arduino/arduino-cli #709: nrf52

tcpipchip, 2020-05-17:
## Bug Report

i cant see nrf52 burn option to Softdevice, only to app

### Current behavior

Dont have to Softdevice

Today i use the NRF5232 and a now i want to use BLE, the upload cli only burn the APP, but i need to burn the Softdevice s132 too

### Expected behavior

### Environment

- CLI version (output of `arduino-cli version`): updated
- OS and platform: Windows

### Additional context

tcpipchip, 2020-05-17:
And how ARDUINO-CLI detects modules that does not use UART to burning ?

tcpipchip, 2020-05-17:
please, add this option to the JSON file:
https://sandeepmistry.github.io/arduino-nRF5/package_nRF5_boards_index.json
https://sandeepmistry.github.io

to
arduino-cli.yaml

You can see that the option is missing

It is a bad JSON file :-(

I get this from the JSON options

when I call :

board details - Arduino CLI
https://arduino.github.io

arduino-cli board details .....

I request it in JSON format

arduino-cli  --format json board details arduino:avr:nano

then it returns the options for the board

and I create the combos

If the JSON file does not have the option, then I don't even know that it exists

"missing feature"

No programmers detect

per1234, 2020-05-17:
>Dont have option to burn the Softdevice

It does. You just need to specify the option via the FQBN. See the detailed explanation here: https://github.com/arduino/arduino-cli/issues/591

>You can see that the option is missing
>
>I get this from the JSON options
>
>when I call :
>
>board details - Arduino CLI

It's not missing when I run the command:
```
$ ./arduino-cli board details sandeepmistry:nRF5:Generic_nRF52832 --additional-urls https://sandeepmistry.github.io/arduino-nRF5/package_nRF5_boards_index.json
Board name:     Generic nRF52

Required tools: sandeepmistry:gcc-arm-none-eabi       5_2-2015q4
                sandeepmistry:openocd                 0.10.0-dev.nrf5

Option:         Softdevice                            softdevice
                None                                ✔ softdevice=none
                S132                                  softdevice=s132

Option:         Low Frequency Clock                   lfclk
                Crystal Oscillator                  ✔ lfclk=lfxo
                RC Oscillator                         lfclk=lfrc
                Synthesized                           lfclk=lfsynt
                                              5_2-2015q4

Required tools:        sandeepmistry:openocd                                                           0.10.0-dev.nrf5

Option:                Softdevice                                                                      softdevice
                       None                                                                          ✔ softdevice=none
                       S132                                                                            softdevice=s132
Option:                Low Frequency Clock                                                             lfclk
                       Crystal Oscillator                                                            ✔ lfclk=lfxo
                       RC Oscillator                                                                   lfclk=lfrc
                       Synthesized                                                                     lfclk=lfsynt
```

>"missing feature"
>
>No programmers detect

That's correct. Arduino CLI does not yet support external programmers. We already have a feature request for this: https://github.com/arduino/arduino-cli/issues/27

Until that feature is added, you'll either need to use the Arduino IDE, or the upload tool directly when you need the "Upload Using Programmer" or "Burn Bootloader" functionality.

---
Since it seems that I've provided explanations for the issues you're having and we already have an existing feature request for external programmer support, I'll go ahead and close this. If you are experiencing an additional bug that hasn't been resolved by this reply, feel free to comment here with more information (exact command you're running and the full and exact output from Arduino CLI is essential) and I'll reopen the issue.

closed 2020-05-17 by per1234