arduino/arduino-cli #713: Document library search paths

itsayellow, 2020-05-19:
This is a request that somewhere in the documentation the library search paths would be documented.

Currently I have my own library which lives in a git repo (and is not part of the arduino libraries index.)  I couldn't find anything in the documentation about where to put it so that `arduino-cli compile` would find it.  

Finally in a non-arduino webpage I found that if you put a library in `~/Documents/Arduino/libraries` (on macOS) then `arduino-cli compile` will find it.  It would be nice if this were documented somewhere.

per1234, 2020-05-19:
This information is documented in the "Sketch Build Process" documentation:
https://arduino.github.io/arduino-cli/sketch-build-process/#location-priority

However, that documentation was written with the Arduino IDE in mind and could use some updates to be more friendly to the Arduino CLI user. Arduino CLI uses the same system, but Arduino CLI uses the term "user directory" in place of the Arduino IDE's "sketchbook".

Note that the user directory location is configurable. You can find the location with `arduino-cli config dump`. You can configure it in the `directories.data` key of the arduino-cli.yaml configuration file created by `arduino-cli config init`, or by setting the `ARDUINO_DIRECTORIES_USER` environment variable.

itsayellow, 2020-05-20:
That is very helpful information, thanks!  Your last paragraph would be wonderful in the docs somewhere.

I did read that exact passage that you linked to, and could not parse that any of it referred to `~/Documents/Arduino/libraries`  

And yes, the "sketchbook" was the most confusing keyword.  I assumed that `{sketchbook path}` referred to my current project directory (e.g. where my *.ino files are for the specific project I was compiling.)  I didn't realize that `{sketchbook path}` means `~/Documents/Arduino`.

Agreed, It would be nice to use the same keyword `user` that the `arduino-cli dump config` uses in the documentation instead of `sketchbook`.

Thanks for the clarifications.

closed 2020-06-18 by per1234

schnoberts1, 2024-01-19:
+1. 

In the IDE if I set the Sketchbook location in preferences it finds all the libraries I have in `<location>/libraries`. If I simply do this:

```
ARDUINO_DIRECTORIES_USER=/Users/andy/code/dbc/gauge/SquareLine_Project/ arduino-cli compile --profile dbc
```

the libraries aren't auto-discovered. I'd like the option for it to work like it does in the IDE. To be fair, what I really want is a transparent build functions in CMake so I can use tested and scalable existing build solutions (I know there's a bunch of open source stuff) but I'd be happy if I could get the CLI to replicate what the IDE does with the same minimal effort.