arduino/arduino-cli #722: Could not upload to Adafruit Feather M0

dajtxx, 2020-05-27:
## Bug Report

### Current behavior

C:\Users\DAVITAYL\sloeber-workspace>c:\users\davitayl\arduino-cli board list
Port Type              Board Name          FQBN                              Core
COM3 Serial Port       Unknown
COM6 Serial Port (USB) Adafruit Feather M0 adafruit:samd:adafruit_feather_m0 adafruit:samd

C:\Users\DAVITAYL\sloeber-workspace>c:\users\DAVITAYL\arduino-cli compile --fqbn adafruit:samd:adafruit_feather_m0 PumpController
Sketch uses 57272 bytes (21%) of program storage space. Maximum is 262144 bytes.

C:\Users\DAVITAYL\sloeber-workspace>c:\users\DAVITAYL\arduino-cli upload -p COM6 --fqbn adafruit:samd:adafruit_feather_m0 PumpController
Error during Upload: cannot execute upload tool: exec: "{runtime.tools.bossac-1.7.0-arduino3.path}/bossac.exe": file does not exist

The upload didn't work because arduino-cli didn't find bossac.

### Expected behavior

The sketch can be compiled and uploaded from within the Arduino IDE so bossac must exist somewhere.

### Environment

- CLI version (output of `arduino-cli version`):
arduino-cli Version: 0.10.0 Commit: ec5c3ed

- OS and platform:
Windows 10

### Additional context

dajtxx, 2020-05-27:
Here is the IDE uploading:

```
Uploading using selected port: COM7
C:\Users\DAVITAYL\Documents\ArduinoData\packages\arduino\tools\bossac\1.7.0-arduino3/bossac.exe -i -d --port=COM7 -U true -i -e -w -v C:\Users\DAVITAYL\AppData\Local\Temp\arduino_build_150081/PumpController.ino.bin -R 
Set binary mode
readWord(addr=0)=0x20007ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010305
version()=v2.0 [Arduino:XYZ] Mar  5 2016 17:46:52
chipId=0x10010005
Connected at 921600 baud
readWord(addr=0)=0x20007ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010305
Atmel SMART device 0x10010005 found
write(addr=0x20004000,size=0x34)
writeWord(addr=0x20004030,value=0x10)
writeWord(addr=0x20004020,value=0x20008000)
Device       : ATSAMD21G18A
```

per1234, 2020-05-27:
Hi @dajtxx. Please post the output from this command:
```
arduino-cli core list
```

dajtxx, 2020-05-27:
```
C:\Users\DAVITAYL\sloeber-workspace>c:\users\DAVITAYL\arduino-cli core list
ID            Installed Latest Name
adafruit:samd 1.5.13    1.5.13
```

I also tried hard-coding the path to bossa by making this change in two places in the adafruit platform.txt file:

```
#tools.bossac.path={runtime.tools.bossac-1.7.0-arduino3.path}
tools.bossac.path=C:/Users/DAVITAYL/Documents/ArduinoData/packages/arduino/tools/bossac/1.7.0-arduino3
```

This is what I'm getting after hard-coding the paths:

```
C:\Users\DAVITAYL\sloeber-workspace>c:\users\DAVITAYL\arduino-cli board list
Port Type              Board Name          FQBN                              Core
COM3 Serial Port       Unknown
COM7 Serial Port (USB) Adafruit Feather M0 adafruit:samd:adafruit_feather_m0 adafruit:samd

C:\Users\DAVITAYL\sloeber-workspace>c:\users\DAVITAYL\arduino-cli upload -v --log-level debug -p COM7 --fqbn adafruit:samd:adafruit_feather_m0 PumpController
No new serial port detected.
Error during Upload: cannot execute upload tool: exec: "{runtime.tools.bossac-1.7.0-arduino3.path}/bossac.exe": file does not exist

C:\Users\DAVITAYL\sloeber-workspace>c:\users\DAVITAYL\arduino-cli board list
Port Type              Board Name          FQBN                              Core
COM3 Serial Port       Unknown
COM6 Serial Port (USB) Adafruit Feather M0 adafruit:samd:adafruit_feather_m0 adafruit:samd

C:\Users\DAVITAYL\sloeber-workspace>c:\users\DAVITAYL\arduino-cli upload -v --log-level debug -p COM6 --fqbn adafruit:samd:adafruit_feather_m0 PumpController
Error during Upload: cannot perform reset: opening port: Invalid serial port

C:\Users\DAVITAYL\sloeber-workspace>echo %ErrorLevel%
1
```

umbynos, 2022-10-18:
Hi @dajtxx could you please try with the latest version of the cli? A lot of fix were made in the upload https://github.com/arduino/arduino-cli/releases

zvonler, 2023-10-18:
I have a Feather M0 and just confirmed that I was able to use
```
arduino-cli  Version: 0.34.2 Commit: 963c1a76 Date: 2023-09-11T09:50:44Z
```
to upload and execute a sketch.

Is there anything else I could test to help close this issue?

alessio-perugini, 2023-10-18:
@zvonler :superhero: Do you mind trying with the CLI [0.10.0](https://github.com/arduino/arduino-cli/releases/tag/0.10.0 ) ? I expect it to fail, if it does we can consider this fixed :nerd_face:

zvonler, 2023-10-18:
Using the downloaded 0.10.0 MacOS binary, after re-compiling the sketch, the same upload command that worked before fails:
```
Error during Upload: cannot execute upload tool: fork/exec {runtime.tools.bossac-1.8.0-48-gb176eee.path}/bossac: no such file or directory
```
It's not exactly the same message dajtxx got but I think that's just a different OS message.

alessio-perugini, 2023-10-19:
@zvonler Perfect, I'm closing this then. Thanks for the help :partying_face:

closed 2023-10-19 by alessio-perugini