arduino/arduino-cli #723: Cannot install AVR code

dajtxx, 2020-05-27:
## Bug Report

### Current behavior

I am trying to install the AVR core to see if uploads work with that, to help debug my Feather problem.

The AVR core installation fails.

Again, I can compile and upload a sketch from within the IDE.

```
C:\Users\DAVITAYL\Documents\Arduino>arduino-cli core list -v
INFO[0000] Using config file: C:\Users\DAVITAYL\AppData\Local\Arduino15\arduino-cli.yaml
INFO[0000] arduino-cli version 0.10.0
INFO[0000] Checking if CLI is Bundled into the IDE
INFO[0000] Loading hardware from: C:\Users\DAVITAYL\AppData\Local\Arduino15\packages
INFO[0000] Loading package adafruit from: C:\Users\DAVITAYL\AppData\Local\Arduino15\packages\adafruit\hardware
INFO[0000] Loaded platform                               platform="adafruit:samd@1.5.13"
INFO[0000] Loading package arduino from: C:\Users\DAVITAYL\AppData\Local\Arduino15\packages\arduino\hardware
INFO[0000] Checking existence of 'tools' path: C:\Users\DAVITAYL\AppData\Local\Arduino15\packages\arduino\tools
INFO[0000] Loading tools from dir: C:\Users\DAVITAYL\AppData\Local\Arduino15\packages\arduino\tools
INFO[0000] Loaded tool                                   tool="arduino:CMSIS@4.5.0"
INFO[0000] Loaded tool                                   tool="arduino:CMSIS-Atmel@1.2.0"
INFO[0000] Loaded tool                                   tool="arduino:arduinoOTA@1.2.1"
INFO[0000] Loaded tool                                   tool="arduino:arduinoOTA@1.3.0"
INFO[0000] Loaded tool                                   tool="arduino:arm-none-eabi-gcc@7-2017q4"
INFO[0000] Loaded tool                                   tool="arduino:avrdude@6.3.0-arduino17"
INFO[0000] Loaded tool                                   tool="arduino:bossac@1.7.0-arduino3"
INFO[0000] Loaded tool                                   tool="arduino:bossac@1.8.0-48-gb176eee"
INFO[0000] Loaded tool                                   tool="arduino:openocd@0.10.0-arduino7"
INFO[0000] Loading package builtin from: C:\Users\DAVITAYL\AppData\Local\Arduino15\packages\builtin
INFO[0000] Checking existence of 'tools' path: C:\Users\DAVITAYL\AppData\Local\Arduino15\packages\builtin\tools
INFO[0000] Loading tools from dir: C:\Users\DAVITAYL\AppData\Local\Arduino15\packages\builtin\tools
INFO[0000] Loaded tool                                   tool="builtin:ctags@5.8-arduino11"
INFO[0000] Loaded tool                                   tool="builtin:serial-discovery@1.0.0"
INFO[0000] Adding libraries dir                          dir="C:\\Users\\DAVITAYL\\Documents\\Arduino\\libraries" location=user
INFO[0000] current best: adafruit:samd@1.5.13            bundle=false managed=true version=1.5.13
INFO[0000] Adding libraries dir                          dir="C:\\Users\\DAVITAYL\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.13\\libraries" location=platform
INFO[0000] Executing `arduino core list`
INFO[0000] current best: adafruit:samd@1.5.13            bundle=false managed=true version=1.5.13
ID            Installed Latest Name
adafruit:samd 1.5.13    1.5.13

C:\Users\DAVITAYL\Documents\Arduino>arduino-cli core install arduino:avr -v
INFO[0000] Using config file: C:\Users\DAVITAYL\AppData\Local\Arduino15\arduino-cli.yaml
INFO[0000] arduino-cli version 0.10.0
INFO[0000] Checking if CLI is Bundled into the IDE
INFO[0000] Loading hardware from: C:\Users\DAVITAYL\AppData\Local\Arduino15\packages
INFO[0000] Loading package adafruit from: C:\Users\DAVITAYL\AppData\Local\Arduino15\packages\adafruit\hardware
INFO[0000] Loaded platform                               platform="adafruit:samd@1.5.13"
INFO[0000] Loading package arduino from: C:\Users\DAVITAYL\AppData\Local\Arduino15\packages\arduino\hardware
INFO[0000] Checking existence of 'tools' path: C:\Users\DAVITAYL\AppData\Local\Arduino15\packages\arduino\tools
INFO[0000] Loading tools from dir: C:\Users\DAVITAYL\AppData\Local\Arduino15\packages\arduino\tools
INFO[0000] Loaded tool                                   tool="arduino:CMSIS@4.5.0"
INFO[0000] Loaded tool                                   tool="arduino:CMSIS-Atmel@1.2.0"
INFO[0000] Loaded tool                                   tool="arduino:arduinoOTA@1.2.1"
INFO[0000] Loaded tool                                   tool="arduino:arduinoOTA@1.3.0"
INFO[0000] Loaded tool                                   tool="arduino:arm-none-eabi-gcc@7-2017q4"
INFO[0000] Loaded tool                                   tool="arduino:avrdude@6.3.0-arduino17"
INFO[0000] Loaded tool                                   tool="arduino:bossac@1.7.0-arduino3"
INFO[0000] Loaded tool                                   tool="arduino:bossac@1.8.0-48-gb176eee"
INFO[0000] Loaded tool                                   tool="arduino:openocd@0.10.0-arduino7"
INFO[0000] Loading package builtin from: C:\Users\DAVITAYL\AppData\Local\Arduino15\packages\builtin
INFO[0000] Checking existence of 'tools' path: C:\Users\DAVITAYL\AppData\Local\Arduino15\packages\builtin\tools
INFO[0000] Loading tools from dir: C:\Users\DAVITAYL\AppData\Local\Arduino15\packages\builtin\tools
INFO[0000] Loaded tool                                   tool="builtin:ctags@5.8-arduino11"
INFO[0000] Loaded tool                                   tool="builtin:serial-discovery@1.0.0"
INFO[0000] Adding libraries dir                          dir="C:\\Users\\DAVITAYL\\Documents\\Arduino\\libraries" location=user
INFO[0000] current best: adafruit:samd@1.5.13            bundle=false managed=true version=1.5.13
INFO[0000] Adding libraries dir                          dir="C:\\Users\\DAVITAYL\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.13\\libraries" location=platform
INFO[0000] Executing `arduino core install`
WARN[0000] Tool already installed                        platform="arduino:avr@1.8.2" tool="arduino:avrdude@6.3.0-arduino17"
Tool arduino:avrdude@6.3.0-arduino17 already installed
WARN[0000] Tool already installed                        platform="arduino:avr@1.8.2" tool="arduino:arduinoOTA@1.3.0"
Tool arduino:arduinoOTA@1.3.0 already installed
Downloading packages...
arduino:avr-gcc@7.3.0-atmel3.6.1-arduino5 already downloaded
arduino:avr@1.8.2 already downloaded
INFO[0000] Installing tool                               Tool="arduino:avr-gcc@7.3.0-atmel3.6.1-arduino5"
Installing arduino:avr-gcc@7.3.0-atmel3.6.1-arduino5...
WARN[0009] Cannot install tool                           Tool="arduino:avr-gcc@7.3.0-atmel3.6.1-arduino5" error="moving extracted archive to destination dir: rename C:\\Users\\DAVITAYL\\AppData\\Local\\Arduino15\\tmp\\package-095788307\\avr C:\\Users\\DAVITAYL\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5: Access is denied."
INFO[0009] Installing platform                           platform="arduino:avr@1.8.2"
Installing arduino:avr@1.8.2...
ERRO[0014] Cannot install platform                       error="moving extracted archive to destination dir: rename C:\\Users\\DAVITAYL\\AppData\\Local\\Arduino15\\tmp\\package-183218518\\avr C:\\Users\\DAVITAYL\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2: Access is denied." platform="arduino:avr@1.8.2"
Error during install: moving extracted archive to destination dir: rename C:\Users\DAVITAYL\AppData\Local\Arduino15\tmp\package-183218518\avr C:\Users\DAVITAYL\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2: Access is denied.
ERRO[0014] Error during install: moving extracted archive to destination dir: rename C:\Users\DAVITAYL\AppData\Local\Arduino15\tmp\package-183218518\avr C:\Users\DAVITAYL\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2: Access is denied.

C:\Users\DAVITAYL\Documents\Arduino>dir C:\Users\DAVITAYL\AppData\Local\Arduino15\packages\arduino\hardware
 Volume in drive C is System
 Volume Serial Number is 0CAD-5409

 Directory of C:\Users\DAVITAYL\AppData\Local\Arduino15\packages\arduino\hardware

27/05/2020  02:36 PM    <DIR>          .
27/05/2020  02:36 PM    <DIR>          ..
               0 File(s)              0 bytes
               2 Dir(s)  76,647,366,656 bytes free
```

### Environment
arduino-cli Version: 0.10.0 Commit: ec5c3ed
Win10.

dajtxx, 2020-06-15:
I have just tried this with a newly built binary from master, and had the same problem.

`DAVITAYL-T460+DAVITAYL@DAVITAYL-T460 MINGW64 ~/Documents/GitHub/arduino-cli (master)
$ ./arduino-cli.exe core list
ID            Installed Latest Name
adafruit:samd 1.5.14    1.5.14

DAVITAYL-T460+DAVITAYL@DAVITAYL-T460 MINGW64 ~/Documents/GitHub/arduino-cli (master)
$ ./arduino-cli.exe core search avr
ID                   Version Name
arduino:avr          1.8.2   Arduino AVR Boards
arduino:megaavr      1.8.6   Arduino megaAVR Boards
atmel-avr-xminis:avr 0.6.0   Atmel AVR Xplained-minis
emoro:avr            3.2.2   EMORO 2560
littleBits:avr       1.0.0   littleBits Arduino AVR Modules

DAVITAYL-T460+DAVITAYL@DAVITAYL-T460 MINGW64 ~/Documents/GitHub/arduino-cli (master)
$ ./arduino-cli.exe core install arduino:avr
Tool arduino:avrdude@6.3.0-arduino17 already installed
Tool arduino:arduinoOTA@1.3.0 already installed
Downloading packages...
arduino:avr-gcc@7.3.0-atmel3.6.1-arduino5 already downloaded
arduino:avr@1.8.2 already downloaded
Installing arduino:avr-gcc@7.3.0-atmel3.6.1-arduino5...
Installing arduino:avr@1.8.2...
Error during install: moving extracted archive to destination dir: rename C:\Users\DAVITAYL\AppData\Local\Arduino15\tmp\package-659673658\avr C:\Users\DAVITAYL\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2: Access is denied.`

dajtxx, 2020-06-15:
I just checked the directory C:\Users\DAVITAYL\AppData\Local\Arduino15 and its properties dialog shows the 'Read-only' checkbox checked. I cleared it, and said apply that to all subfolders etc. I tried to install the AVR code again and got the same error.

The 'Read-only' attr is set again. I cleared it again and reopened the props dialog without doing anything else and it is still set. The security tab says my user has full permissions.

Anyway, I can do things in that dir tree in the shell:

```
DAVITAYL-T460+DAVITAYL@DAVITAYL-T460 MINGW64 ~/Documents/GitHub/arduino-cli (master)
$ ls -ld /c/Users/DAVITAYL/AppData/Local/Arduino15/
drwxr-xr-x 1 DAVITAYL-T460+DAVITAYL 197121 0 Jun 13 18:31 /c/Users/DAVITAYL/AppData/Local/Arduino15//

DAVITAYL-T460+DAVITAYL@DAVITAYL-T460 MINGW64 ~/Documents/GitHub/arduino-cli (master)
$ mkdir -p /c/Users/DAVITAYL/AppData/Local/Arduino15/packages/arduino/hardware/avr/1.8.2

DAVITAYL-T460+DAVITAYL@DAVITAYL-T460 MINGW64 ~/Documents/GitHub/arduino-cli (master)
$ ls -ld /c/Users/DAVITAYL/AppData/Local/Arduino15/packages/arduino/hardware/avr/1.8.2
drwxr-xr-x 1 DAVITAYL-T460+DAVITAYL 197121 0 Jun 15 17:12 /c/Users/DAVITAYL/AppData/Local/Arduino15/packages/arduino/hardware/avr/1.8.2/

DAVITAYL-T460+DAVITAYL@DAVITAYL-T460 MINGW64 ~/Documents/GitHub/arduino-cli (master)
$ ./arduino-cli.exe core install arduino:avr
Tool arduino:avrdude@6.3.0-arduino17 already installed
Tool arduino:arduinoOTA@1.3.0 already installed
Downloading packages...
arduino:avr-gcc@7.3.0-atmel3.6.1-arduino5 already downloaded
arduino:avr@1.8.2 already downloaded
Installing arduino:avr-gcc@7.3.0-atmel3.6.1-arduino5...
Installing arduino:avr@1.8.2...
Error during install: moving extracted archive to destination dir: rename C:\Users\DAVITAYL\AppData\Local\Arduino15\tmp\package-670048666\avr C:\Users\DAVITAYL\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2: Access is denied.
```

dajtxx, 2020-06-15:
I just tried it in a Windows sandbox instance and it worked fine. Must be my env, but I don't know what. Could installing the Windows Store version of the IDE cause this?

```
C:\Users\WDAGUtilityAccount>arduino-cli.exe core update-index
Updating index: package_index.json downloaded

C:\Users\WDAGUtilityAccount>arduino-cli.exe core install arduino:avr
Downloading packages...
arduino:avr-gcc@7.3.0-atmel3.6.1-arduino5 downloaded
arduino:avrdude@6.3.0-arduino17 downloaded
arduino:arduinoOTA@1.3.0 downloaded
arduino:avr@1.8.2 downloaded
Installing arduino:avr-gcc@7.3.0-atmel3.6.1-arduino5...
arduino:avr-gcc@7.3.0-atmel3.6.1-arduino5 installed
Installing arduino:avrdude@6.3.0-arduino17...
arduino:avrdude@6.3.0-arduino17 installed
Installing arduino:arduinoOTA@1.3.0...
arduino:arduinoOTA@1.3.0 installed
Installing arduino:avr@1.8.2...
arduino:avr@1.8.2 installed

C:\Users\WDAGUtilityAccount>arduino-cli.exe core list
ID          Installed Latest Name
arduino:avr 1.8.2     1.8.2  Arduino AVR Boards
```

dajtxx, 2020-06-15:
Rebooted, removed the AppData\local\Arduino15 dir entirely and tried to start from scratch with core update-index, core install arduino:avr.

Same error :(

Can I tell it to install stuff somewhere else?

ywueoei, 2020-10-12:
user\AppData\Local\Arduino15\packages\builtin\tools\serial-discovery\1.0.0: Access is denied.

I have the same problem. Is there a solution

ubidefeo, 2020-10-12:
@ywueoei 
you need to be have a certain level of system admin's rights in order to install packages and tools

matthijskooijman, 2020-10-12:
>  you need to be have a certain level of system admin's rights in order to install packages and tools

Are you sure? I would think this only applies to driver installs, the package and tools themselves are just unpacked into your user directory, which should be writable by the current user without extra permissions, right?

I wonder if there is maybe a virus scanner that blocks access? Or maybe the Windows 10 store version has additional sandboxing going on that restricts access to files even in the user's directory?

ubidefeo, 2020-10-12:
@matthijskooijman 
you're right.
I'm not a Windows user so had to go check on my VM.
what's strange is the path reported by @ywueoei .
My path is `C:\Users\ubi\AppData\Local\Arduino15>`, while I see `user\APpData\Local\`

sastorer-ms, 2021-07-06:
I'm still seeing this issue in 0.18.3

per1234, 2021-07-07:
Hi @sastorer-ms. I would first try restarting your computer. Some process might have the target path locked, although you could try to find it and end it, restarting is an easier way to make sure to start fresh.

---
If that doesn't help, this sort of error is might be caused by interference from the antivirus software doing an on-access scan. You can check that by **TEMPORARILY** disabling your antivirus just long enough to run the installation to see if the problem goes away, then turn the antivirus back on.

If the problem doesn't occur with the antivirus off you will need to adjust the settings of your antivirus to put the appropriate file, folder, or process on the "allow list" so it doesn't interfere with compilation.

Please be cautious about working without an antivirus. This is only about temporarily disabling it for a quick test. If you don't feel comfortable doing that, fine. You can try going straight to configuring the antivirus to not interfere with the Arduino software.

sudobob, 2021-12-07:
Had the same problem with **arduino-cli.exe alpha Version: 0.18.3 Commit: d710b642 Date: 2021-05-14T12:36:58Z** under windows  **arduino-cli lib install xxx : Access is denied** My destination dir was in my dropbox and the solution was to temporarily install to C:\ then move them manually. In git for windows bash:
  
```
#given your regular arduino-cli lib location is set like this:
export ARDUINO_DIRECTORIES_USER="C:\dropbox\arduino\sketches"
# temporarily set lib dest to C: somewhere
export ARDUINO_DIRECTORIES_USER="C:\windows\temp"

# install your our libraries
ac lib install foo
ac lib install bar 
# (copy libraries/ dir from src to the regular lib location
# reset where arduino-cli  to look at the regular lib location
export ARDUINO_DIRECTORIES_USER="C:\dropbox\arduino\sketches"
# prove it to yourself it worked
arduino-cli lib list
```