arduino/arduino-cli #740: Breaking my license terms

bouk, 2020-06-04:
This project [is using](https://github.com/arduino/arduino-cli/blob/ded057e60d1ed8f79700b9f22862b6956dcf6033/commands/daemon/monitor_test.go#L27) my monkey project, which has a [license](https://github.com/bouk/monkey/blob/master/LICENSE.md) that states you're not allowed to use it.

Please remove all references to it.

matthijskooijman, 2020-06-04:
First off, I'm not a core arduino-cli developer, so do not treat this comment as an official reply (and IMHO it seems clear-cut: there is no license that allows using your project, so its usage should be removed).

However, personally I'm a bit confused about your license choice. Why go through all the effort of publishing your code, writing a proper README and even a long blogpost (that says "If you’re not interested in how it works and you just want to do monkey patching, then you can find the library here.") but then do not put an open license on this? All this seems to encourage people to use the code, except for the license. Of course you are completely entitled to make your own licensing choices and you have no obligation to even answer this, but I'm just genuinely curious what the rationale is here... Would you care to share your ideas about this?

bouk, 2020-06-04:
It's a satirical project, one big joke for my own entertainment. And I've gotten a lot of laughs out of it! There's no way to implement this safely, I put that license on it because people read the blogpost and README and still thought it would be a good idea to use it.

rsora, 2020-06-04:
Thanks @bouk for taking the time to warn us and to reply to the @matthijskooijman question.

matthijskooijman, 2020-06-04:
@bouk, hm, I hadn't really looked at the method you used, and now I looked more closely I can indeed see how that's fun to figure out, but not really suited to be used in practice. Thanks for replying :-)

rsora, 2020-09-04:
Solved by https://github.com/arduino/arduino-cli/pull/937

closed 2020-09-04 by rsora