arduino/arduino-cli #745: windows - following symlink to network share not working

novalisek, 2020-06-09:
## Bug Report

### Current behavior

`C:\Program Files (x86)\arduino-nightly\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\arduino-nightly\hardware -hardware C:\Users\martin\AppData\Local\Arduino15\packages -hardware D:\programy\arduino\hardware -tools C:\Program Files (x86)\arduino-nightly\tools-builder -tools C:\Program Files (x86)\arduino-nightly\hardware\tools\avr -tools C:\Users\martin\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\arduino-nightly\libraries -libraries D:\programy\arduino\libraries -fqbn=arduino:sam:arduino_due_x_dbg -ide-version=10813 -build-path d:\temp\arduino_build_807214 -warnings=none -build-cache d:\temp\arduino_cache_153221 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\martin\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.bossac-1.6.1-arduino.path=C:\Users\martin\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\martin\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\martin\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose D:\git\arduino\test\test.ino`

Hi,
I am running Windows 7 under VirtualBox and I have folder shared via Virtualbox, it can be seen as \\vboxsrv\programy
I have mapped it using  
`mklink /D d:\programy \\vboxsrv\programy`
Now I only see

`loading hardware from D:\programy\arduino\hardware: following possible symlink D:\programy\arduino\hardware: CreateFile \\VBOXSVR\programyarduino: Network name cannot be found.`

it seems that it is missing \ in programyarduino (it should be \\VBOXSVR\programy\arduino)

### Expected behavior

It should be possible to disable following symlinks or it should be repaired so that following them doesn't cause errors

### Environment

- CLI version (output of `arduino-cli version`):
Arduino Builder 1.5.3

- OS and platform:
Windows 7 64bit running under Virtualbox with network shares mapped to local directories

### Additional context

Arduino Builder 1.4.7 from Arduino 1.8.10 is working fine, I noticed this problem for the first time in Arduino 1.8.12

novalisek, 2020-09-18:
anybody knows how to repair that?

umbynos, 2022-10-20:
Hi @novalisek I think you are using the old and deprecated https://github.com/arduino/arduino-builder,
could you please try if this issue applied also to the latest version of the cli? https://github.com/arduino/arduino-cli/releases

novalisek, 2022-10-21:
Hi @umbynos 
 tried to install Arduino IDE 1.8.19,  still using same link to virtualbox shared folder made by " mklink /D d:\programy \\vboxsrv\programy"  (which also contains hardware definitions for ESP32) and I am still getting

```
Arduino: 1.8.19 (Windows 7), Board: "ESP32 Wrover Module, Default, QIO, 80MHz, 921600, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware D:\programy\arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\programy\arduino\libraries -fqbn=espressif:esp32:esp32wrover:PartitionScheme=default,FlashMode=qio,FlashFreq=80,UploadSpeed=921600,DebugLevel=none -ide-version=10819 -build-path d:\temp\arduino_build_567424 -warnings=none -build-cache d:\temp\arduino_cache_34451 -prefs=build.warn_data_percentage=75 -verbose d:\temp\untitled1162953461.tmp\sketch_oct21a\sketch_oct21a.ino

loading hardware from D:\programy\arduino\hardware: following possible symlink D:\programy\arduino\hardware: CreateFile \\vboxsrv\programyarduino: Síťový název nelze nalézt.

Error compiling for board ESP32 Wrover Module.
```

Is arduino-builder from Arduino 1.8.19 deprecated?  I didn't try with Arduino IDE 2.0.0 yet.

I think the problem is somewhere in the path - it is showing "\\vboxsrv\programyarduino" but it should be "\\vboxsrv\programy\arduino"

per1234, 2022-10-22:
>tried to install Arduino IDE 1.8.19

You must use Arduino CLI @novalisek. Reporting results you are getting from Arduino IDE 1.8.19 is not useful to us because that is an outdated version of the IDE in maintenance mode.

I'll provide instructions you can follow to test this with Arduino CLI:

1. Click the following link to download the latest nightly build of Arduino CLI:
   https://downloads.arduino.cc/arduino-cli/nightly/arduino-cli_nightly-latest_Windows_64bit.zip
1. Unzip the downloaded file.
1. Open a Windows cmd terminal (I assume cmd is your preferred shell due to your use of `mklink`, feel free to adapt the instructions if you prefer another shell) in the folder that contains the file `arduino-cli.exe` that was unzipped.
1. Type the following command:
   ```text
   arduino-cli config init
   ```
1. Press the <kbd>**Enter**</kbd> key.
1. Type the following command:
   ```text
   arduino-cli config set directories.user D:\programy\arduino
   ```
1. Press the <kbd>**Enter**</kbd> key.
1. Type the following command:
   ```text
   arduino-cli sketch new d:\temp\TestSketch
   ```
1. Press the <kbd>**Enter**</kbd> key.
1. Type the following command:
   ```text
   arduino-cli compile --verbose --fqbn espressif:esp32:esp32wrover:PartitionScheme=default,FlashMode=qio,FlashFreq=80,UploadSpeed=921600,DebugLevel=none d:\temp\TestSketch
   ```
1. Press the <kbd>**Enter**</kbd> key.

This will produce a compilation under conditions equivalent to what you did in Arduino IDE 1.8.19.

Do you still get an error when you do that, or is the compilation successful?

>Is `arduino-builder` from Arduino 1.8.19 deprecated?

Yes, as was explained to you already.

>I didn't try with Arduino IDE 2.0.0 yet.

Please use Arduino CLI. There is already a known issue specific to the Arduino IDE 2.x codebase related to dealing with this type of UNC path: https://github.com/arduino/arduino-ide/issues/1475. That bug might interfere with your ability to test whether there is some problem with Arduino CLI's handling of these conditions.

novalisek, 2022-10-22:
@per1234 hello, sorry for using IDE with deprecated arduino-cli, may be they should add some note to download page, that it contains deprecated utils and shouldn't be used.

Anyway I tried your steps, and I am still getting that error:

```
D:\install\arduino-cli_nightly-20221022_Windows_64bit>arduino-cli compile --verbose --fqbn espressif:esp32:esp32wrover:PartitionScheme=default,FlashMode=qio,FlashFreq=80,UploadSpee
d=921600,DebugLevel=none d:\temp\TestSketch
Error initializing instance: Error loading hardware platform: following symlink D:\programy\arduino\hardware: CreateFile \\vboxsrv\programyarduino: Síťový název nelze nalézt. (= the network name cannot be found)

Used platform Version Path

Error during build: Platform 'espressif:esp32' not found: platform not installed
Platform espressif:esp32 is not found in any known index
Maybe you need to add a 3rd party URL?
```

I think there is still "\\"  missing in the path

cmaglie, 2022-11-15:
> I think there is still "\" missing in the path

Do you mean that `\\vboxsrv\programyarduino` should be `\\\vboxsrv\programyarduino`?

novalisek, 2022-11-15:
@cmaglie it should be   `\\vboxsrv\programy\arduino`  because   `d:\programy\`  is directory symbolic link (mklink /D) to `\\vbox\programy`  and `arduino` is subdirectory of that.