arduino/arduino-cli #751: Cannot build on windows due to assumption of *nix shell and utils

dajtxx, 2020-06-15:
## Bug Report

### Current behavior
arduino-cli>task build
go: downloading golang.org/x/lint v0.0.0-20190313153728-d0100b6bd8b3
go: downloading golang.org/x/tools v0.0.0-20190524140312-2c0ae7006135
"grep": executable file not found in $PATH
"tr": executable file not found in $PATH
go: downloading github.com/sirupsen/logrus v1.4.2
...
go: downloading golang.org/x/crypto v0.0.0-20190308221718-c2843e01d9a2
io: read/write on closed pipe
task: Command "echo `go list ./... | grep -v legacy | tr '\n' ' '`" in taskvars file failed: exit status 1

### Expected behavior

The build prerequisites do not specify you need to be in *nix environment so I assumed I could build in a Windows cmd shell.

### Environment

- master branch source code
- go 1.14
- task 2.8.1
- Win10/amd64.

### Additional context

ubidefeo, 2020-06-15:
@dajtxx 
for windows you're better off downloading the binary.
If you look at where the error is it's trying to run `echo` and `grep` and I believe unless you're using PowerShell it won't work on Windows.
@per1234 can you confirm?

per1234, 2020-06-15:
I just tried it out and the same error happens with PowerShell.

I'll confess that I frequently build Arduino CLI on Windows. I use the Git Bash that comes with Git for Windows and `task build` works perfectly:
https://gitforwindows.org/

dajtxx, 2020-06-15:
Thanks, that works. It would be worth documenting.

closed 2020-06-15 by dajtxx