arduino/arduino-cli #755: Arduino Cli is not working on rpi zero w.

raj-priyanshu, 2020-06-19:
## Bug Report
This issue is already discussed before. here, https://github.com/arduino/arduino-cli/issues/340
but its old Arduino-CLI version (0.4.0)  and it is not able to find the board(STM32) connected to it. So I tried arm7 build and it is able to find the board but give the unknown in board name, it can compile successfully it but givesome memory error when I trying to upload it.
`
pi@raspberrypi:~ $ ./arduino-cli board list
Port         Type        Board Name FQBN Core
/dev/ttyAMA0 Serial Port Unknown
/dev/ttyS0   Serial Port Unknown
`

`pi@raspberrypi:~ $ ./arduino-cli upload -p /dev/ttyS0 --fqbn stm32duino:STM32F1:genericSTM32F103C $HOME/Arduino/test
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x385c50]

goroutine 1 [running]:
github.com/arduino/go-paths-helper.(*Path).String(...)
	/github/home/go/pkg/mod/github.com/arduino/go-paths-helper@v1.0.1/paths.go:470
github.com/arduino/arduino-cli/arduino/cores.(*ToolRelease).RuntimeProperties(0x2e67760, 0x4)
	/__w/arduino-cli/arduino-cli/arduino/cores/tools.go:136 +0x7c
github.com/arduino/arduino-cli/commands/upload.Upload(0x914bb8, 0x2c2a078, 0x48a3e88, 0x90d358, 0x2c100e0, 0x90d358, 0x2c100e8, 0x60040c, 0x2c8f4f0, 0x2d46540)
	/__w/arduino-cli/arduino-cli/commands/upload/upload.go:128 +0x6d0
github.com/arduino/arduino-cli/cli/upload.run(0x2d3c140, 0x2c67e30, 0x1, 0x5)
	/__w/arduino-cli/arduino-cli/cli/upload/upload.go:73 +0x180
github.com/spf13/cobra.(*Command).execute(0x2d3c140, 0x2c67e00, 0x5, 0x6, 0x2d3c140, 0x2c67e00)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:830 +0x1f4
github.com/spf13/cobra.(*Command).ExecuteC(0xdfb388, 0x42754, 0x0, 0x2c000e0)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:914 +0x230
github.com/spf13/cobra.(*Command).Execute(...)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:864
main.main()
	/__w/arduino-cli/arduino-cli/main.go:26 +0x20
`

rsora, 2020-06-22:
Hi @raj-priyanshu,
Thanks for the Heads up!
Can you please try with the latest release candidate for ARMv6
https://github.com/arduino/arduino-cli/releases/download/0.11.0-rc2/arduino-cli_0.11.0-rc2_Linux_ARMv6.tar.gz
and tell us if you get errors?

Thanks!

raj-priyanshu, 2020-06-29:
still getting the similar problem. i am connecting the board with gpio serial and expecting only `/dev/ttyS0` and board name but the output is as follow.
`pi@raspberrypi:~ $ ./arduino-cli board list`
`Port                Type           Board Name    FQBN Core`
`/dev/ttyAMA0 Serial Port   Unknown                            `
`/dev/ttyS0      Serial Port   Unknown                             `

rsora, 2020-06-29:
@raj-priyanshu Can you please confirm that you installed the `stm32duino:STM32F1` core?

raj-priyanshu, 2020-06-29:
@rsora 
here it is
pi@raspberrypi:~ $ ./arduino-cli core list
ID                 Installed Latest    Name                     
stm32duino:STM32F1 2019.2.28 2019.2.28 STM32F1xx/GD32F1xx boards

rsora, 2020-06-29:
I'll ask to @per1234 to chime in as I'm not an expert in the STM32F1 chip, from CLI side it seems to me that everything was done correctly by the user. 

Maybe this particular board requires some additional config parameters to pass via FQBN?

ubidefeo, 2020-07-01:
hi @raj-priyanshu 
Sorry for the late reply on this, I've assigned the issue to myself in order to take some stuff off of @per1234 's plate, and because this kind of investigation is fun.

Let me share with you what Per found out and what I'm up to in order to resolve the issue and point you in the right direction.
I don't have a Blue Pill handy now, and modifying an ST-Link V2 has failed miserably, but I'll have one in house tomorrow to test.

I brought myself up to speed with STM32duino and setup a Pi Zero with Arduino-CLI which works well, and as soon as I have an F103 with bootloader I'll attempt what you are having problems with.

Here's what Per found out and observed in our conversation based on you referring to GPIO serial on the Pi.:
* you are trying to upload code to the STM32 via the serial port connected to the chip A9 and A10
* you use Raspberry Pi's on-board GPIO UART pins 15 (TXD) and 14 (RXD) respectively
* you have been using [this guide](https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Uploading#generic--unbranded-stm32f103-boards)
* you seem to use an outdated core (2019.2.28) vs the (2020.6.20)

The first outcome is not seeing a board name, which is normal because boards can be identified solely from their USB VID:PID pair, while a serial port is just a serial port.
Per observed that uploading via serial port will require a parameter to be added as extra configuration for the board, and such parameter is `upload_method=serialMethod`, otherwise the core defaults to `DFUUploadMethod` and that won't work over serial.

As I gathered more information I also found out that you'll need the pins BOOT0 set to HIGH and BOOT1 set to LOW in order to be able to upload a new firmware via the Maple bootloader.

My suggestion is that you begin with making sure BOOT0 and BOOT1 are properly configured and that you add the `upload_method` parameter to your `fqbn`
Try the following

 ```
./arduino-cli upload -p /dev/ttyS0 --fqbn stm32duino:STM32F1:genericSTM32F103C:upload_method=serialMethod $HOME/Arduino/test
```

and let me know.
I'll test the same and let you know, but if you succeed, please, report back to us :)

raj-priyanshu, 2020-07-01:
Hi @ubidefeo, here is the process what I did.
1. I Install the new version `Arduino-CLI Version: 0.11.0 Commit: 0296f4d` provided by **rsora** in this thread.
2. I add this URL `http://dan.drown.org/stm32duino/package_STM32duino_index.json`
3. I install the STM32F103 board with this command `$ ./arduino-cli core install stm32duino:STM32F1`
4. I connect A9 and A10 of STM 32 with Raspberry pi's 15 and 14 pins respectively. 
5. change boot mode as boot0 at 1 and boot1 at 0.
6. press the reset button.
7. upload the code by `$ ./arduino-cli upload -p /dev/ttyS0 --fqbn stm32duino:STM32F1:genericSTM32F103C:upload_method=serialMethod /home/pi/MyFirstSketch`
8. I got this result. 

`panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x388a2c]

goroutine 1 [running]:
github.com/arduino/go-paths-helper.(*Path).String(...)
/github/home/go/pkg/mod/github.com/arduino/go-paths-helper@v1.2.0/paths.go:482
github.com/arduino/arduino-cli/arduino/cores.(*ToolRelease).RuntimeProperties(0x1bcde20, 0x4)
/__w/arduino-cli/arduino-cli/arduino/cores/tools.go:136 +0x7c
github.com/arduino/arduino-cli/commands/upload.runProgramAction(0x19d7380, 0x3135eb0, 0x0, 0x0, 0xbec5e678, 0x3f, 0xbec5e666, 0xa, 0x0, 0x0, ...)
/__w/arduino-cli/arduino-cli/commands/upload/upload.go:196 +0x850
github.com/arduino/arduino-cli/commands/upload.Upload(0x99cd48, 0x182a078, 0x34f3e30, 0x994be0, 0x18100e0, 0x994be0, 0x18100e8, 0x6, 0xa51031a5, 0x396c74)
/__w/arduino-cli/arduino-cli/commands/upload/upload.go:56 +0x2ac
github.com/arduino/arduino-cli/cli/upload.run(0x19e4c60, 0x19d0630, 0x1, 0x5)
/__w/arduino-cli/arduino-cli/cli/upload/upload.go:109 +0x1ec
github.com/spf13/cobra.(*Command).execute(0x19e4c60, 0x19d0600, 0x5, 0x6, 0x19e4c60, 0x19d0600)
/github/home/go/pkg/mod/github.com/spf13/cobra@v1.0.0/command.go:846 +0x1f4
github.com/spf13/cobra.(*Command).ExecuteC(0x1895a20, 0x0, 0x0, 0x18000e0)
/github/home/go/pkg/mod/github.com/spf13/cobra@v1.0.0/command.go:950 +0x26c
github.com/spf13/cobra.(*Command).Execute(...)
/github/home/go/pkg/mod/github.com/spf13/cobra@v1.0.0/command.go:887
main.main()
/__w/arduino-cli/arduino-cli/main.go:31 +0x24`
9. I tried with **usb to ttl serial cable** onUbuntu 20.04  and still not succeded.
`$ ./arduino-cli upload -p /dev/ttyUSB0 --fqbn `

`stm32duino:STM32F1:genericSTM32F103C /home/raj-priyanshu/MyFirstSketch/
dfu-util 0.7

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Filter on vendor = 0x1eaf product = 0x0003
No DFU capable USB device found
Waiting for /dev/ttyUSB0 serial...Done`

ubidefeo, 2020-07-01:
hey @raj-priyanshu 
thank you for giving it a shot.
bummer!

I have failed to hack my ST-Link into a bare board, but tomorrow I'm receiving a blue pill.

I'm concerned by this message 
`Filter on vendor = 0x1eaf product = 0x0003
No DFU capable USB device found
Waiting for /dev/ttyUSB0 serial...Done` because it means that it's still trying DFU upload.
I look forward to trying this out.

Keep you posted, and thank you for helping us squash some bugs 🙏🏼

raj-priyanshu, 2020-07-01:
Hi @ubidefeo... 
I forgot to mention... the last one (STEP 9) I tried on my computer and not on raspberry pi... Also, I was making some mistake in the uploading process. So, I found that and got a successful upload on Bluepill. :) 
Since the same version of Arduino-CLI is working on my computer (Linux x64) I think it is more like build issue for arm6.

ubidefeo, 2020-07-02:
That is an interesting turn of events, but before blaming the ARMv6 version I'd like to look at why it shouldn't work on the Pi.
Forgive me for asking, but even 15 years later I sometimes make this mistake myself:
Have you connected Blue Pill this way?

| Blue Pill   |      direction      |  RPi |
|----------|:-------------:|------:|
| RX |  < | TXD |
| TX |  > | RXD |
| GND | - |    GND |
| 3V3  | - | 3V3|

raj-priyanshu, 2020-07-02:
Hi @ubidefeo, 
Yes, apart from that I make a pcb for this, Where rpi communicating to stm32 over serial connection and both already able to communicate properly. I wanted some cli solutions so that I could easly program the arm over ssh connection and I found Arduino-cli could be helful. 
To be make sure it is not the problem with my pcb I already checked the same connection on breadboard first. And then tried to flash.so, I am very sure about connection side nothing is wrong. 
Ps.- also found that some basic features will be helpful with Arduino cli so once this issue is covered I will put some feature request.

ubidefeo, 2020-07-02:
All right, I'm happy you picked our CLI for your needs and I'll further investigate this with the team.
I can't wait for my Blue Pill to arrive and get cracking :)

Please feel free to propose features, we're at a phase of the project in which we can still shape a lot and you'll see more releases coming out as we add/remove/adjust workflows based on users' needs

Thank you for helping hunt this one down for now
u.

ubidefeo, 2020-07-03:
hi @raj-priyanshu 
just a quick update

I received my Blue Pill boards yesterday and this morning I uploaded the bootloader and managed to make it work on my Mac.
Once I transfer to the Pi Zero I get the same error as you do.

```
ino upload -p /dev/ttyAMA0 -b stm32duino:STM32F1:genericSTM32F103C6:upload_method=serialMethod -v
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x388a2c]

goroutine 1 [running]:
github.com/arduino/go-paths-helper.(*Path).String(...)
	/github/home/go/pkg/mod/github.com/arduino/go-paths-helper@v1.2.0/paths.go:482
github.com/arduino/arduino-cli/arduino/cores.(*ToolRelease).RuntimeProperties(0x300d920, 0x4)
	/__w/arduino-cli/arduino-cli/arduino/cores/tools.go:136 +0x7c
github.com/arduino/arduino-cli/commands/upload.runProgramAction(0x31db3c0, 0x48fe100, 0x0, 0x0, 0xbeca973e, 0x40, 0xbeca972e, 0xc, 0x0, 0x0, ...)
	/__w/arduino-cli/arduino-cli/commands/upload/upload.go:196 +0x850
github.com/arduino/arduino-cli/commands/upload.Upload(0x99cd48, 0x302a088, 0x4d7be30, 0x994be0, 0x30100e0, 0x994be0, 0x30100e8, 0x6, 0x26929c0a, 0x396c74)
	/__w/arduino-cli/arduino-cli/commands/upload/upload.go:56 +0x2ac
github.com/arduino/arduino-cli/cli/upload.run(0x31e4dc0, 0x31d0630, 0x0, 0x5)
	/__w/arduino-cli/arduino-cli/cli/upload/upload.go:109 +0x1ec
github.com/spf13/cobra.(*Command).execute(0x31e4dc0, 0x31d0600, 0x5, 0x6, 0x31e4dc0, 0x31d0600)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v1.0.0/command.go:846 +0x1f4
github.com/spf13/cobra.(*Command).ExecuteC(0x3095b80, 0x0, 0x0, 0x30000e0)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v1.0.0/command.go:950 +0x26c
github.com/spf13/cobra.(*Command).Execute(...)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v1.0.0/command.go:887
main.main()
	/__w/arduino-cli/arduino-cli/main.go:31 +0x24
```

Honestly I think that the Pi Zero is excruciatingly slow to work with, and I don't think anyone should go through the pain of a simple Blink sketch taking forever to compile and upload, but we'll investigate this error a bit further.
I'll later also test on a Pi 3 and 4 just for my peace of mind :)

@cmaglie , care to take a look at this error?
Feel free to transfer the issue from me to you 😉

raj-priyanshu, 2020-07-03:
Thanks, I think it is not easy and worth enough to solve now. But entire logic to work with rpi zw and Arduino cli was to easy to update the binary without using some extrnal bridge. Also I tried to get bin file from Gui Arduino but I found it only generate elf for stm32. May be another bug. It was the primary reason to do this. Also, I liked cli version. However. I will wait for the future replies. 
Also, I found if we can save fqbn as a profile it will be easier to work with. Another way to this is alias but then things are not that manegable. Also didn't find any parameters which is available in Arduino Gui for uploading. And one serial monitor will be nice. I am thinking to put this in feature request.

ubidefeo, 2020-07-03:
my pleasure, @raj-priyanshu 
please keep following and testing, your feedback is appreciated.

I can tell you that a Serial Monitor is in our plans, but for now I suggest you get acquainted with the `screen` command on Linux, which is what I sometimes use to connect to serial ports.
There are of course many Serial console apps around that can do what the Serial Monitor in the Java IDE does :)

You can definitely save the fqbn as a profile, but we're still working on the most convenient way to do this.
You can simply run the command `arduino-cli board attach` and see what the help suggests.
This will fill the file `sketch.json` with all that's necessary to compile for a target.
We're currently looking at how to implement multiple profiles which has been in my list since the beginning but we have a bit of UX work to do before we roll it out 😜 

Your are all great points, by the way, and I will be taking notes of this.
Also consider that once we have a better working auto-completion by pressing tab you should be able to see all your options and pick what you need. Well... that still require quite a lot of work.

Be patient, good things are coming :)

u.

raj-priyanshu, 2020-07-03:
Hi @ubidefeo,
I already using `screen` for this purpose. I am really looking forward to use Arduino cli with full features and shifting from Gui to cli. Because cli is love😄

ubidefeo, 2020-07-03:
>Because cli is love😄
I look forward to the rest of the team seeing this :D

raj-priyanshu, 2020-07-07:
Hi @ubidefeo,
Any new updates on this issue?

ubidefeo, 2020-07-07:
hi @raj-priyanshu 

We just finished reviewing this, and it seems that the upload tool is not available for `armv6` but during core installation you don't get an error for missing tools.
Basically you helped us to find 2 bugs which we are fixing right away, but unfortunately we won't be able to supply the tools missing in your setup.
I also started investigating another STM32duino core which seems to be more solid but doesn't have Raspberry Pi support yet.

I'm afraid you'll have to be more patient and we don't have an ETA for this, but thanks to you in the next release (nightly) you'll get better errors during core installation and compile/upload

closed 2020-07-07 by cmaglie

raj-priyanshu, 2020-07-07:
Hi @ubidefeo,
Ok, I will wait. Btw please provide a better way to build Arduino-cli for arm6. Docker is also through a error on rpi zero. And I don't know build using go tool chain. And this part is not covered in documentation.

cmaglie, 2020-07-07:
The two fixes are now merged (#794 and #795), so the cli should not crash anymore.

it should be available as "nightly build" here https://arduino.github.io/arduino-cli/installation/#download-a-nightly-build within 24 hours (the nighlty build job runs every night at 1 AM).

You can also cross-build the CLI from any other PC by running the following command on the root of the git repository:

```
GOOS=linux GOARCH=arm go build
```

it will produce an `arduino-cli` executable for arm32 (even if the PC is not arm nor linux), you just need a recent golang installed.

ubidefeo, 2020-07-07:
@raj-priyanshu 
I advice you to switch to a more recent architecture.
`armv6` is extremely slow and underperforming for this kind of activities. You'd already be better set with a Pi 3.
Consider that the world is heading towards a full ARM 64 bit conversion (Raspian OS 64 bit will soon be the standard release) and slowly nobody will bother building tools for 32 bit targets.
For instance we cannot deploy some things on Pi Zero because 3rd party components simply won't build or run on `armv6` and we have to accept that.
Some things are just out of our control.

For documentation on building for `armv6` I don't know much about it, but I think @per1234 might chime in on this.

I'd like to close this issue not too late, since it's not something we can address and needs to be brought forward to the developers/maintainers of the core you're trying to use.

u.

raj-priyanshu, 2020-07-07:
@ubidefeo ok, and thanks for advice.😀 I will try to find some other suitable architecture.😀

ubidefeo, 2020-07-07:
Thank you, it's been nice bug-hunting with you :)