arduino/arduino-cli #759: arduino-cli board details crashes with external FQBNs

a7hybnj2, 2020-06-24:
## Bug Report

### Current behavior

```
arduino-cli board details ATTinyCore:avr:attinyx4
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x46b324f]

goroutine 1 [running]:
github.com/arduino/arduino-cli/commands/board.Details(0x4b080a0, 0xc000126008, 0xc000547d60, 0x46bebc7, 0xc000190200, 0xc000589d70)
        /private/tmp/arduino-cli-20200624-55394-jn0e1u/commands/board/details.go:65 +0x4df
github.com/arduino/arduino-cli/cli/board.runDetailsCommand(0xc000168b00, 0xc000119ba0, 0x1, 0x1)
        /private/tmp/arduino-cli-20200624-55394-jn0e1u/cli/board/details.go:57 +0x145
github.com/spf13/cobra.(*Command).execute(0xc000168b00, 0xc000119b80, 0x1, 0x1, 0xc000168b00, 0xc000119b80)
        /Users/a7hybnj2/Library/Caches/Homebrew/go_mod_cache/pkg/mod/github.com/spf13/cobra@v1.0.0/command.go:846 +0x29d
github.com/spf13/cobra.(*Command).ExecuteC(0xc000168000, 0x0, 0x0, 0xc00010e058)
        /Users/a7hybnj2/Library/Caches/Homebrew/go_mod_cache/pkg/mod/github.com/spf13/cobra@v1.0.0/command.go:950 +0x349
github.com/spf13/cobra.(*Command).Execute(...)
        /Users/a7hybnj2/Library/Caches/Homebrew/go_mod_cache/pkg/mod/github.com/spf13/cobra@v1.0.0/command.go:887
main.main()
        /private/tmp/arduino-cli-20200624-55394-jn0e1u/main.go:31 +0x36
```

### Expected behavior
I would like for the command to show me all the available options for the board I am specifying.

### Environment
- CLI version (output of `arduino-cli version`): `arduino-cli Version: HEAD-7541c80 Commit: 7541c80d180106f4370731de45a3ff4cc6709657`
- OS and platform: `macOS 10.15.5`

### Additional Info
```
arduino-cli board listall
Board Name                    FQBN                         
ATtiny1634 (No bootloader)    ATTinyCore:avr:attiny1634    
ATtiny1634 (Optiboot)         ATTinyCore:avr:attiny1634opti
ATtiny167/87 (Optiboot)       ATTinyCore:avr:attinyx7opti  
ATtiny2313/4313               ATTinyCore:avr:attinyx313    
ATtiny24/44/84                ATTinyCore:avr:attinyx4      
ATtiny25/45/85                ATTinyCore:avr:attinyx5      
ATtiny261/461/861             ATTinyCore:avr:attinyx61     
ATtiny43                      ATTinyCore:avr:attiny43      
ATtiny44/84 (optiboot)        ATTinyCore:avr:attinyx4opti  
ATtiny441/841 (No bootloader) ATTinyCore:avr:attinyx41     
ATtiny441/841 (Optiboot)      ATTinyCore:avr:attinyx41opti 
ATtiny45/85 (Optiboot)        ATTinyCore:avr:attinyx5opti  
ATtiny461/861 (optiboot)      ATTinyCore:avr:attinyx61opti 
ATtiny48/88                   ATTinyCore:avr:attinyx8      
ATtiny48/88 (optiboot)        ATTinyCore:avr:attinyx8opti  
ATtiny828 (No bootloader)     ATTinyCore:avr:attiny828     
ATtiny828 (Optiboot)          ATTinyCore:avr:attiny828opti 
ATtiny87/167 (No bootloader)  ATTinyCore:avr:attinyx7      
Arduino Nano Every            arduino:megaavr:nona4809     
Arduino Uno WiFi Rev2         arduino:megaavr:uno2018      
```

fpistm, 2020-07-01:
Hi
I've got the same issue with our CI and the [STM32](https://github.com/stm32duino/Arduino_Core_STM32) core. It works fine with 0.10.0.
I've updated our python build scripts to manage the breaking changes (`--output`) anyway we could not get the boards details.
One strange things is that I could not reproduce locally (under Windows 10 or Linux Ubuntu 18.04), only with the GitHub action running under Linux Ubuntu 18.04.
See: https://github.com/stm32duino/Arduino_Core_STM32/pull/1107/checks?check_run_id=827270930

Extract of the log:

> panic: runtime error: invalid memory address or nil pointer dereference
> [signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0xac705f]
> 
> goroutine 1 [running]:
> github.com/arduino/arduino-cli/commands/board.Details(0xf33100, 0xc000098010, 0xc0000f1d38, 0xad2ab7, 0xc000104a00, 0xc0001cdd48)
> 	/__w/arduino-cli/arduino-cli/commands/board/details.go:65 +0x4df
> github.com/arduino/arduino-cli/cli/board.runDetailsCommand(0xc000191080, 0xc00026ebd0, 0x1, 0x3)
> 	/__w/arduino-cli/arduino-cli/cli/board/details.go:57 +0x145
> github.com/spf13/cobra.(*Command).execute(0xc000191080, 0xc00026eba0, 0x3, 0x3, 0xc000191080, 0xc00026eba0)
> 	/github/home/go/pkg/mod/github.com/spf13/cobra@v1.0.0/command.go:846 +0x2aa
> github.com/spf13/cobra.(*Command).ExecuteC(0xc000190840, 0x0, 0xc6f6e0, 0xc000082058)
> 	/github/home/go/pkg/mod/github.com/spf13/cobra@v1.0.0/command.go:950 +0x349
> github.com/spf13/cobra.(*Command).Execute(...)
> 	/github/home/go/pkg/mod/github.com/spf13/cobra@v1.0.0/command.go:887
> main.main()
> 	/__w/arduino-cli/arduino-cli/main.go:31 +0x36
> panic: runtime error: invalid memory address or nil pointer dereference

Here the full Raw log:
[FullRawLogOfTheCI.txt](https://github.com/arduino/arduino-cli/files/4858997/FullRawLogOfTheCI.txt)

matthijskooijman, 2020-08-26:
I also ran into this issue, getting exactly the same backtrace with the Adafruit SAMD platform, installed through the board manager and running `arduino-cli board details adafruit:samd:adafruit_feather_m0` with current git master (locally compiled).
 
The problem seems to be that `boardPlatform.Resource` is `nil` here: https://github.com/arduino/arduino-cli/blob/48383dad5e53f155beff1178742ed5e2ea810c5f/commands/board/details.go#L65

I looked around a little bit for how this value is set, but couldn't quickly find my way, so I'm just leaving this here instead.

fpistm, 2020-08-26:
Hi @matthijskooijman 
Which version of the cli you used ?

matthijskooijman, 2020-08-26:
Current git master, just pulled and built it.

fpistm, 2020-08-26:
Ok thanks. Sorry mireading your first comment.

I've tested with the latest release 0.12 and same issue:
For ref: https://github.com/stm32duino/Arduino_Core_STM32/pull/1154/checks?check_run_id=1030660344
```
Arduino CLI version used: 0.12.0
Cores configuration JSON file that will be used:
/github/home/.arduino15/packages/STM32/hardware/stm32/2.0.0-dev/CI/build/conf/cores_config_ci.json
Build configuration for 'STM32' maintainer and 'stm32' architecture
"arduino-cli board details --format json STM32:stm32:Nucleo_64" failed with code: 2!
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0xbae548]

goroutine 1 [running]:
github.com/arduino/arduino-cli/commands/board.Details(0x1058da0, 0xc000038048, 0xc00013bd30, 0xa, 0xa, 0x9)
	/__w/arduino-cli/arduino-cli/commands/board/details.go:65 +0x4e8
github.com/arduino/arduino-cli/cli/board.runDetailsCommand(0xc0001f3600, 0xc0002b5aa0, 0x1, 0x3)
	/__w/arduino-cli/arduino-cli/cli/board/details.go:67 +0x11e
github.com/spf13/cobra.(*Command).execute(0xc0001f3600, 0xc0002b5a70, 0x3, 0x3, 0xc0001f3600, 0xc0002b5a70)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v1.0.1-0.20200710201246-675ae5f5a98c/command.go:846 +0x2aa
github.com/spf13/cobra.(*Command).ExecuteC(0xc0001f2dc0, 0x0, 0xd4bb80, 0xc0000440b8)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v1.0.1-0.20200710201246-675ae5f5a98c/command.go:950 +0x349
github.com/spf13/cobra.(*Command).Execute(...)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v1.0.1-0.20200710201246-675ae5f5a98c/command.go:887
main.main()
	/__w/arduino-cli/arduino-cli/main.go:31 +0x36

##[error]Process completed with exit code 1.
```

cmaglie, 2020-08-26:
Looking at stacktrace, the `Resource.URL` value comes from the various package_index jsons.
Recently, in the `board details` command, we added a print of the platform resources URLs. So the first though I had is that platforms manually installed in the `sketchbook/hardware` directory may be affected... and after testing it, indeed they are, so I will produce a fix for this problem.

Anyway I'm wondering how a package installed from a package_index (even from a 3rd party index) may end up in this situation.
I'm also unable to reproduce the issue with any of the example posted above, I tried with:

`ATTinyCore:avr:attinyx4` (using http://drazzy.com/package_drazzy.com_index.json)
`adafruit:samd:adafruit_feather_m0` (using https://adafruit.github.io/arduino-board-index/package_adafruit_index.json)
`STM32:stm32:Nucleo_64` (using https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json)

and I always get the correct output without any issue, the procedure I followed is:

1. add the 3rd party url in the `arduino-cli` configuration
2. install the platform
3. run `board details` with the board FQBN

Hopefully the patch will fix the issue in general but, in any case, if someone may provide a step-by-step procedure to reproduce the original issue, starting from a clean install, it may help to see if there are other bugs to fix.

matthijskooijman, 2020-08-26:
> So the first though I had is that platforms manually installed in the sketchbook/hardware directory may be affected... 

Yeah, that was also my first thought, but it also happened for other platforms.

> add the 3rd party url in the arduino-cli configuration

Ah, that's probably the problem. I installed this platform using the Java IDE, so I have no URLs configured for it in arduino-cli.yaml.

fpistm, 2020-08-26:
In fact,
Several calls uses the `--additional-urls` 
- core update-index
- core install
-  core search

Then it fails on the `board details` without `--additional-urls`
I can try to add the `--additional-urls` to this call.

fpistm, 2020-08-26:
I've tested to add the `--additional-urls` to the `board details` and it seems OK.
The question is why before it works without and now not, I guess? 
https://github.com/stm32duino/Arduino_Core_STM32/pull/1154/commits/2e5f4401895ca5cd850f5b3ea05c8d9b92b8246a

cmaglie, 2020-08-27:
> The question is why before it works without and now not, I guess?

The reason is that we changed the `board details` command: now it prints more information (like the package URL) but it doesn't check if the information is available or not, so the nil excpetion.

This is totally our oversight, BTW this is now fixed after merging #924 and will be available in the next release.
Thanks for the bug report!

closed 2020-08-27 by cmaglie