arduino/arduino-cli #764: arduino-cli upload `-i` flag removed

janjongboom, 2020-06-25:
## Bug Report

### Current behavior

We currently upload via e.g.:

```
$ arduino-cli upload -p /dev/cu.usbmodem401101 --fqbn arduino:mbed:nano33ble -i path-to-our.bin
```

This no longer works in v0.11. This is a breaking change for us and I do not see any alternatives for specifying the bin file in the latest docs. Is there any way to achieve the same behavior? We use the Arduino CLI to flash prebuilt binaries, not sketches.

### Environment

- CLI version (output of `arduino-cli version`): arduino-cli Version: 0.11.0 Commit: 0296f4d
- OS and platform: macOS Catalina

### Additional context

rsora, 2020-06-26:
Hi @janjongboom, thanks for the heads up!
In 0.11.0 with https://github.com/arduino/arduino-cli/pull/687, we changed the way the CLI manages the binary artifacts.

janjongboom, 2020-06-26:
@rsora `--input-dir` does not work either, as it requires the `.bin` file to be named `foldername.ino.bin`. Users download this as a ZIP file and we have no control what the name of the folder where they extract this binary is.

cmaglie, 2020-06-26:
@janjongboom I see that your usecase is a good one and is now lacking support, we'll reintroduce the `--input-file` flag to perform "upload without the sketch". We're are going to make another release soon, I'll try to strech a fix for this one.

For completeness, the current `compile/upload` flow is meant to be something like:

```
# sketch directory
$ find ~/Arduino/Blink/
/home/cmaglie/Arduino/Blink/
/home/cmaglie/Arduino/Blink/Blink.ino

# compile using the default build path (inside the sketch)
$ arduino-cli compile ~/Arduino/Blink -b arduino:samd:mkr1000
Sketch uses 11236 bytes (4%) of program storage space. Maximum is 262144 bytes.
Global variables use 2384 bytes of dynamic memory.

$ find ~/Arduino/Blink/
/home/cmaglie/Arduino/Blink/
/home/cmaglie/Arduino/Blink/build
/home/cmaglie/Arduino/Blink/build/arduino.samd.mkr1000
/home/cmaglie/Arduino/Blink/build/arduino.samd.mkr1000/Blink.ino.map
/home/cmaglie/Arduino/Blink/build/arduino.samd.mkr1000/Blink.ino.hex
/home/cmaglie/Arduino/Blink/build/arduino.samd.mkr1000/Blink.ino.elf
/home/cmaglie/Arduino/Blink/build/arduino.samd.mkr1000/Blink.ino.bin
/home/cmaglie/Arduino/Blink/build/arduino.samd.mkr1000/Blink.ino.with_bootloader.hex
/home/cmaglie/Arduino/Blink/Blink.ino

# compile with custom build path (--output-dir)
$ arduino-cli compile ~/Arduino/Blink -b arduino:samd:mkr1000 --output-dir mybuild
Sketch uses 11236 bytes (4%) of program storage space. Maximum is 262144 bytes.
Global variables use 2384 bytes of dynamic memory.

$ find mybuild
mybuild
mybuild/Blink.ino.map
mybuild/Blink.ino.with_bootloader.bin
mybuild/Blink.ino.hex
mybuild/Blink.ino.elf
mybuild/Blink.ino.bin
mybuild/Blink.ino.with_bootloader.hex

# upload from custom build path (--input-dir)
$ arduino-cli upload ~/Arduino/Blink -b arduino:samd:mkr1000 --input-dir mybuild -p /dev/ttyACM0
Waiting for upload port...
No new serial port detected.
No device found on ttyACM0
```

so we replaced the pair `--output-file / --input-file` with `--output-dir / --input-dir` because some platforms produces more than one artifact out of the build that is required for upload (for example a .bin to upload via USB-DFU and a .zip to push via bootloader...), so we had to replace the "file" with a "dir".

cmaglie, 2020-09-14:
@janjongboom
we reintroduced the flag in 0.13.0, would you like to give it a try?

closed 2020-09-25 by janjongboom