arduino/arduino-cli #769: Issue with Keyboard.h library

dbhaig, 2020-06-28:
## Bug Report

### Current behavior

#include <Keyboard.h>
void setup() {
}
void loop() {
}

arduino-cli compile
/home/don/test/test.ino:1:10: fatal error: Keyboard.h: No such file or directory
 #include <Keyboard.h>
          ^~~~~~~~~~~~

Tried to install library:

arduino-cli lib install Keyboard
Keyboard depends on Keyboard@1.0.2
Downloading Keyboard@1.0.2...
Keyboard@1.0.2 downloaded                                                                                                                             
Installing Keyboard@1.0.2...
Error installing name:"Keyboard" versionRequired:"1.0.2" : extracting archive: Read the zip file: zip: not a valid zip file

### Expected behavior
Sketch compiles using Arduino IDE

### Environment

- CLI version (output of `arduino-cli version`):

arduino-cli Version: 0.11.0 Commit: 0296f4d

- OS and platform:

Linux 4.4.0-184-generic #214-Ubuntu SMP Thu Jun 4 10:14:11 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

### Additional context

Sketch compiles using the Arduino IDE

umbynos, 2020-06-29:
Hello, could you provide wich core are you using? 
I've tried with `arduino-cli compile --fqbn arduino:samd:mkrwifi1010 .` and the compile process was ok. (Obviously you have to install the required core before)

dbhaig, 2020-06-29:
more sketch.json 
{
  "cpu": {
    "fqbn": "arduino:avr:leonardo",
    "name": "Arduino Leonardo",
    "port": "serial:///dev/ttyACM0"
  }
}

arduino-cli core list
ID                 Installed Latest    Name                                        
STM32:stm32        1.9.0     1.9.0                                                 
SparkFun:avr       1.1.12    1.1.12                                                
adafruit:samd      1.5.14    1.5.14                                                
arduino:avr        1.8.3     1.8.3     Arduino AVR Boards                          
arduino:samd       1.8.6     1.8.6     Arduino SAMD Boards (32-bits ARM Cortex-M0+)
esp32:esp32        1.0.4     1.0.4                                                 
esp8266:esp8266    2.7.1     2.7.1     esp8266                                     
espressif:esp32                                                                    
stm32duino:STM32F1 2020.5.26 2020.5.26

umbynos, 2020-06-30:
Actually I tried again with avr core and compile for the arduino leonardo and I can't reproduce your problem..
`arduino-cli compile --fqbn arduino:avr:leonardo test-cli-keyboard/test-cli-keyboard.ino` shows the expected result for me.
Also installing the Keyboard library works for me: `arduino-cli lib install Keyboard` shows:
```
Keyboard depends on Keyboard@1.0.2
Downloading Keyboard@1.0.2...
Keyboard@1.0.2 downloaded                                                                                                                    
Installing Keyboard@1.0.2...
Installed Keyboard@1.0.2
```
What installation procedure did you follow? DId you made some customization regarding libraries path?

dbhaig, 2020-06-30:
I tried again today and everything worked:
 
arduino-cli lib install Keyboard
Keyboard depends on Keyboard@1.0.2
Downloading Keyboard@1.0.2...
Keyboard@1.0.2 downloaded                                                       
Installing Keyboard@1.0.2...
Installed Keyboard@1.0.2

arduino-cli compile
Sketch uses 5102 bytes (17%) of program storage space. Maximum is 28672 bytes.
Global variables use 224 bytes (8%) of dynamic memory, leaving 2336 bytes for local variables. Maximum is 2560 bytes.

I'm completely baffled as to why it is now working but I'm glad that it is.

Thank you for taking the time to look into this.

closed 2020-06-30 by dbhaig