arduino/arduino-cli #771: Portenta H7 (arduino-beta:mbed) - Hyphens in FQBN not supported?

Dygear, 2020-06-28:
## Bug Report
Trying to upload a sketch to the Portenta H7, but I get an invalid FQBN error. According to my searching around the internet, this seems to be a known issue. [pert (Global Moderator - Arduino Forms)](https://forum.arduino.cc/index.php?topic=691052.0#) has mentioned that this is a known issue on the Pro IDE with a link to this issue there. https://github.com/arduino/arduino-pro-ide/issues/272 It seems that this is also a problem within the CLI interface as well.

### Current behavior
`arduino-cli core install arduino-beta:mbed`
`arduino-cli core list`
```
ID                Installed Latest Name                       
arduino-beta:mbed 1.2.0     1.2.0  Arduino mbed-enabled Boards
```
`arduino-cli compile -b arduino-beta:mbed`
```
Error during build: incorrect FQBN: invalid fqbn: arduino-beta:mbed
```

Arduino-CLI does not appear to support FQBN that have a hyphens in it's name.

### Expected behavior
For the compile process to succeed.

### Environment
- CLI version (output of `arduino-cli version`):
`arduino-cli Version: 0.11.0 Commit: 0296f4d`
- OS and platform:
`Linux Dygear-TR4 5.4.0-39-generic #43-Ubuntu SMP Fri Jun 19 10:28:31 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`

### Additional context
Does not work in IDE Pro either.

rsora, 2020-06-29:
Hi @Dygear Thanks for your heads up!

the `-b` argument for the `compile` commands requires a _board FQBN_, which means a Fully Qualified Board Name.

the `arduino-beta:mbed` core supports 3 different boards (as you can see running `arduino-cli board listall`):

```
...
Arduino Nano 33 BLE              arduino-beta:mbed:nano33ble   
Arduino Portenta H7 (M4 core)    arduino-beta:mbed:envie_m4    
Arduino Portenta H7 (M7 core)    arduino-beta:mbed:envie_m7 
...
```

So, when you try to compile a sketch for the Portenta you should specify, for example, 
```
arduino-cli compile -b arduino-beta:mbed:envie_m4
```

or 

```
arduino-cli compile -b arduino-beta:mbed:envie_m7
```

Please let us know if it works your side :+1:

Dygear, 2020-06-29:
Hey that works. Sorry for my confusion!

closed 2020-06-29 by Dygear