arduino/arduino-cli #775: Unknown Command

Ashteroide, 2020-07-01:
## Bug Report

### Current behavior
[Starting] Verify sketch - ArduPong.ino
[Warning] Output path is not specified. Unable to reuse previously compiled files. Verify could be slow. See README.
Error: unknown command "arduboy-homemade:avr:arduboy-homemade:based_on=promicro,boot=cathy3k,core=arduino-core,display=sh1106,flashselect=rx" for "arduino-cli"
Run 'arduino-cli --help' for usage.
[Error] Exit with code=1

I have an Arduboy which is a third part library and get the above error

### Expected behavior

Verifies my sketch file

### Environment

arduino-cli Version: 0.11.0 Commit: 0296f4df116385f868b67c5ffa7393936c3345c9
MacOS Catalina 10.15.5

### Additional context

I had an issue with Arduino IDE opening and making it the focus while I'm working in VSCode

alranel, 2020-07-29:
It looks like you're using https://github.com/microsoft/vscode-arduino rather than arduino-cli directly. Can you confirm this?

ubidefeo, 2020-07-29:
@Ashteroide sorry for the late reply, this fell off my radar.
Thank you @alranel for picking it up 🙏 

Please detail your setup and attempted workflow.
We have recently noticed that some board configurations in `arduboy-homemade` core had malformed settings which can cause issues during compile.
If I have more details I can try and reproduce your issue  :)

Ashteroide, 2020-07-29:
Hi, thank you both for replying. I have the Arduino extension for VSCode but setup the `Arduino: Command Path` set to arduino-cli. I suspect you mean uninstalling the VSCode Arduino extension, then setting the arduino-cli up. And what are some extra details I could provide?

My settings for the Arduboy Homemade are
```
{
    "programmer": "AVRISP mkII",
    "board": "arduboy-homemade:avr:arduboy-homemade",
    "port": "/dev/tty.usbmodem14101",
    "sketch": "ArduPong.ino",
    "configuration": "based_on=promicro,boot=cathy3k,core=arduino-core,display=sh1106,flashselect=rx"
}
```

Ashteroide, 2020-07-30:
I've gathered a bit more information about the problem, I ran a few suggested lines and got these results
```
(base) home Blink % arduino-cli board list
Port                            Type              Board Name  FQBN            Core       
/dev/cu.Bluetooth-Incoming-Port Serial Port       Unknown                                
/dev/cu.usbmodem14201           Serial Port (USB) Arduino Uno arduino:avr:uno arduino:avr

(base) home Blink % arduino-cli upload -p /dev/tty.usbmodem14201 --fqbn arduino:avr:uno Blink
avrdude: can't open input file build/arduino.avr.uno/..ino.hex: No such file or directory
avrdude: read from file 'build/arduino.avr.uno/..ino.hex' failed
Error during Upload: uploading error: uploading error: exit status 1
```

I think it's more to do that the documentation I read doesn't fully explain everything you have to do from installing any VSCode extensions

ubidefeo, 2020-07-30:
hey @Ashteroide 
I'm not sure that's something that should work as a drop-in replacement for the Arduino IDE in VS Code.
Sparkfun has written a nice article on setting up VS Code + Arduino Plugin (mostly for code sensing etc) and Arduino CLI.
I haven't had time to set it up and try, but I'm sure it can give you a good pointer :)
https://learn.sparkfun.com/tutorials/efficient-arduino-programming-with-arduino-cli-and-visual-studio-code/all

Essentially you need to initially install some cores, and listing a board that way won't work.
Please take a look at the [Getting Started page](https://arduino.github.io/arduino-cli/getting-started/)

Once a core is installed, you can find a board by 
`arduin-cli board listall | grep -i mkr` (you need the `-i` to make `grep` case insensitive

Ashteroide, 2020-07-30:
@ubidefeo I used your suggested command `arduin-cli board listall | grep -i mkr` and got this result
```
(base) HOME Blink % arduino-cli board listall | grep -i mkr
Arduino MKR FOX 1200                arduino:samd:mkrfox1200                   
Arduino MKR GSM 1400                arduino:samd:mkrgsm1400                   
Arduino MKR NB 1500                 arduino:samd:mkrnb1500                    
Arduino MKR Vidor 4000              arduino:samd:mkrvidor4000                 
Arduino MKR WAN 1300                arduino:samd:mkrwan1300                   
Arduino MKR WAN 1310                arduino:samd:mkrwan1310                   
Arduino MKR WiFi 1010               arduino:samd:mkrwifi1010                  
Arduino MKR1000                     arduino:samd:mkr1000                      
Arduino MKRZERO                     arduino:samd:mkrzero  
```
I also found a weird bug when updating AVR Boards to 1.8.3 through Arduino IDE the folder would disappear and then VSCode would report a missing file, after some more trying I still get this
```
(base) HOME Blink % arduino-cli upload -p /dev/tty.usbmodem14201 --fqbn arduino:avr:uno Blink
avrdude: can't open input file build/arduino.avr.uno/..ino.hex: No such file or directory
avrdude: read from file 'build/arduino.avr.uno/..ino.hex' failed
Error during Upload: uploading error: uploading error: exit status 1
```

ubidefeo, 2020-07-30:
@Ashteroide 
I want to stress that the configuration you have now is beyond experimental and not endorsed by us at all :D
We do not develop the VS Code integration plugin, so have no idea how it integrates with Arduino IDE or the CLI.

Please update to Arduino IDE 1.8.13 because it introduces a fix that prevents index issues between the CLI and the IDE :)

Ashteroide, 2020-07-30:
@ubidefeo Thanks for the update info on Arduino IDE I’ll download it when I get time today (Most likely in about 20mins), if you’re able can you run a mini setup guide to help me, so the Arduino plugin for VSCode isn’t needed when using Arduino CLI. After some thinking I think I understand the problem a bit more, I’ve been using the Arduino plugin to try and run Arduino CLI but I need to upload my code through the terminal, the issue I’m having is the uno not showing, I’ll dig as deep as I can but unfortunately I can’t test it this works until this afternoon because it’s at my house

ubidefeo, 2020-07-31:
sorry, @Ashteroide 

>if you’re able can you run a mini setup guide to help me, so the Arduino plugin for VSCode isn’t needed when using Arduino CLI. 

I don't think I understand this :)

You got the rest correctly, the VS Code plugin is built for the Java IDE, and it wouldn't even work with the Alpha IDE we're working on right now.

Follow the Sparkfun link I gave you above, it's a good tutorial to use the Arduino CLI with VS Code. A very detailed one ✌️

Ashteroide, 2020-07-31:
@ubidefeo I followed the Sparkfun link, but I've made some recent progress. I managed to get 1 sketch on my Uno, I'll list the commands I used, just a note I opened the whole folder with all my other projects their too, I think this may contribute something
`arduino-cli sketch new SketchTest`
`arduino-cli compile SketchTest` (I think their was more to this but you should get the idea, I am able to see the build folder with the SketchTest.ino.hex)
`arduino-cli upload -p /dev/tty.usbmodem14201 --fqbn arduino:avr:uno Sketchtest`
This only worked once and then I couldn't upload anymore, here is what I uploaded
```
void setup()
{
    pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
    digitalWrite(LED_BUILTIN, HIGH);
    delay(4000);
    digitalWrite(LED_BUILTIN, LOW);
    delay(4000);
}
```
and as suspected the builtin LED blinks in a 4 sec interval
I'll do some more testing to see what I can find

Ashteroide, 2020-07-31:
@ubidefeo I am fully able to upload to my Uno easily just by compiling and the uploading, but my Arduboy doesn't currently work because I don't know how to specify the bootloader, display or core within arduino-cli, Here is some feedback from arduino-cli when I try to upload to my Arduboy (Homemade by the way)
```
(base) ajones23@Ashlees-Macbook-Air My Projects % arduino-cli compile -b arduboy-homemade:avr:arduboy ArduPong
Sketch uses 10770 bytes (37%) of program storage space. Maximum is 28672 bytes.
Global variables use 1244 bytes (48%) of dynamic memory, leaving 1316 bytes for local variables. Maximum is 2560 bytes.
(base) ajones23@Ashlees-Macbook-Air My Projects % arduino-cli upload -p /dev/tty.usbmodem14201 --fqbn arduboy-homemade:avr:arduboy ArduPong
Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44
```

ubidefeo, 2020-07-31:
hey @Ashteroide 

if you use the Java IDE and enable full verbose in the compile/upload you'll see that additional parameters are added to the `--fqbn`

you'll have things like 
`--fqbn=arduboy-homemade:avr:arduboy-homemade:based_on=leonardo,boot=org,core=arduino-core,display=sh1106,flashselect=sda`

this is how you chain parameters :)

Ashteroide, 2020-07-31:
@ubidefeo Thank you very much for your help that you provided, I have discovered my Homemade Arduboy is a bit broken but I had a different to confirm, yes I tried the `--fqbn=arduboy-homemade:avr:arduboy-homemade:based_on=leonardo,boot=org,core=arduino-core,display=sh1106,flashselect=sda` and that works which is good. I did try in Arduino IDE too which still didn't work on my own, so it's a physical issue that I'll have to fix. Would you know of any easier ways to compile and upload to Arduino's with the CLI?

ubidefeo, 2020-07-31:
@Ashteroide 
can you try to download annolder version of the Java IDE like 1.8.10?
The Arduboy core has some malformed configuration parameters that can cause the IDE 1.8.13 to fail to build menu items.

also downloading a nightly build of the IDE should work

Ashteroide, 2020-08-01:
@ubidefeo What do you mean by Java IDE? I'll check the nightly builds of Arduino IDE too

per1234, 2020-08-01:
@Ashteroide "Java IDE" == "Arduino IDE".

The Arduino IDE is written in Java, as opposed to the Arduino Pro IDE, which is based on the TypeScript Eclipse Theia IDE framework.

Ashteroide, 2020-08-05:
@per1234 Thanks for the info. @ubidefeo apologies for the late reply, I tried 1.8.10 which I know previously worked with my ArduBoy, but still no success for it working so it is most likely a physical issue, considering I've had this problem before and the previous fix isn't working

per1234, 2021-02-22:
This mostly turned out to be simply a matter of misusing Arduino CLI. However, there was one bug:
>avrdude: read from file 'build/arduino.avr.uno/..ino.hex' failed

We had a dedicated issue report for this: https://github.com/arduino/arduino-cli/issues/691, but it evolved over time from a cosmetic issue to a true functionality defect. However, that has since been fixed, so I'll close this as resolved.

closed 2021-02-22 by per1234