arduino/arduino-cli #792: Error installing: There were errors loading platform indexes

jonathangjertsen, 2020-07-06:
## Bug Report

### Current behavior
I have a Travis CI setup which runs a build script with the following commands

```
curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh
./bin/arduino-cli core install arduino:avr
```

The first command succeeds with the following output:

```
Installing in /home/travis/build/jonathangjertsen/dht11/build/bin
ARCH=64bit
OS=Linux
Using curl as download tool
TAG=0.11.0
CLI_DIST=arduino-cli_0.11.0_Linux_64bit.tar.gz
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_0.11.0_Linux_64bit.tar.gz
arduino-cli not found. You might want to add /home/travis/build/jonathangjertsen/dht11/build/bin to your $PATH
```

The second command fails with the following output:

```
Downloading missing tool builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 downloaded                                          
Installing builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 installed
Downloading missing tool builtin:serial-discovery@1.0.0...
builtin:serial-discovery@1.0.0 downloaded                                       
Installing builtin:serial-discovery@1.0.0...
builtin:serial-discovery@1.0.0 installed
Updating index: library_index.json downloaded                                   
Error installing: There were errors loading platform indexes
```

### Expected behavior
I would expect the installation to succeed, or at least to provide an actionable error message

### Environment

- CLI version (output of `arduino-cli version`): `arduino-cli Version: 0.11.0 Commit: 0296f4d`
- OS and platform: Ubuntu 16.04.6 LTS, linux/amd64 (Docker)

per1234, 2020-07-07:
Hi @jonathangjertsen. You forgot to download the platform indexes by running the command:
```
./bin/arduino-cli core update-index
```
There is a nice tutorial about using Arduino CLI here:
https://arduino.github.io/arduino-cli/latest/getting-started

jonathangjertsen, 2020-07-07:
How silly of me, thanks!

I would suggest updating the error message to include more information about the error. The code that emits this error message does not seem to print the previous error message(s) that caused the error https://github.com/arduino/arduino-cli/blob/6903076cd3e0ba05791dbd248907a7d42c7f9e16/cli/instance/instance.go#L94

per1234, 2020-07-07:
>thanks!

You're welcome. I'm glad if was able to assist you in using Arduino CLI for your CI needs. It's certainly the right tool for the job!

>I would suggest updating the error message to include more information about the error.

I updated the title and left the issue open to allow the Arduino CLI team to consider your suggestion.

Inc0ns1st3nt, 2020-07-09:
Hi,

I was using arduino-cli on my mkr-wifi-1010, after successfully compile the sketch, I had this error during the upload it to the board.
I had run the update commands several times
```
arduino-cli core update-index
```
The error message:
```
Error during Upload: There were errors loading platform indexes
```

namlee2, 2020-08-12:
> Hi,
> 
> I was using arduino-cli on my mkr-wifi-1010, after successfully compile the sketch, I had this error during the upload it to the board.
> I had run the update commands several times
> 
> ```
> arduino-cli core update-index
> ```
> 
> The command I ran:
> 
> ```
> sudo arduino-cli upload --fqbn arduino:samd:mkrwifi1010 -p /dev/ttyACM0 test
> ```
> 
> The full error message:
> 
> ```
> Error during Upload: There were errors loading platform indexes
> ```
> 
> EDIT: I think my issue is because I ran the command as sudo, and without it, my issue is about permission, which is unrelated as this issue

Hi!

I had the same issue as well, and thank god for forums (https://forum.arduino.cc/index.php?topic=495039.0)!
As you said, it's a permission issue, so here's the magic command:
`sudo chmod a+rw /dev/ttyACM0`
Once you did this, you don't even have to sudo your upload command.

Have a nice day!

umbynos, 2022-10-20:
Hi @jonathangjertsen, this has changed a lot, but I think that now the error should be more clear: https://github.com/arduino/arduino-cli/blob/bdd0cf9267926f81b91ddbde30189dac6eeb1829/cli/instance/instance.go#L187
Furthermore, the index is downloaded when required #908, feel free to reopen this if needed