arduino/arduino-cli #796: arduino-cli 11 on arm6 with sense33ble board (compile) Illegal Instruction

DrMomentum, 2020-07-07:
## Bug Report

### Current behavior:

_arduino-cli compile -b arduino:mbed:nano33ble blinkYou/ -v_

Using board 'nano33ble' from platform in folder: /root/.arduino15/packages/arduino/hardware/mbed/1.1.4
Using core 'arduino' from platform in folder: /root/.arduino15/packages/arduino/hardware/mbed/1.1.4
Detecting libraries used...
/root/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -g -Os -nostdlib @/root/.arduino15/packages/arduino/hardware/mbed/1.1.4/variants/ARDUINO_NANO33BLE/defines.txt @/root/.arduino15/packages/arduino/hardware/mbed/1.1.4/variants/ARDUINO_NANO33BLE/cxxflags.txt -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED -I/root/.arduino15/packages/arduino/hardware/mbed/1.1.4/cores/arduino -I/root/.arduino15/packages/arduino/hardware/mbed/1.1.4/variants/ARDUINO_NANO33BLE -I/root/.arduino15/packages/arduino/hardware/mbed/1.1.4/cores/arduino/api/deprecated -iprefix/root/.arduino15/packages/arduino/hardware/mbed/1.1.4/cores/arduino @/root/.arduino15/packages/arduino/hardware/mbed/1.1.4/variants/ARDUINO_NANO33BLE/includes.txt /tmp/arduino-sketch-E4324F3987644454FD681029F1E24959/sketch/blinkYou.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Generating function prototypes...
/root/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -g -Os -nostdlib @/root/.arduino15/packages/arduino/hardware/mbed/1.1.4/variants/ARDUINO_NANO33BLE/defines.txt @/root/.arduino15/packages/arduino/hardware/mbed/1.1.4/variants/ARDUINO_NANO33BLE/cxxflags.txt -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED -I/root/.arduino15/packages/arduino/hardware/mbed/1.1.4/cores/arduino -I/root/.arduino15/packages/arduino/hardware/mbed/1.1.4/variants/ARDUINO_NANO33BLE -I/root/.arduino15/packages/arduino/hardware/mbed/1.1.4/cores/arduino/api/deprecated -iprefix/root/.arduino15/packages/arduino/hardware/mbed/1.1.4/cores/arduino @/root/.arduino15/packages/arduino/hardware/mbed/1.1.4/variants/ARDUINO_NANO33BLE/includes.txt /tmp/arduino-sketch-E4324F3987644454FD681029F1E24959/sketch/blinkYou.ino.cpp -o /tmp/arduino-sketch-E4324F3987644454FD681029F1E24959/preproc/ctags_target_for_gcc_minus_e.cpp -DARDUINO_LIB_DISCOVERY_PHASE
/root/.arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino-sketch-E4324F3987644454FD681029F1E24959/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/root/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -g -Os -nostdlib @/root/.arduino15/packages/arduino/hardware/mbed/1.1.4/variants/ARDUINO_NANO33BLE/defines.txt @/root/.arduino15/packages/arduino/hardware/mbed/1.1.4/variants/ARDUINO_NANO33BLE/cxxflags.txt -DARDUINO_ARCH_NRF52840 -MMD -mcpu=cortex-m4 -DARDUINO=10607 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED -I/root/.arduino15/packages/arduino/hardware/mbed/1.1.4/cores/arduino -I/root/.arduino15/packages/arduino/hardware/mbed/1.1.4/variants/ARDUINO_NANO33BLE -I/root/.arduino15/packages/arduino/hardware/mbed/1.1.4/cores/arduino/api/deprecated -iprefix/root/.arduino15/packages/arduino/hardware/mbed/1.1.4/cores/arduino @/root/.arduino15/packages/arduino/hardware/mbed/1.1.4/variants/ARDUINO_NANO33BLE/includes.txt /tmp/arduino-sketch-E4324F3987644454FD681029F1E24959/sketch/blinkYou.ino.cpp -o /tmp/arduino-sketch-E4324F3987644454FD681029F1E24959/sketch/blinkYou.ino.cpp.o
In file included from /root/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/bits/uniform_int_dist.h:35:0,
                 from /root/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/bits/stl_algo.h:66,
                 from /root/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/algorithm:62,
                 from /root/.arduino15/packages/arduino/hardware/mbed/1.1.4/cores/arduino/mbed/drivers/FlashIAP.h:31,
                 from /root/.arduino15/packages/arduino/hardware/mbed/1.1.4/cores/arduino/mbed/mbed.h:75,
                 from /root/.arduino15/packages/arduino/hardware/mbed/1.1.4/cores/arduino/Arduino.h:32,
                 from /tmp/arduino-sketch-E4324F3987644454FD681029F1E24959/sketch/blinkYou.ino.cpp:1:
/root/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/limits:1599:7: internal compiler error: Illegal instruction
       min() _GLIBCXX_USE_NOEXCEPT { return __FLT_MIN__; }

### Expected behavior

To compile cleanly the way if does with other tried boards
### Environment

- CLI version (output of `arduino-cli version`): arduino-cli Version: 0.11.0 Commit: 0296f4d
- OS and platform: Linux raspberrypi 4.19.118+ #1311 Mon Apr 27 14:16:15 BST 2020 armv6l GNU/Linux

### Additional context

I encounter this issue only when working with the new arduino nano 33BLE sense, but this is the only recent board I have tried

ubidefeo, 2020-07-08:
hi @DrMomentum 
do you have the same compiling issue if building on the Java IDE on another computer?

Do you have the most recent core for the Nano 33 BLE?

I can try building it on my Pi Zero if you share th code

DrMomentum, 2020-07-08:
Thanks for jumping into it @ubidefeo ,

- I have indeed worked with this board on my desktop computer (another Linux beast) previously using the Java IDE with no problems at all.

- Core seams to be the lattest from: [https://github.com/arduino/ArduinoCore-nRF528x-mbedos](url)
**arduino-cli core list**
ID           Installed Latest Name                            
arduino:avr  1.8.2     1.8.2  Arduino AVR Boards              
arduino:mbed 1.1.4     1.1.4  Arduino nRF528x Boards (Mbed OS)

And, about the code I used, I always start from the bottom when introducing a new board to any project I work on

This is it: ;)

byte led = LED_BUILTIN;

void setup() {
	pinMode(led, OUTPUT);
}

void loop() {
	digitalWrite(led, 1);
	delay(200);
	digitalWrite(led, 0);
	delay(200);
}

Thank you for giving it a shot!

umbynos, 2020-07-09:
Hello, I tried your code with my x86 machine with no problem at all. At this point, I think that it could be a problem regarding the toolchain and the armv6 architecture.

ubidefeo, 2020-07-09:
@umbynos would you have time to uninstall the core and reinstall it with the `-v` flag?
Also the nightly build already addresses missing tools installation by triggering a fail and notifying the user.
There was a bug which didn't display such info upon failure

ubidefeo, 2020-07-09:
@DrMomentum 
I checked with @umbynos and he doesn't have a Pi Zero to test on, so this test is on me as soon as I make it back to the lab today.
Thank you for your patience :)

ubidefeo, 2020-07-09:
hey @DrMomentum 
I took a shot at installing the `arduino:mbed` core for the Nano 33 BLE on a Pi Zero and it didn't even let me install the tools, failing with an `out of memory` error.

I'm using one of the first Pi Zero, which one are you running on?
Also this thing is so excruciatingly slow it's a pain to work with :D

DrMomentum, 2020-07-09:
@ubidefeo ,

I'm at this moment going through the uninstall-reinstall process my self. Using a Pi Zero W... indeed it is an excruciatingly slow process, but "the end justifies the means"

I'll post my results in a short while

Thank you both for your time on the matter

ubidefeo, 2020-07-09:
I brought a Zero W home so I might try later on that.
Have to say that we're not too keen on investing too much time on the `armv6` version because of how underperforming it is as well as the lack of some 3rd party tools.
The Pi 3B+ and 4, on the other hand, are incredibly satisfying :)

DrMomentum, 2020-07-10:
Well, no good new for this topic
This time it crashed like   @ubidefeo mentioned before with a: "fatal error: runtime: out of memory" while trying to install: "arm-none-eabi-gcc@7-2017q4", which by the way I was unable to find for 32 bit anywhere (output bellow)

Unfortunately I'm running out of time for the project I'm working on so, sadly I'll have to go back to using an old AVR (micro) in order to move forward  (and to aboid diverting too much of you guys time from more important projects)

I really appreciate you all's efforts on the matter

Downloading packages...
arduino:openocd@0.10.0-arduino9 already downloaded
arduino:**arm-none-eabi-gcc@7-2017q4** already downloaded
arduino:bossac@1.9.1-arduino1 already downloaded
arduino:mbed@1.1.4 already downloaded
Installing arduino:openocd@0.10.0-arduino9...
arduino:openocd@0.10.0-arduino9 installed
Installing arduino:**arm-none-eabi-gcc@7-2017q4...**
**````fatal error: runtime: out of memory````**

runtime stack:
runtime.throw(0x86f19e, 0x16)
	/usr/local/go/src/runtime/panic.go:774 +0x5c
runtime.sysMap(0x20000000, 0x800000, 0xed6f68)
	/usr/local/go/src/runtime/mem_linux.go:169 +0xa8
runtime.(*linearAlloc).alloc(0xec9fe8, 0x800000, 0x400000, 0xed6f68, 0xed61d0)
	/usr/local/go/src/runtime/malloc.go:1387 +0x94
runtime.(*mheap).sysAlloc(0xec9cf8, 0x800000, 0x326e0, 0x32740)
	/usr/local/go/src/runtime/malloc.go:607 +0x54
runtime.(*mheap).grow(0xec9cf8, 0x400, 0xffffffff)
	/usr/local/go/src/runtime/mheap.go:1252 +0x2c
runtime.(*mheap).allocSpanLocked(0xec9cf8, 0x400, 0xed6f78, 0x0)
	/usr/local/go/src/runtime/mheap.go:1163 +0x290
runtime.(*mheap).alloc_m(0xec9cf8, 0x400, 0x101, 0x11acdf87)
	/usr/local/go/src/runtime/mheap.go:1015 +0xd0
runtime.(*mheap).alloc.func1()
	/usr/local/go/src/runtime/mheap.go:1086 +0x3c
runtime.(*mheap).alloc(0xec9cf8, 0x400, 0xff010101, 0xffde9057)
	/usr/local/go/src/runtime/mheap.go:1085 +0x60
runtime.largeAlloc(0x7ffe00, 0x3000101, 0x5e3e7)
	/usr/local/go/src/runtime/malloc.go:1138 +0x74
runtime.mallocgc.func1()
	/usr/local/go/src/runtime/malloc.go:1033 +0x38
runtime.systemstack(0x152)
	/usr/local/go/src/runtime/asm_arm.s:354 +0x84
runtime.mstart()
	/usr/local/go/src/runtime/proc.go:1146

goroutine 1 [running]:
runtime.systemstack_switch()
	/usr/local/go/src/runtime/asm_arm.s:298 +0x4 fp=0x58594d8 sp=0x58594d4 pc=0x6b524
runtime.mallocgc(0x7ffe00, 0x786c30, 0x1, 0x0)
	/usr/local/go/src/runtime/malloc.go:1032 +0x8d0 fp=0x585953c sp=0x58594d8 pc=0x1d3c4
runtime.makeslice(0x786c30, 0x7ffe00, 0x7ffe00, 0x3314210)
	/usr/local/go/src/runtime/slice.go:49 +0x6c fp=0x5859550 sp=0x585953c pc=0x57280
bytes.makeSlice(0x7ffe00, 0x0, 0x0, 0x0)
	/usr/local/go/src/bytes/buffer.go:229 +0x60 fp=0x5859584 sp=0x5859550 pc=0xf1dc4
bytes.(*Buffer).grow(0x42b55a0, 0x200, 0xc14b0)
	/usr/local/go/src/bytes/buffer.go:142 +0x138 fp=0x58595a8 sp=0x5859584 pc=0xf175c
bytes.(*Buffer).ReadFrom(0x42b55a0, 0x993fb0, 0x3314220, 0xa5b76c08, 0x42b55a0, 0x1, 0x72)
	/usr/local/go/src/bytes/buffer.go:202 +0x48 fp=0x58595e4 sp=0x58595a8 pc=0xf1bc0
io.copyBuffer(0x993728, 0x42b55a0, 0x993fb0, 0x3314220, 0x0, 0x0, 0x0, 0x20, 0x7dbfb8, 0x5828201, ...)
	/usr/local/go/src/io/io.go:388 +0x2fc fp=0x5859624 sp=0x58595e4 pc=0x7ad5c
io.Copy(...)
	/usr/local/go/src/io/io.go:364
github.com/codeclysm/extract.copyCancel(0x99cd28, 0x4c70620, 0x993728, 0x42b55a0, 0x993680, 0x3098240, 0xc0, 0x0, 0x0, 0x0)
	/github/home/go/pkg/mod/github.com/codeclysm/extract@v2.2.0+incompatible/cancelable_reader.go:10 +0x94 fp=0x5859654 sp=0x5859624 pc=0x377080
github.com/codeclysm/extract.(*Extractor).Tar(0x58597b4, 0x99cd28, 0x4c70620, 0x994820, 0x4424110, 0x4296810, 0x26, 0x0, 0x869c6b, 0x11)
	/github/home/go/pkg/mod/github.com/codeclysm/extract@v2.2.0+incompatible/extractor.go:145 +0x2b0 fp=0x5859774 sp=0x5859654 pc=0x377d18
github.com/codeclysm/extract.Tar(...)
	/github/home/go/pkg/mod/github.com/codeclysm/extract@v2.2.0+incompatible/extract.go:90
github.com/codeclysm/extract.(*Extractor).Bz2(0x585991c, 0x99cd28, 0x4c70620, 0x994820, 0x47ebf60, 0x4296810, 0x26, 0x0, 0x86bb61, 0x13)
	/github/home/go/pkg/mod/github.com/codeclysm/extract@v2.2.0+incompatible/extractor.go:67 +0x16c fp=0x58597dc sp=0x5859774 pc=0x377814
[... 19 lines cut ...]
	/github/home/go/pkg/mod/github.com/spf13/cobra@v1.0.0/command.go:950 +0x26c fp=0x5859f90 sp=0x5859f28 pc=0x6083d4
github.com/spf13/cobra.(*Command).Execute(...)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v1.0.0/command.go:887
main.main()
	/__w/arduino-cli/arduino-cli/main.go:31 +0x24 fp=0x5859fa4 sp=0x5859f90 pc=0x6f75b4
runtime.main()
	/usr/local/go/src/runtime/proc.go:203 +0x208 fp=0x5859fe4 sp=0x5859fa4 pc=0x42784
runtime.goexit()
	/usr/local/go/src/runtime/asm_arm.s:868 +0x4 fp=0x5859fe4 sp=0x5859fe4 pc=0x6d380

goroutine 6 [syscall, 3 minutes]:
os/signal.signal_recv(0x0)
	/usr/local/go/src/runtime/sigqueue.go:147 +0x130
os/signal.loop()
	/usr/local/go/src/os/signal/signal_unix.go:23 +0x14
created by os/signal.init.0
	/usr/local/go/src/os/signal/signal_unix.go:29 +0x30

goroutine 10 [select, 2 minutes]:
go.bug.st/cleanup.InterruptableContext.func1(0x99cd28, 0x4c70620, 0x5104c78)
	/github/home/go/pkg/mod/go.bug.st/cleanup@v1.0.0/context.go:21 +0xe0
created by go.bug.st/cleanup.InterruptableContext
	/github/home/go/pkg/mod/go.bug.st/cleanup@v1.0.0/context.go:18 +0x50

Best,...

rsora, 2020-07-21:
Thanks for your investigation @DrMomentum, 
I'll close this one since it seems that there is not so much that we can do on the specific armv6 platform ATM.

Happy coding!

closed 2020-07-21 by rsora