arduino/arduino-cli #809: Cannot open port Permission denied on uploading to board

Inc0ns1st3nt, 2020-07-09:
## Bug Report
I am having trouble to upload my sketch after a successful compilation.
My device is a mkr wifi 1010 variant.
 
### Current behavior

Command ran:
```
arduino-cli upload --fqbn arduino:samd:mkrwifi1010 -p /dev/ttyACM0 test
```
Full message output:
```
Error during Upload: cannot perform reset: opening port: Permission denied
```

### Expected behavior
To upload the sketch onto the board

### Environment
- CLI version: arduino-cli Version: 0.11.0-arch Commit: 0296f4df116385f868b67c5ffa7393936c3345c
- OS and platform: Manjaro Lysia 20.0.3

Inc0ns1st3nt, 2020-07-09:
The solution is to add yourself to the uucp user group to get write access to tty.
```
sudo usermod -a -G uucp $USER
```

closed 2020-07-09 by Inc0ns1st3nt

antonkesy, 2023-07-26:
Ensure you sign in/out, or simply reboot your system to apply those changes