arduino/arduino-cli #821: Local .h files should not be treated as libraries

ArminJo, 2020-07-10:
## Bug Report

### Current behavior
```
#include <DigisparkOLED.h>
#include <Wire.h>
// ============================================================================

#include "img0_128x64c1.h"
#include "digistump_128x64c1.h"
```
[Compiles with 0.10.0](https://github.com/ArminJo/DigistumpArduino/runs/769713710?check_suite_focus=true#step:4:90) 
[Failes with 0.11.0](https://github.com/ArminJo/DigistumpArduino/runs/857294172?check_suite_focus=true#step:4:93) with
[fatal error: img0_128x64c1.h: No such file or directory](https://github.com/ArminJo/DigistumpArduino/runs/857294172?check_suite_focus=true#step:4:111), but file is in the [same directory](https://github.com/ArminJo/DigistumpArduino/tree/master/digistump-avr/libraries/DigisparkOLED/examples/DigisparkOLED) as the .ino file.

### Expected behavior
Local .h files should not be treated as libraries and if so, should not lead to an error!

### Environment
- OS and platform: Linux

ArminJo, 2020-07-10:
Duplicate of #484
Only popping up if build-path is used???

per1234, 2020-07-12:
A minimal demonstration of the issue:

Set things up:
```
$ arduino-cli version
arduino-cli Version: 0.11.0-rc1 Commit: e7638c7

$ mkdir /tmp/Foo

$ printf "#include \"Bar.h\"\nvoid setup() {}\nvoid loop() {}\n" > /tmp/Foo/Foo.ino

$ touch /tmp/Foo/Bar.h
```
Now let's test with a build path not in the sketch folder just to make sure everthing is fine:
```
$ arduino-cli compile -b arduino:avr:uno --build-path /tmp/build-path /tmp/Foo/Foo.ino
Sketch uses 444 bytes (1%) of program storage space. Maximum is 32256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.

$ tree /tmp/build-path
/tmp/build-path
├── build.options.json
├── core
├── Foo.ino.eep
├── Foo.ino.elf
├── Foo.ino.hex
├── Foo.ino.with_bootloader.bin
├── Foo.ino.with_bootloader.hex
├── includes.cache
├── libraries
├── preproc
│   └── ctags_target_for_gcc_minus_e.cpp
└── sketch
    ├── Bar.h
    ├── Foo.ino.cpp
    ├── Foo.ino.cpp.d
    └── Foo.ino.cpp.o

4 directories, 12 files
```
Looks good. Note that `Bar.h` was correctly copied into the `sketch` subfolder of the build folder:
```
$ ls /tmp/build-path/sketch/Bar.h
/tmp/build-path/sketch/Bar.h
```

---
Now to attempt using the sketch folder as the build folder:
```
$ arduino-cli compile -b arduino:avr:uno --build-path /tmp/Foo /tmp/Foo/Foo.ino
/tmp/Foo/Foo.ino:1:10: fatal error: Bar.h: No such file or directory
 #include "Bar.h"
          ^~~~~~~
compilation terminated.
Error during build: exit status 1

$ tree /tmp/Foo
/tmp/Foo
├── Bar.h
├── build
│   └── arduino.avr.uno
│       ├── Foo.ino.eep
│       ├── Foo.ino.elf
│       ├── Foo.ino.hex
│       ├── Foo.ino.with_bootloader.bin
│       └── Foo.ino.with_bootloader.hex
├── build.options.json
├── Foo.ino
└── sketch
    └── Foo.ino.cpp

3 directories, 9 files
```
Note that this time `Bar.h` was not copied to the `sketch` subfolder of the build folder:
```
$ ls /tmp/Foo/sketch/Bar.h
ls: cannot access '/tmp/Foo/sketch/Bar.h': No such file or directory
```
I notice it works differently on Windows
```
$ arduino-cli version
arduino-cli.exe Version: 0.11.0-rc1 Commit: e7638c7

$ arduino-cli compile -b arduino:avr:uno --build-path /tmp/Foo /tmp/Foo/Foo.ino
Sketch uses 444 bytes (1%) of program storage space. Maximum is 32256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.

$ ls /tmp/Foo/sketch/Bar.h
/tmp/Foo/sketch/Bar.h
```
Looks good, but if you want to do it again, you must remember to remove build.option.json first:
```
$ rm /tmp/Foo/build.options.json

$ arduino-cli compile -b arduino:avr:uno --build-path /tmp/Foo /tmp/Foo/Foo.ino
Sketch uses 444 bytes (1%) of program storage space. Maximum is 32256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.
```
If you don't, everything in the sketch folder is deleted:
```
$ arduino-cli compile -b arduino:avr:uno --build-path /tmp/Foo /tmp/Foo/Foo.ino
Error during build: error reading source file: open C:\Users\per\AppData\Local\Temp\Foo\Foo.ino: The system cannot find the file specified.

$ ls /tmp/Foo
build.options.json
```

---
Hi @ArminJo. Just so I can try to fully understand the use case, is there a specific reason for wanting the build output to be saved to the root of the sketch folder?

https://github.com/arduino/arduino-cli/pull/687 means that when using Arduino CLI 0.11.0 and newer, all build artifacts (e.g., .hex, .elf, .bin) are saved to the `build/<FQBN>` subfolder of the sketch, as you can see from the tree in the above output:
```
$ tree /tmp/Foo
/tmp/Foo
├── Bar.h
├── build
│   └── arduino.avr.uno
│       ├── Foo.ino.eep
│       ├── Foo.ino.elf
│       ├── Foo.ino.hex
│       ├── Foo.ino.with_bootloader.bin
│       └── Foo.ino.with_bootloader.hex
```
Do you foresee that build files other than the artifacts will be useful in the context of CI/CD?

matthijskooijman, 2020-07-13:
I somewhat recall that the Arduino IDE actually actively forbids making the build path equal to the sketch path (which makes sense, since the build path should really contain only temp files that can be entirely deleted if needed). So indeed, I wonder if this is something that should be fixed, or maybe must forbidden.

ArminJo, 2020-07-13:
@per1234 Thank you so much for investigating further. 🥇 
I adapted my action now, to set build path to the /build subdirectory. But the best news is, it is almost not longer required, since the request to have the build files available is now fulfilled by arduino-cli itself 👍 👍

closed 2020-07-13 by ArminJo