arduino/arduino-cli #824: With esp32 can't use baud=115200 option

rmisko11, 2020-07-12:
## Bug Report

### Current behavior
arduino-cli upload -vb esp32:esp32:esp32:baud=115200 -p /dev/ttyUSB4 myDualCore

Error during Upload: incorrect FQBN: getting build properties for board esp32:esp32:esp32: invalid option 'baud'

### Expected behavior

Expected the baud rate option to be accepted and used for upload.

### Environment

- CLI version (output of `arduino-cli version`): arduino-cli Version: 0.11.0 Commit: 0296f4d
- OS and platform: Linud 5.3.0-62-generic Ubuntu 18.04.4 LTS

### Additional context

I only have a problem with ESP32 boards which want to upload at 921600.  I have no problem when set baud rate for an ESP8266 board.

per1234, 2020-07-12:
Hi @rmisko11. The custom option ID used by the ESP32 platform for the "Upload Speed" option is `UploadSpeed`, not `baud`, thus the error you got:
https://github.com/espressif/arduino-esp32/blob/1.0.4/boards.txt#L123-L136
```
esp32.menu.UploadSpeed.921600=921600
esp32.menu.UploadSpeed.921600.upload.speed=921600
esp32.menu.UploadSpeed.115200=115200
esp32.menu.UploadSpeed.115200.upload.speed=115200
esp32.menu.UploadSpeed.256000.windows=256000
esp32.menu.UploadSpeed.256000.upload.speed=256000
esp32.menu.UploadSpeed.230400.windows.upload.speed=256000
esp32.menu.UploadSpeed.230400=230400
esp32.menu.UploadSpeed.230400.upload.speed=230400
esp32.menu.UploadSpeed.460800.linux=460800
esp32.menu.UploadSpeed.460800.macosx=460800
esp32.menu.UploadSpeed.460800.upload.speed=460800
esp32.menu.UploadSpeed.512000.windows=512000
esp32.menu.UploadSpeed.512000.upload.speed=512000
```
So the correct FQBN is `esp32:esp32:esp32:UploadSpeed=115200`.

More information on determining the FQBN here: https://github.com/arduino/arduino-cli/issues/138

closed 2020-07-12 by per1234

rmisko11, 2020-07-12:
Thank you for the prompt reply and the additional information that I wasn't
able to find otherwise.