arduino/arduino-cli #846: Provide dedicated sets of "extra flags" properties for platform developers and users

per1234, 2020-07-20:
Arduino has established a convention of providing "extra_flags" properties in platform.txt, which are intended to allow the compilation process to be customized by the addition of arbitrary flags in the command lines:
https://github.com/arduino/ArduinoCore-avr/blob/1.8.3/platform.txt#L39-L49
```
# This can be overridden in boards.txt
build.extra_flags=

# These can be overridden in platform.local.txt
compiler.c.extra_flags=
compiler.c.elf.extra_flags=
compiler.S.extra_flags=
compiler.cpp.extra_flags=
compiler.ar.extra_flags=
compiler.objcopy.eep.extra_flags=
compiler.elf2hex.extra_flags=
```
These are referenced in the appropriate compilation recipes in platform.txt.

The empty definitions in platform.txt ensure that if the extra flags are not defined elsewhere, the compilation recipes that reference them will still work. They can be overrriden from any of the other platform configuration files:
- [boards.txt](https://arduino.github.io/arduino-cli/latest/platform-specification/#boardstxt)
- [boards.local.txt](https://arduino.github.io/arduino-cli/latest/platform-specification/#boardslocaltxt)
- [platform.local.txt](https://arduino.github.io/arduino-cli/latest/platform-specification/#platformlocaltxt)
- [global platform.txt](https://arduino.github.io/arduino-cli/latest/platform-specification/#global-platformtxt)

or from the command line:
- [`arduino-cli compile --build-properties`](https://arduino.github.io/arduino-cli/latest/commands/arduino-cli_compile/#options) - [`arduino --pref`])(https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc#options)

From the comments, it seems that `build.extra_flags` is indended to be for the use of the platform developer, while the others are intended to be for the use of the user (because a platform developer has no need for platform.local.txt).

However, the developer often has need for the recipe-specific properties and uses them in boards.txt.

This means that the user risks overriding the flags the platform intended if they use the "extra flags" properties to modify the compilation command. A common example would be to add `-D` flags (https://github.com/arduino/arduino-cli/issues/210#issuecomment-493892258, https://github.com/arduino/Arduino/issues/6344), but there could be other reasons.

---
A solution for this conflict could be to establish two sets of the properties, each set clearly documented as being dedicated to the sole use by either the platform developer or the user.

These are only arbitrary flags, without any special use by the build system, so no changes would required in the build system. It's only a matter of setting an example for 3rd party platform developers  by adding these additional properties to the platform.txts of the official platforms with comments clearly explaining their intent, and ideally documenting this in the [Arduino platform specification](https://arduino.github.io/arduino-cli/latest/platform-specification/) as well (https://github.com/arduino/Arduino/issues/4141).

---
On a certain level, this does provide the user global macro setting capability that has been often requested, and often rejected: 

- https://github.com/arduino/arduino-builder/issues/15
- https://github.com/arduino/arduino-builder/pull/29
- https://github.com/arduino/arduino-builder/pull/282
- https://github.com/arduino/arduino-builder/issues/283
- https://github.com/arduino/Arduino/issues/2547
- https://github.com/arduino/Arduino/issues/3635
- https://github.com/arduino/Arduino/pull/3717
- https://github.com/arduino/Arduino/pull/3757
- https://github.com/arduino/Arduino/issues/4528
- https://github.com/arduino/Arduino/issues/5084
- https://github.com/arduino/Arduino/issues/5390
- https://github.com/arduino/arduino-cli/issues/159

However, I think this proposal strikes the right balance of:
- Allowing advanced users to do advanced things
- Not exposing standard users to additional complexity that makes the Arduino development software less beginner friendly
- Being very easy to implement

The concern has been that allowing users to set global macros will result in library developers unnecessarily using these as part of the API, which will be much less beginner friendly (https://github.com/arduino/arduino-builder/issues/15#issuecomment-145558252). However, if that interface can only be used via the user is required to use a command line interface or create a specially formatted configuration file in a difficult to access location, the average user won't be interested, and so it will only make sense for a library developer to use this feature for advanced, rarely used features of the library (e.g., enable debug output for troubleshooting of the library. We have already had this "extra flags" capabilities for years, yet I haven't seen any significant incorporation of it into the user-level API of libraries.

alranel, 2020-07-22:
Some additional thoughts:

* Even if we clarify the semantics in the documentation, we'd still have a legacy. Maybe with a simple crawler we can estimate how many existing cores are setting these flags (and which ones). We might consider opening PRs on all existing cores that are setting a flag intended for users.
* We could also print warnings to the console at compile time whenever something is overridden, to help users troubleshoot.
* Those `compiler.*.extra_flags` seem a bit unfriendly and fragmented for the average user who may just want to add a `-DFOOBAR` and may be unsure about where exactly should that put
* As an alternative to all this, could we consider a syntax that allows to _append_ flags? Such as `build.extra_flags+=-DFOOBAR`

bxparks, 2020-08-23:
So I discovered through hours of trial and error that the`--build-properties` flag has a broken parser, so it cannot be a substitute for `-D` (arduino-cli version: 0.12.0.-rc3, on Ubuntu Linux 20.04.):

1) It seems to currently handle only `-DMACRO` or `-DMACRO=2`, but not strings like `-DMACRO="This Is A String"`:
```
$ arduino-cli  compile --fqbn arduino:avr:nano:cpu=atmega328old  --build-properties 'compiler.cpp.extra_flags=-DPIN=2 -DSSID="This is a String"' PrintStringTest.ino
Error: invalid argument "compiler.cpp.extra_flags=-DPIN=2 -DSSID=\"This is a String\"" for "--build-properties" flag: parse error on line 1, column 40: bare " in non-quoted-field
```

2) It does not parse embedded commas properly:
```
$ arduino-cli  compile --fqbn arduino:avr:nano:cpu=atmega328old  --build-properties 'compiler.cpp.extra_flags=-DPIN=2 -DSSID="Embedded,Comma"' PrintStringTest.ino
Error during build: Error reading from slice (index:1): Invalid line format, should be 'key=value'
```

3) No amount of backslash escaping works:
```
$ arduino-cli  compile --fqbn arduino:avr:nano:cpu=atmega328old  --build-properties 'compiler.cpp.extra_flags=-DPIN=2 -DSSID=\"Embedded,Comma\"' PrintStringTest.ino
Error: invalid argument "compiler.cpp.extra_flags=-DPIN=2 -DSSID=\\\"Embedded,Comma\\\"" for "--build-properties" flag: parse error on line 1, column 41: bare " in non-quoted-field
```

4) Note that the Arduino IDE's `--pref` flag handles the parsing correctly, so the CLI's `--build-properties` is not a replacement for the IDE's `--pref`:
```
$ /home/brian/dev/arduino-1.8.13/arduino  --verify --board arduino:avr:nano:cpu=atmega328old  --pref 'compiler.cpp.extra_flags=-DPIN=2 -DSSID="Embedded,Comma"' PrintStringTest.ino
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
[...]
Sketch uses 6136 bytes (19%) of program storage space. Maximum is 30720 bytes.
Global variables use 717 bytes (35%) of dynamic memory, leaving 1331 bytes for local variables. Maximum is 2048 bytes.
```

5) With regards to the problem of defining SECRETS, you write:

> A system for doing this has been established in the Arduino Web Editor:
> https://arduino.github.io/arduino-cli/dev/sketch-specification/#secrets

Hmm, I've never heard of the Arduino Web Editor. I'm someone who writes code using the vim(1) editor and uses git(1) on the command line for source control. You want me to code in a web browser, while forced to be connected to the network? I'm not sure how this is the solution to the SECRETS problem.

6) With regards to the needing the `-D` flag to target different hardware, you write:

> A macro is defined for each board according to ARDUINO_{build.board}:
> https://arduino.github.io/arduino-cli/dev/platform-specification/#boardstxt

I think you misunderstood. I am not just looking for detecting `ARDUINO_AVR_NANO` or `ESP8266`. I have built various hardware devices, using different microcontrollers, each with different switches, buttons, sensors, displays on different GPIO pins (e.g. `PIN=2` or `PIN=D3`) and so on. I want to compile the same program targeted to different hardware devices, with slightly different configurations using the `-D` flag in a shell script or a `Makefile`.

I also write a fair number of Libraries, which have test programs (sketches) which I need to run across many different microcontrollers to make sure that they run properly on various boards. It would be impractical to run multiple dozens of validation programs manually through the Arduino IDE, manually changing various configuration options, without being able to create scripts to automate the testing.

7) I am not in favor of overloading the `--fqbn` flag to hold the `-D` macros. Here is the fqbn for one of my ESP8266:
```
esp8266:esp8266:generic:xtal=80,vt=flash,exception=disabled,ResetMethod=ck,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=512K,led=2,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200
```

Hiding a `-D` macro in there seems to be unnecessarily punitive to developers.

**Additional Thoughts on the -D flag**

I have to say that I am constantly surprised at the push-back against the `-D` flag. It is a feature that has been requested many times, by many people, over many years, as indicated by the original post. It's a significant pain point for many people. It's a simple, focused request. It is a well-defined feature, since it is directly supported by the `gcc` compiler, and is familiar to any programmer who knows about the `gcc` compiler. We don't even have to write a manual for it, we can just point people to the [gcc compiler preprocessor options](https://gcc.gnu.org/onlinedocs/gcc/Preprocessor-Options.html#Preprocessor-Options).

I am reaching a point where the `-D` feature might be the thing that finally breaks the camel's back. In other words, I am considering giving up on the Arduino platform without the `-D` flag. I spent 8 hours yesterday trying to get `--build-properties` flag to work with strings. It doesn't work, at least I couldn't find any way to escape the broken parser. I'm tired of fighting the Arduino tools. I would rather spend my time writing interesting Libraries that I can share with others, and write interesting Arduino applications.

per1234, 2020-08-24:
>So I discovered through hours of trial and error that the--build-properties flag has a broken parser

Thanks for reporting this. Would you mind opening a dedicated issue report?

>You want me to code in a web browser, while forced to be connected to the network? 

Not at all. But note that link also explains how the system is applied to local sketches as well. I already explained how you could use that to solve your issue of needing to define secrets in a way that is not checked in to version control.

However, the main reason I mention it is because we always need to consider the systems that have been previously established in the other official Arduino development tools in order to provide as consistent a user experience as possible.

>I want to compile the same program targeted to different hardware devices, with slightly different configurations using the -D flag in a shell script or a Makefile.

I understand now. Thanks for the explanation.

>I have to say that I am constantly surprised at the push-back against the -D flag. It is a feature that has been requested many times, by many people, over many years, as indicated by the original post. It's a significant pain point for many people. It's a simple, focused request.

OK, well this is my proposal for providing that feature.

You didn't answer my question from before: Assuming it became possible to add quotes to property values set via `--build-flags`, would the proposal I present here then be sufficient to make you feel that Arduino CLI provides the "easy -D flag mechanism" you are requesting?

>I spent 8 hours yesterday trying to get `--build-properties` flag to work with strings. It doesn't work, at least I couldn't find any way to escape the broken parser.

I'm sorry to hear that you had a bad experience using Arduino CLI. I hope that anyone who writes code will understand that bugs do happen sometimes. This project is a work in progress, hopefully steadily improving overall. High quality bug reports and feature requests are valuable contributions to that effort.

matthijskooijman, 2020-08-27:
One addition here: If we provide such a set of "user-provided flags", we should probably make sure to provide both recipe-specific flags (e.g. only for .c files, or .cpp files, or only for linking, maybe even also for the ar and elf2hex recipes) and a property to add common flags (passed to the `c`, `cpp` and `S` recipes, maybe also to the link, though that's probably not required). That should make it convenient to just add extra flags "to the build", but also allow full control for flags that apply to only one or two filetypes.

One flag var that is missing in the first post is the one for `recipe.preproc.macros`, which should probably also get a user-changeable flags variable.

Furthermore, I would suggest using the name `link` (rather than `c.elf`) for the link flags (e.g. `build.user_link_extra_flags`) to make it more clear where it is used (an extension of that would be to rename the recipe itself, too, but that's a different issue).

What would be the next step to move this forward? I think writing a more detailed proposal about which properties a core is recommended to expose (ideally in a form that can be included in the documentation as-is)? This proposal would at least need to talk about the "user-facing" properties, but maybe it could also make some (more casual) recommendations about internal properties to use (for consistency)?

bxparks, 2020-08-31:
**Broken Parser in --build-properties**

It seems to me that the`--build-properties` flag, as currently defined, cannot provide the functionality of the `-D` flag because the comma-character is used as a delimiter: `--build-properties strings   List of custom build properties separated by commas. Or can be used multiple times for multiple properties.`

Let's say that we fix the current parser so that the comma-character is ignored inside a double-quote. That doesn't fix the parsing, because the following will not work:
```C++
#define COMMA ,
#define GREETINGS {"hello", "hey", "g'day mate"}
```

Maybe we attempt to fix that by requiring the right-side of `compiler.cpp.extra_flags=` to be encoded in a single quote, like this: `--build-properties "compiler.cpp.extra_flags='-DPIN=2 -DSSID=\"This is a String\"'"`. But we can break that parser trivially by attempting to encode this:
```C++
#define NEWLINE '\n'
```

Any attempt to add backslash-escaping of the comma (`\,`), the single-quote (`\'`), or the double-quote (`\"`) will result in an incomprehensible nesting of multiple backslashes, as we try to escape the quoting and backslash mechanism of the bash shell.

The only way I can see `--build-properties` working is to allow only a single `xxx=yyy` Java `.properties` line to be associated with a single instance of the `--build-properties` flag.

**Unreasonable Complexity of --build-properties**

My biggest issue with the `--build-properties` is the unnecessary complexity of this solution for solving such a simple problem. Application and Library developers want a simple thing, the `-D` flag. They should not need to know the details of how Arduino platforms are assembled and configured. In particular, application and library writers should not need to know the internal defails of the various `boards.txt` and `platform.txt` files.

The original propose in this ticket, with regards to splitting the extra flags between "platform" flags and "end-user" flags solves the problem with accidental clobbering. But I don't think it solves the problem of unnecessary complexity. I don't think the Application and Library writer should need to know that the configuration they need to override is in the `boards.txt` file, and it is called `compiler.cpp.extra_flags` (or under @per1234 proposal, it might be `compiler.cpp.user_extra_flags.user`).

I have a similar feeling about the `+=` syntax. I don't think end-users (i.e. app and library developers) should need to go into the internal details of `boards.txt`.

**Thoughts on arduino_secrets.h**

I've never used the Arduino Web Editor. If I understand the https://arduino.github.io/arduino-cli/dev/sketch-specification/#secrets page correctly, the `arduino_secrets.h` file is generated upon downloading of the sketch. I have a couple of thoughts:

1) I personally don't make much operational distinction between "secrets" (such as SSIDs and PASSWORDs) and configurations such as:
  * `#define LED_PIN 3`
  * `#define BUTTON_PINS {4,5}`

    Both types of things are configuration parameters that need to be injected into an app using a `-D` flag. I agree that there is some semantic difference in how carefully we should treat SECRETs. But for the purposes of Arduino programs, I think they can be treated about the same. After all, not many people will need to use encrypted KMS (Key Management Systems) for Arduino secrets. And if they did, they could still use the `-D` flag to inject those secrets.

2) There seems to be an assumption that a single `.ino` file has only a single set of secrets or a single set of configurations. For many people asking for the `-D`, I don't think this is the case. For example, I have 2 SSIDs, an "internal" SSID and an "external" SSID. Certain IoT devices go on certain networks. (I actually have 6 SSIDs, but I'm simplifying here.) Sometimes I want to place the same Arduino app on different networks, on different hardware devices, configured slightly differently, for debugging purposes. Having a single `arduino_secrets.h` file will be limiting. I would probably write a script to dynamically generate the `arduino_secrets.h` file. But if I'm going to do that, it would be far easier to just directly call the compiler with the appropriate `-D` flags.

    It looks like I currently maintain about 200 `.ino` files. It is impractical to maintain 200 `arduino_secrets.h` files. What I have is a set of 5-10 hardware devices (with known PIN numbers and other I/O configurations), a set of 6 SSIDs, and some other secrets, all of them organized into about 10-15 "environments". The 200 INO files are frequently validated against the 10-15 environments, not all of them on all environments, but maybe the total of 400-600 different combinations of INO files and environments. The only way to maintain these is through automated scripting, and validating them through continuous integration.

**Summary**

1) I think the current `--build-properties` flag is broken. I think it can be fixed by removing the support for comma-delimiting.
2) I think the `--build-properties` is an unnecessarily complex solution to the simple problem of `-D`. Splitting the various properties into 'platform developers' versus 'end-users' does not solve the complexity problem.
3) The `arduino_secrets.h` is not a good substitute for `-D` flag when there is a combinatorial explosion of INO programs, combined with various secrets and configuration parameters.
[... cut ...] that solves the widest range of configuration problems (secrets and configurations).

Given how long and how often the `-D` flag has been requested, I am sure that I am not making any new arguments. So I expect my request to be rejected as well. But if a miracle happens and the proposal is accepted, then I can also volunteer to do the work. I will learn the Go Language just to implement this feature. (I'm already pretty fluent in C, C++, Java, Python and JavaScript. Go should not be impossible.)

per1234, 2020-08-31:
>I think the current --build-properties flag is broken. I think it can be fixed by removing the support for comma-delimiting.

If I understand https://github.com/arduino/arduino-cli/issues/532 correctly, it's a report of this same type of problem caused by the support for a comma-separated list of properties, and also proposes the comma-separated list support be dropped.

matthijskooijman, 2020-08-31:
> If I understand #532 correctly, it's a report of this same type of problem caused by the support for a comma-separated list of properties, and also proposes the comma-separated list support be dropped.

Yup, that seems to be *exactly* this problem, so let's continue discussion about it there, and assume it will be solved somehow.

>  In particular, application and library writers should not need to know the internal defails of the various boards.txt and platform.txt files.

The purpose of this issue is to define a standard set of properties, so that users do not need to know the internals of the platform.txt file. If this standard property set is adopted, it can be documented in the `arduino-cli` manpage, and users do not need to look further than that.

The only caveat is that existing cores will not automatically and instantly adopt the new properties, so it might be important to let cores that support the new properties define them with empty values in platform.txt, as an indication that they support it. If so, arduino-cli could maybe provide a warning when setting a property that is not defined (with an empty value, maybe only specifically for these properties?).

However, since platform.txt *is* the canonical source for how to run the compiler, it has to be involved when adding compiler -D options somehow, and I think there is currently no standardized attribute that the IDE/arduino-cli can define that is always included in the build. There is `{includes}`, which could work, but that would be an abuse. arduino-cli could also just append something to the resulting recipe, but that's a hack that will bite us in the future, so I'm strongly against that.

Hence, the only reasonable way to add flags to the commandline, is by defining a standardized set of build properties like this issue suggests.

What we could maybe do, of course, is *in addition to* the `--build-properties` flag add something like `--extra-define FOO=BAR`, which is a convenience helper that internally just does the same as `--build-properties build.user_extra_flags=-DFOO=BAR`, so it uses the same interface to the platform, but improves the interface for the user.

One additional suggestion: maybe it would be good to define three sets of properties. The two suggested before (one for the platform/board developers to set internally, one for the user to set through e.g. `--build-properties`), and then a third to be set by the tooling (i.e. arduino-cli). This keeps the door open for arduino-cli to generate additional flags when needed later (I can imagine at some point we allow libraries to declare compile-time options in some generic format, converted into -D options by arduino-cli, or the `--extra-define` suggested above could also use this "tooling" property set, to prevent conflict with manually supplied `--build-properties`).

d-a-v, 2020-11-20:
Since sketches.ino are parsed first, is it a silly idea to hijack gcc's \#pragma directive like this:
```
#pragma message ArduinoBuildFlag A=1 B=2
```
This sentence will be displayed by gcc at compile time, but before that Arduino builder can parse and recognise it and add relevant `-DA=1` `-DB=2` to `build-extra-flags`.

Having the ability to define macros in such way directly from a sketch would be IMHO extremely useful.

matthijskooijman, 2020-11-20:
> Since sketches.ino are parsed first, is it a silly idea to hijack gcc's #pragma directive like this:

I think that having *some* way to configure build flags from a sketch would be a useful addition, but I do not think that this is the right issue to dicuss this. I would want to keep this issue focused on standardizing the build properties offered by cores to allow customizing the build flags in a generic way. If we solve this issue, we can build a per-sketch-flags feature on top of it. As for the per-sketch-flags feature, I know this has been discussed repeatedly in the past, but I'm not sure if there is any currently open issue to track this? Closest I could find is https://github.com/arduino/Arduino/issues/421, but that's really an old and broader issue. @per1234, do you know a better one to refer to?

Coming back to the topic of this issue, I previously said:

> One additional suggestion: maybe it would be good to define three sets of properties. The two suggested before (one for the platform/board developers to set internally, one for the user to set [...]

If some way is added to define per-sketch flags, I think these could be included in the "tooling" set as well, so no extra changes are needed here to prepare for per-sketch flags later.

ubidefeo, 2020-11-21:
@matthijskooijman this also to be discussed as a `sketch.json` feature.
@cmaglie and I are meant to have a sit-down on the first implementation but our schedules are a bit insane at the moment.
We'll share our first idea of an implementation as soon as we have it ✌️

d-a-v, 2021-06-29:
There still are efforts from platforms implementers to try to do *something* for their users, which are sometimes [broken at mid-flight](https://github.com/arduino/arduino-cli/issues/1338).
It seems the need of some standardization from the official Arduino designers is not fading away.

baltitenger, 2022-03-23:
Apparently `--build-property` isn't *completely* broken, I think I've managed to get it to work in all cases. I'm using a makefile for my arduino projects, and setting this flag for the `arduino-cli compile` call seems to work: `--build-property compiler.cpp.extra_flags='$(shell printf "'\\\\''%s'\\\\'' " $(flags))'`.
Basically, what it's parsing is supposed to be shell quoting, but not really, as it only recognises quotes if they are at the start and end of the "words".

e.g. setting `-I/custom/include/dir -Dfoo='"bar, baz\n"'` (literally how you'd set them on the command line) results in the flag being set to `compiler.cpp.extra_flags=''\''-I/custom/include/dir'\'' '\''-Dfoo="bar, baz\n"'\'' '`. It's not pretty but it does the job.

zfields, 2022-05-20:
To make a clear and obvious place for "users" to modify the build configuration, why not make a `user` name space?

For example:

```
# This can be overridden in boards.txt
build.extra_flags=

# These can be overridden in platform.local.txt
compiler.c.extra_flags=
compiler.c.elf.extra_flags=
compiler.S.extra_flags=
compiler.cpp.extra_flags=
compiler.ar.extra_flags=
compiler.objcopy.eep.extra_flags=
compiler.elf2hex.extra_flags=
```

Would become something like:

```
# This can be overridden in boards.txt
build.user.extra_flags=

# These can be overridden in platform.local.txt
compiler.user.c.extra_flags=
compiler.user.c.elf.extra_flags=
compiler.user.S.extra_flags=
compiler.user.cpp.extra_flags=
compiler.user.ar.extra_flags=
compiler.user.objcopy.eep.extra_flags=
compiler.user.elf2hex.extra_flags=
```

A dedicated namespace would make it clear which variables are reserved for users vs. which variables are to be used by platform vendors. Currently there is much confusion resulting from the lack of clear delineation of variable names.

To provide an example, `build.extra_flags` is not available on Adafruit boards, but the seemingly more specific `compiler.cpp.extra_flags` is available. It is neither obvious or intuitive that this would be the case. Having a dedicated namespace for the end user would make it easy for users to know what to do and for vendors to know how to support them.

ObviousInRetrospect, 2022-08-11:
There is another reason not mentioned here explicitly to do this. Some cores allow configuration tuning by define. As an example, https://github.com/SpenceKonde/DxCore/blob/master/megaavr/cores/dxcore/UART.h will accept SERIAL_RX_BUFFER_SIZE but today the only way to change it is globally for all sketches (either by editing UART.h - yuck, or perhaps in platform.local.txt).

It would be far better if these could be overridden on a per-sketch basis. For example, I have a couple sketches that use GPS units. Those sketches really want large RX buffers to deal with the bursty data coming off the GPS. On larger bart-count parts that consumes a significant fraction of the available memory. The global hacks also tend to get undone during upgrades.

A platform.sketch.txt doing this in the sketch (or sketch.json, or really any other mechanism) for sketch-level global defines would be extremely valuable. IMO this is one of the major gaps that pushes people to platformio.

DaleSchultz, 2023-01-23:
I would like to add my weight and support for a feature to achieve the ability to define a macro on the compile command - that won't destroy things in the depths of Arduino, i.e. at the pure user level.

My use case is that I need to build my code in different versions for different screen hardware. I have a define and some some conditional statements as follows:

```#define SCREEN 32

#if (SCREEN == 64)
// yellow/blue or all white screen:
U8G2_SSD1306_128X64_NONAME_F_HW_I2C  u8g2  (U8G2_R0);
const String FW_VERSION = "64p" ; 
#elif (SCREEN == 32)
// SSD1306 128x32 4P  small monochrome screen
U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C u8g2 (U8G2_R0);
const String FW_VERSION = "32p" ;
etc....
```

When I saw the ability to set a macro via build-properties I thought I would be able to set SCREEN on the fly. When I tried that it was replacing all the build properties of the board, and the build failed.

I think it would be great to have the ability to define things on the fly, for example:
`Arduino-ide compile -b fqbn -e --define "SCREEN 32" --define "NODEBUG"`

The effect of this new compiler flag is to _prepend_ the defines to the file to be compiled. Perhaps "--macro" string or "--prepend" would be more generic.  --prepend "#define SCREEN 32"

In the meantime I have developed a workaround in the form of a batch file:

```
echo #define SCREEN 0 > SCREEN.h
arduino-cli compile -b esp8266:esp8266:d1_mini GenericRS.ino -e 
copy build\esp8266.esp8266.d1_mini\GenericRS.ino.bin NoScreen.vL.bin
 
echo #define SCREEN 64 > SCREEN.h
arduino-cli compile -b esp8266:esp8266:d1_mini GenericRS.ino -e 
copy build\esp8266.esp8266.d1_mini\GenericRS.ino.bin SSD1306-64x128.vL.bin
 
echo #define SCREEN 32 > SCREEN.h
arduino-cli compile -b esp8266:esp8266:d1_mini GenericRS.ino -e 
copy build\esp8266.esp8266.d1_mini\GenericRS.ino.bin SSD1306-32x128.vL.bin
```

The batch file creates a dedicated .h file and I have simply included that file in my .ino sketch. I can edit that file from the Arduino IDE and manually compile as needed.

egnor, 2024-05-21:
In _theory_ this proposal is basically a documentation change, along with a lot of elbow grease and cat-herding to get the proposal (e.g. `build.user_defines`) implemented, right? It doesn't need major (or any) changes to the Arduino IDE or CLI code base that I can see?

And it's necessary for any effort to allow project-specific definitions (e.g. https://github.com/arduino/tooling-rfcs/issues/9 where I recently spelled out many use cases and why standard workarounds aren't great).

The one thing I don't know how to readily manage is that different compilers may have different syntax for specifying defines (`-DFOO=BAR` vs `/DFOO=BAR`, different conventions for quoting arguments, etc). I'm not sure how easy it is for board/platform definitions to translate from something common into whatever their compiler expects. Maybe this is already an issue platform definitions have to deal with?

per1234, 2024-05-21:
> In theory this proposal is basically a documentation change

That is correct.

> a lot of elbow grease and cat-herding to get the proposal (e.g. build.user_defines) implemented

The necessary changes to implement these properties in each platform are quite minimal, straightforward, and risk-free. The undocumented `compiler.*.extra_flags` convention is already implemented in many platforms.

> It doesn't need major (or any) changes to the Arduino IDE or CLI code base that I can see?

None at all. It leverages the standard platform configuration functionality that is already in place in any Arduino development tool in use today (even in the outdated 1.x versions of Arduino IDE).

> The one thing I don't know how to readily manage is that different compilers may have different syntax for specifying defines (-DFOO=BAR vs /DFOO=BAR

It is not relevant to this proposal. The user passes the complete arguments via the property so the user simply must use the appropriate syntax.

egnor, 2024-05-21:
> [Syntax for defines] is not relevant to this proposal. The user passes the complete arguments via the property so the user simply must use the appropriate syntax.

So the user would need to know the syntax of the compiler in use by the platform? (And how to quote values, etc.?)

per1234, 2024-05-21:
Yes. These properties are intended to allow advanced users to do advanced things. The user must understand the compiler's command line interface before they start injecting arguments into the commands.

Non-advanced users don't have any need for this functionality. If a developer is designing their project in a way that forces a novice user to use this feature, that is a mistake on their part and we will not make any effort to cater to such misuse of the feature.

drankinatty, 2025-09-09:
Picking up an Arduino Nano a couple of days ago, I immediately ran into this issue (see: [Reference for .ino file structure? Beyond setup() and loop() - comment 11](https://forum.arduino.cc/t/reference-for-ino-file-structure-beyond-setup-and-loop/1405552/11) where my sole need was to pass `-DCHGDIR` to change the order in which the opamps on a LM324 chip were called.

I found and used `--build-property build.extra_flags=-DCHGDIR` not understanding that `.extra_flags=` was used interally by other boards. (noting that theses flags **overwrite**, I had looked for a extra-flag that was empty for Nano to use) 

That is where this issue is important. There needs to be a simple way to "Add" or "Append" to the build string (command-line) that doesn't alter the current platform/board setup. I had thought that `arduino-cli` may just "pass what was left over" on the command-line making the addition of a define no different from from adding `-DWHAT_EVER` to the gcc/clang command-line, but that wasn't available.

Even for very basic example code, I may have several defines I need to be able to pass to alter the code behavior by conditional inclusion, just to isolate whatever part of the code behavior I want to focus on.

There are good ideas in this issue already that would provide this capability and I don't have any that would be better. so I'll leave how to do it up to the powers that be, but it would be much appreciated to simply have a way to pass a define (or any other part of the build string) on the `arduino-cli` command line without first having to search to make sure whatever `.extra_flag=` was already used in some manner somewhere else. If there was a way to pass "whatever is left over" on the command line as part of the build string, that would be more than adequate as well.