arduino/arduino-cli #863: Can't compile with fqbn arduino:avr:nano

lefuturiste, 2020-07-23:
## Bug Report

### Current behavior

`arduino-cli compile --fqbn arduino:avr:nano:cpu=atmega328old ./arduino_switches/`

The compile command isn't working it output:

`Error during build: fork/exec /bin/avr-g++: no such file or directory`

### Expected behavior

Well, I expected to compile

### Environment

`arduino-cli Version: 0.11.0 Commit: 0296f4d`

`Linux raspberrypi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux`

### Additional context

I've installed these packages: `arduino arduino-core arduino-mk`

ubidefeo, 2020-07-23:
hi @lefuturiste 
could you `cd` to the sketch folder, run 
`arduino-cli compile --fqbn arduino:avr:nano:cpu=atmega328old -v`
and report here?

also, could you run `aduino-cli core list` and also copy/paste it here?

You're running on `armv7l which tells me Pi 2, and when I asked @per1234 he confirmed that AVR tools for arm should be installed unless you specifically install a really old version of the core.

You may have run in a download issue during install, and version `0.11.0` has a bug we fixed in the nightly.
Basically if a download fails during installation you receive confirmation of a successful installation process but some tools might be missing.

Try this:
`arduino-cli core uninstall arduino:avr -v`
`arduino-cli core install arduino:avr -v`

it might fix your woes.
please keep us posted as we're iterating very fast through issues and bug-fixing, and your feedback is precious :)

lefuturiste, 2020-07-24:
Hi,

Actually I'm not on a raspberry pi 2 but I'm on a raspberry pi 3 B+

```
OS: Raspbian 10 buster
Kernel: armv7l Linux 4.19.118-v7+
CPU: ARMv7 rev 4 (v7l) @ 4x 1.4GHz
GPU: BCM2708
RAM: 107MiB / 874MiB
```
```
$ /usr/bin/arduino-cli core list
ID           Installed Latest Name                                        
arduino:avr  1.8.3     1.8.3  Arduino AVR Boards                          
arduino:samd 1.8.6     1.8.6  Arduino SAMD Boards (32-bits ARM Cortex-M0+)
```

- `/usr/bin/arduino-cli core uninstall arduino:samd -v` [OUTPUT](https://gist.github.com/lefuturiste/f394f7e58c8ddc173bcaa86848ee8e72)
- `/usr/bin/arduino-cli core uninstall arduino:avr -v` [OUTPUT](https://gist.github.com/lefuturiste/1a4870d19b5e0fe4c0612008eea549e0)
- `/usr/bin/arduino-cli core install arduino:avr -v` [OUTPUT](https://gist.github.com/lefuturiste/2968117e8dfaf30f62cadbad8163f341)

After doing all of these operations
Output of `/usr/bin/arduino-cli core list`:
```
ID          Installed Latest Name              
arduino:avr 1.8.3     1.8.3  Arduino AVR Boards
```

ubidefeo, 2020-07-25:
hey @lefuturiste 
I ran a test on a Pi 3B+ and it works but it seems your install fails for some reason.
I'm running Buster and it compiles under 1.8.2
```
pi@automator:~/BlinkMe $ arduino-cli core list
ID           Installed Latest Name
arduino:avr  1.8.2     1.8.3  Arduino AVR Boards
arduino:samd 1.8.6     1.8.6  Arduino SAMD Boards (32-bits ARM Cortex-M0+)

pi@automator:~/BlinkMe $ arduino-cli compile -b arduino:avr:nano:cpu=atmega328old
Sketch uses 444 bytes (1%) of program storage space. Maximum is 30720 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.
```

from your log (thank you) I can see this error, and I need to check with @facchinm if the file may be damaged in case of timeout (takes quite a long time for me as well), since an EOF shouldn't be there
```
INFO[0964] Installing tool                               Tool="arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7"
Installing arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7...
arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7 15.93 MiB / 33.08 MiB [==========================================>---------------------------------------------]  48.17% 16m42sWARN[0993] Cannot install tool                           Tool="arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7" error="extracting archive: Read contents of file /home/pi/.arduino15/tmp/package-982116798/avr/lib/gcc/avr/7.3.0/avrxmega4/libgcc.a: unexpected EOF"
```

I have installed arduino:avr*1.8.3 and don't get that error, hence my sketch compiles.

Please give a few more tries at uninstalling and re-installing, or even try
`arduino-cli core install arduino:avr@1.8.2` just in case

u

closed 2020-09-27 by github-actions[bot]

ubidefeo, 2020-09-27:
hi @lefuturiste 
the bot automatically closed the issue, have you solve it with a more recent version of the CLI?