arduino/arduino-cli #873: Request for clarification arduino IDE toolchain selection

jantje, 2020-07-27:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.6.x-package_index.json-format-specification

states:

> Note that the IDE does not use this information to select the toolchain during verify. If you want the IDE to use this specific version you should use the notation {runtime.tools.TOOLNAME-VERSION.path} in the platform.txt.

Based on the hardware specification:
>A {runtime.tools.TOOL_NAME.path} and {runtime.tools.TOOL_NAME-TOOL_VERSION.path} property is generated for the tools of Arduino AVR Boards and any other platform installed via Boards Manager. {runtime.tools.TOOL_NAME.path} points to the latest version of the tool available.

I always understood the arduino IDE selected the tool(chain)? with the highest version number (making the assumption the latest is the one with the highest version number). However testing shows this is  not Arduino IDE behaviour.
 
FI installing sparkfun (https://raw.githubusercontent.com/sparkfun/Arduino_Boards/master/IDE_Board_Manager/package_sparkfun_index.json) apollo boards means installing arm-none-eabi-gcc\\8-2018-q4-major
After installing the sparkfun platform, Arduino zero compiles with arm-none-eabi-gcc\\7-2017q4 (version 7) and not with the sparkfun tool (version  8) .

Can you please specify how arduino IDE selects the tool(chain)?

Note: IMHO the arduino framework specifies/uses platform versions and tool versions but does not specify/use toolchain versions.

ubidefeo, 2020-09-22:
@per1234 
I saw that documentation has been made regarding this task
can we close it?

jantje, 2020-09-22:
I read this
> {runtime.tools.TOOL_NAME.path} points to the latest version of the tool available.

which is IMHO not correct

umbynos, 2022-10-13:
@jantje this issue should be solved by https://github.com/arduino/arduino-cli/pull/1887, can you confirm it?

jantje, 2022-10-13:
This indeed solves the problem of lack of documentation.
However IMHO with the described strategy there will be boards that no longer compile after the change.

>     What is the new behavior?
>     In the situation above where a tool is provided by multiple packages then the compiler will pick in order of priority:
>         the tool provided by the board's package
>         in absence of it, the tool (from a 3rd party) with the greatest version according to semver
>         in case of a tie, the tool from the packager whose name comes first in alphabetic order
>         The last rule is merely a way to make the tool selection fully deterministic even in the most uncommon situation.

closed 2022-10-13 by jantje