arduino/arduino-cli #876: Feature Request: Serial Monitor

sohnryang, 2020-07-27:
A built-in serial monitor in arduino-cli would be great.

ubidefeo, 2020-07-28:
hi @sohnryang 
Being this a command line tool we believe it's up to the user to choose their preferred way of interacting with the serial connection.
If we were to integrated it into the CLI we'd end up putting a tool inside a tool, and this would be something that we're trying to avoid.
CLI is for users who want to extend their workflows and introduce Arduino into them.
On Linux and Mac OS you can use command line tools to connect to serial ports, `screen` is stock in both, I believe, or it can be installed quickly via a package manager.

We welcome this feature request nonetheless, as it will be part of our upcoming FAQ.
I'd suggest you use [Putty](https://www.chiark.greenend.org.uk/~sgtatham/putty/latest.html) on Windows and I personally use [Serial](https://www.decisivetactics.com/products/serial/) on Mac OS (it seems they released a v2 which I look forward to upgrading to).

Anyway if you search the web for "serial communication app" and add your OS you'll get a lot of options.
Again, if your workflow is terminal based, I'd suggest you implement it through `screen`.
I often just pull that `screen /dev/cu.xxxxxxxx 9600` and it gets the job done if I only need to receive data.
You can of course use this tool to send data to your board.
Look up a tutorial, the terminal can let you do so many beautiful and powerful things :)

sohnryang, 2020-07-28:
@ubidefeo I know tools like `screen` can be used as serial monitor. I thought that serial monitor integrated to `arduion-cli` can give users more "Arduino IDE-like" features, such as setting line ending characters on the fly, and showing timestamps in outputs.

ubidefeo, 2020-07-28:
I can see the usefulness for a few users. We'd have to implement cross-platform solutions for this, but we can't yet give way for these features, we've definitely talked about this but we're leaning towards adding this to an FAQ and advising users to choose a terminal communication mean, and if we get a lot of requests we might being it to the roadmap :)

swankun, 2020-08-27:
Might not be the exact replacement for Serial Monitor, but [this is a project](https://hackaday.io/project/5334-serialplot-realtime-plotting-software) that I find very useful. It reads and plots data from your serial port, and also lets you send commands to the board in ASCII or binary format.

sohnryang, 2020-09-04:
Thanks for responding!

closed 2020-09-04 by sohnryang

ubidefeo, 2021-04-06:
@giripriyadarshan 
This would imply having one `arduino-cli` running as a process doing the Serial Monitor and another one doing your compile/upload.
We should make the two processes be aware of each other and it's out of the scope of the CLI.
Offering Serial Monitor is in our roadmap, but you'll have to stop it to continue the operations on the same board anyway

giripriyadarshan, 2021-04-06:
> but you'll have to stop it to continue the operations on the same board anyway

Not really.... currently, Arduino IDE offers a Serial monitor which is stopped by the IDE (without erasing the previous outputs) when we are uploading to the board and continues the Serial monitor when the board restarts(after upload).

> This would imply having one arduino-cli running as a process doing the Serial Monitor and another one doing your compile/upload.

Yes, probably that's what the IDE does too right?

ubidefeo, 2021-04-06:
@giripriyadarshan 
the IDE does some behind-the-scene gymnastics to do that
The CLI is meant to do one operation at the time.
Even IDE 2 has a GUI handling the disconnection of the Serial Monitor when it needs to upload.

ubidefeo, 2021-04-06:
we'll do our best :D

notcos, 2022-07-30:
For anyone looking for a simple solution to handling serial monitoring on Linux automatically, just write a simple bash script. Here is an example that can be used inside of tmux to compile, upload, and automatically handle serial monitoring with screen. It kills and restarts screen inside of the second window of tmux.

```
#!/bin/bash
arduino-cli compile --fqbn espressif:esp32:adafruit_feather_esp32_v2 httpget.ino
pkill screen
arduino-cli upload -p /dev/ttyACM0 --fqbn espressif:esp32:adafruit_feather_esp32_v2 httpget.ino
tmux send-keys -t 2 C-z 'screen /dev/ttyACM0 115200' Enter
```

per1234, 2022-07-30:
Resolved by https://github.com/arduino/arduino-cli/pull/1495