arduino/arduino-cli #880: ERROR detecting boards: error getting board info from Arduino Cloud

happy228, 2020-07-29:
I run arduino-cli in my React app and when I tested app without network i got this message: 
```
"Error: Command failed: .\extra\arduino-cli.exe board list --format json
Error detecting boards: error getting board info from Arduino Cloud"
```
I want my app work without network. How can I download board info and work with it offline?

ubidefeo, 2020-07-29:
hi @happy228 

by when you first run `arduino-cli` you don't have any index related to available boards and cores.
This allows you to always have the most up-to-date cores to download.

Make sure that the first command you launch is `arduino-cli.exe core update-index` followed by `arduino-cli.exe lib update-index`.
[In the next release you'll be able to optimise this workflow with a single command :) ]
Now you can install the platforms you want/need.
`arduino-cli.exe core install arduino:avr`
`arduino-cli.exe core install arduino:samd`
and so on.

please read the README and the getting started carefully, these procedures are all listed in that guide.
u.

happy228, 2020-07-29:
@ubidefeo Thank u very much! Is there some way to make a "portable" version of my app? I mean that every device with my app does not need to install those packages, because I want to run my app on devices without any access to network.

ubidefeo, 2020-07-30:
hi @happy228 

I haven't tried the portable way, but you can pass the configuration file to the command every time you run it (we're in the process of making some enhancements and fixes which might render this unusable but not without an alternative way).

`arduino-cli --config-file your_config_file.yaml {other commands}`

your `.yaml` file can contain paths to the cores and other packages.
You might want to investigate that way.

make a copy of your `arduino-cli.yaml` file and start playing with that :)

rsora, 2020-07-30:
@happy228 we received a similar question here https://github.com/arduino/arduino-cli/issues/519, check it!

Happy Coding!

per1234, 2022-01-27:
I have reopened this because the bug is not resolved.

## To reproduce

1. Disconnect your computer from the Internet.
1. Run the command `arduino-cli board list`

The command fails and no ports are listed:

```
$ arduino-cli version
arduino-cli.exe  Version: nightly-20220125 Commit: 23e7cb5 Date: 2022-01-25T01:29:37Z

$ arduino-cli board list
Error detecting boards: Error getting board info from Arduino Cloud

$ echo $?
5
```

It is reasonable to print a warning that it could not collect the supplemental convenience information by querying Arduino's online database. However, there is no reason this should cause the command to fail to list the ports and the board identifications from the data of the installed platforms.

ubidefeo, 2022-11-14:
@per1234 , @cmaglie 
Upon further examination, I have noticed the following:
if a user has multiple boards connected, one of which with a resolvable `fqbn` via means of installed `boards.txt`, and the other which cannot be resolved unless the CLI is able to query the service, even the normally recognised one will be skipped and CLI will generate an error and even break the JSON output.

I'll be analyzing the following scenario
```
arduino-cli@0.28.0
arduino:mbed_nano@3.4.1
esp32:esp32:esp32@2.0.5
```
Command executed: `arduino-cli board list --format json`

### Internet connection ACTIVE
Single board
- Nano RP2040 Connect

```
[
  {
    "matching_boards": [
      {
        "name": "Arduino Nano RP2040 Connect",
        "fqbn": "arduino:mbed_nano:nanorp2040connect"
      }
    ],
    "port": {
      "address": "/dev/cu.usbmodem2101",
      "label": "/dev/cu.usbmodem2101",
      "protocol": "serial",
      "protocol_label": "Serial Port (USB)",
      "properties": {
        "pid": "0x005E",
        "serialNumber": "416523501C9D8090",
        "vid": "0x2341"
      }
    }
  }
]
```

Single board
- ESP32 module, VID:PID not defined as a pair in a board

```
[
  {
    "matching_boards": [
      {
        "name": "ESP32S2 Native USB",
        "fqbn": "esp32:esp32:esp32s2usb"
      }
    ],
    "port": {
      "address": "/dev/cu.usbmodem03",
      "label": "/dev/cu.usbmodem03",
      "protocol": "serial",
      "protocol_label": "Serial Port (USB)",
      "properties": {
        "pid": "0x0003",
        "serialNumber": "0",
        "vid": "0x303A"
      }
    }
  }
]
```

Two boards
- Nano RP2040 Connect
- ESP32 module, VID:PID not defined as a pair in a board

```
[
  {
    "matching_boards": [
      {
        "name": "Arduino Nano RP2040 Connect",
        "fqbn": "arduino:mbed_nano:nanorp2040connect"
      }
    ],
    "port": {
      "address": "/dev/cu.usbmodem2101",
      "label": "/dev/cu.usbmodem2101",
      "protocol": "serial",
      "protocol_label": "Serial Port (USB)",
      "properties": {
        "pid": "0x005E",
        "serialNumber": "416523501C9D8090",
        "vid": "0x2341"
      }
    }
  },
  {
    "matching_boards": [
      {
        "name": "ESP32S2 Native USB",
        "fqbn": "esp32:esp32:esp32s2usb"
      }
    ],
    "port": {
      "address": "/dev/cu.usbmodem03",
      "label": "/dev/cu.usbmodem03",
      "protocol": "serial",
      "protocol_label": "Serial Port (USB)",
      "properties": {
        "pid": "0x0003",
        "serialNumber": "0",
        "vid": "0x303A"
      }
    }
  }
]
```

### Internet connection INACTIVE
Single board
- Nano RP2040 Connect

```
[
  {
    "matching_boards": [
      {
        "name": "Arduino Nano RP2040 Connect",
        "fqbn": "arduino:mbed_nano:nanorp2040connect"
      }
    ],
    "port": {
      "address": "/dev/cu.usbmodem2101",
      "label": "/dev/cu.usbmodem2101",
      "protocol": "serial",
      "protocol_label": "Serial Port (USB)",
      "properties": {
        "pid": "0x005E",
        "serialNumber": "416523501C9D8090",
        "vid": "0x2341"
      }
    }
  }
]
```

Single board
- ESP32 module, VID:PID not defined as a pair in a board

```
Error detecting boards: Error getting board info from Arduino Cloud
null
```

Two boards
- Nano RP2040 Connect
- ESP32 module, VID:PID not defined as a pair in a board

```
Error detecting boards: Error getting board info from Arduino Cloud
null
```

Besides the Nano RP2040 Connect not being recognised when an unrecognisable board is part of the set,
the JSON formatting is now broken, rendering any chance of parsing impossible

closed 2022-11-21 by cmaglie