arduino/arduino-cli #881: Upload fails with  "ioctl("TIOCMGET"): Invalid argument"

oligu, 2020-07-29:
## Bug Report
Uploading from a Raspberry Pi to an Controllino Maxi Automation fails with following output:

```
pi@raspi:~/my_sketch $ arduino-cli upload -v --log-file log.txt -p /dev/tty --fqbn CONTROLLINO_Boards:avr:controllino_maxi_automation
avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/pi/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf"
         User configuration file is "/home/pi/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/tty
         Using Programmer              : wiring
         Overriding Baud Rate          : 115200
ioctl("TIOCMGET"): Invalid argument
                                   ioctl("TIOCMGET"): Invalid argument
                                                                      vrdude: stk500v2_ReceiveMessage(): timeout
                                                                                                                vrdude: stk500v2_ReceiveMessage(): timeout
                                                                                                                                                          vrdude: stk500v2_ReceiveMessage(): timeout
                      vrdude: stk500v2_ReceiveMessage(): timeout
                                                                vrdude: stk500v2_ReceiveMessage(): timeout
                                                                                                          vrdude: stk500v2_ReceiveMessage(): timeout
                                                                                                                                                    avrdude: stk500v2_getsync(): timeout communicating with programmer

                                        avrdude done.  Thank you.

                                                                 Error during Upload: uploading error: exit status 1 pi@raspi:~/my_sketch $
```
The log file looks as follows:
```
time="2020-07-29T18:09:56+02:00" level=info msg="Checking for config file in: /home/pi/.arduino15/arduino-cli.yaml"
time="2020-07-29T18:09:56+02:00" level=info msg="Checking if CLI is Bundled into the IDE"
time="2020-07-29T18:09:56+02:00" level=info msg="Candidate IDE Directory: /home/pi/bin"
time="2020-07-29T18:09:56+02:00" level=info msg="CLI is not bundled into the IDE"
time="2020-07-29T18:09:56+02:00" level=info msg="Checking for config in: /arduino-cli.yaml"
time="2020-07-29T18:09:56+02:00" level=info msg="error loading" error="open /arduino-cli.yaml: no such file or directory"
time="2020-07-29T18:09:56+02:00" level=info msg="Checking for config in: /home/arduino-cli.yaml"
time="2020-07-29T18:09:56+02:00" level=info msg="error loading" error="open /home/arduino-cli.yaml: no such file or directory"
time="2020-07-29T18:09:56+02:00" level=info msg="Checking for config in: /home/pi/arduino-cli.yaml"
time="2020-07-29T18:09:56+02:00" level=info msg="error loading" error="open /home/pi/arduino-cli.yaml: no such file or directory"
time="2020-07-29T18:09:56+02:00" level=info msg="Checking for config in: /home/pi/my_sketch/arduino-cli.yaml"
time="2020-07-29T18:09:56+02:00" level=info msg="error loading" error="open /home/pi/my_sketch/arduino-cli.yaml: no such file or directory"
time="2020-07-29T18:09:56+02:00" level=info msg="Configuration set"
time="2020-07-29T18:09:56+02:00" level=info msg=arduino-cli-0.5.0
time="2020-07-29T18:09:56+02:00" level=info msg="Starting root command preparation (`arduino`)"
time="2020-07-29T18:09:56+02:00" level=info msg="Formatter set"
time="2020-07-29T18:09:56+02:00" level=info msg="Initializing package manager"
time="2020-07-29T18:09:56+02:00" level=info msg="Checking if CLI is Bundled into the IDE"
time="2020-07-29T18:09:56+02:00" level=info msg="Candidate IDE Directory: /home/pi/bin"
time="2020-07-29T18:09:56+02:00" level=info msg="Loading hardware from: /home/pi/.arduino15/packages"
time="2020-07-29T18:09:56+02:00" level=info msg="Loading package CONTROLLINO from: /home/pi/.arduino15/packages/CONTROLLINO"
time="2020-07-29T18:09:56+02:00" level=info msg="Loading package CONTROLLINO_Boards from: /home/pi/.arduino15/packages/CONTROLLINO_Boards/hardware"
time="2020-07-29T18:09:56+02:00" level=info msg="Loaded platform" platform="CONTROLLINO_Boards:avr@3.1.0"
time="2020-07-29T18:09:56+02:00" level=info msg="Loading package arduino from: /home/pi/.arduino15/packages/arduino/hardware"
time="2020-07-29T18:09:56+02:00" level=info msg="Loaded platform" platform="arduino:avr@1.8.1"
time="2020-07-29T18:09:56+02:00" level=info msg="Checking existence of 'tools' path: /home/pi/.arduino15/packages/arduino/tools"
time="2020-07-29T18:09:56+02:00" level=info msg="Loading tools from dir: /home/pi/.arduino15/packages/arduino/tools"
time="2020-07-29T18:09:56+02:00" level=info msg="Loaded tool" tool="arduino:arduinoOTA@1.3.0"
time="2020-07-29T18:09:56+02:00" level=info msg="Loaded tool" tool="arduino:avr-gcc@7.3.0-atmel3.6.1-arduino5"
time="2020-07-29T18:09:56+02:00" level=info msg="Loaded tool" tool="arduino:avrdude@6.3.0-arduino17"
time="2020-07-29T18:09:56+02:00" level=info msg="Loading package builtin from: /home/pi/.arduino15/packages/builtin"
time="2020-07-29T18:09:56+02:00" level=info msg="Checking existence of 'tools' path: /home/pi/.arduino15/packages/builtin/tools"
time="2020-07-29T18:09:56+02:00" level=info msg="Loading tools from dir: /home/pi/.arduino15/packages/builtin/tools"
time="2020-07-29T18:09:56+02:00" level=info msg="Loaded tool" tool="builtin:ctags@5.8-arduino11"
time="2020-07-29T18:09:56+02:00" level=info msg="Loaded tool" tool="builtin:serial-discovery@1.0.0"
time="2020-07-29T18:09:56+02:00" level=info msg="Adding libraries dir" dir=/home/pi/Arduino/libraries location=sketchbook
time="2020-07-29T18:09:56+02:00" level=info msg="Adding libraries dir" dir=/home/pi/.arduino15/packages/arduino/hardware/avr/1.8.1/libraries location=platform
time="2020-07-29T18:09:57+02:00" level=info msg="Reading sketch from dir: /home/pi/my_sketch"
time="2020-07-29T18:09:57+02:00" level=info msg="Executing `arduino upload`"
time="2020-07-29T18:09:57+02:00" level=info msg="Searching tools required for board CONTROLLINO_Boards:avr:controllino_maxi_automation"
time="2020-07-29T18:10:28+02:00" level=error msg="Error during Upload: uploading error: exit status 1"
```

### Current behavior

Upload fails.

### Expected behavior

The compiled sketch gets uploaded to the Controllino.

### Environment

- CLI version (output of `arduino-cli version`):  0.5.0 Commit: 3be2287
- OS and platform: Raspbian GNU/Linux 10 (buster)

### Additional context

arduino-cli was installed on the RasPi with the `install.sh` file. Then the Arduino IDE was installed on a Windows computer and the Controllino libs installed as [described on their page](https://user-images.githubusercontent.com/3225652/88827578-69fcde00-d1ca-11ea-823e-36ce7d816a2b.png). Then the folders CONTROLLINO and CONTROLLINO_Boards where copied from the Windows PC to the RasPi at ~/.arduino15/packages. Then following command were executed:
`arduino-cli core install CONTROLLINO_Boards:avr`
`arduino-cli lib install "CONTROLLINO"`
Compiling works as expected.

ubidefeo, 2020-07-30:
hi @oligu 

The link to the tutorial on Controllino's website is wrong, points to a PNG.
I believe the issue might simply be with the selected USB Serial port you're targeting.
I doubt it's just `/dev/tty/`.
Please try `arduino-cli board list` once your device is connected and identify the port.

cheers.ubi

oligu, 2020-07-30:
Hi @ubidefeo 

Thanks for the quick reply and pointing out the two hiccups.
I edited the link to the Controllino page in the main post.
And yes, setting the correct port `/dev/ttyACM0/` solved the problem.

Cheers
o/igu

closed 2020-07-30 by oligu