arduino/arduino-cli #898: Q: How to use sketchbook/hardware directory instead of ~/.arduino15/packages/

sej7278, 2020-08-15:
I don't like the way boards manager in the IDE dumps everything into a hidden directory, is there a way to avoid that with arduino-cli?

I tried setting the following in arduino-cli.yaml but it ended up just creating a packages directory where I expected the cores to go:

```
directories:
  data: /home/user/.arduino15
  downloads: /home/user/arduino16/hardware
  user: /home/simon/user/arduino16
```

Is there a way to force the cores to go into the hardware directory inside the user sketchbook instead of ~/.arduino15/packages/arduino/hardware/avr/1.8.3 ?

`arduino-cli core list` seems to list the cores from there, so surely it can install to there too?

Libraries seem to work this way (install/update work fine) but not cores.

ubidefeo, 2020-08-15:
hi @sej7278 
the `data` directory can be set to whatever you like, that's where your core data will be.
try to set it or create a new config with `arduino-cli config init` (this will erase your current config, so watch out. we're working on adding confirmation) and run `arduino-cli core update-index` followed by `arduino-cli lib update-index`

Let me know how it pans out ✌️

ubidefeo, 2020-08-15:
@sej7278 
Also, you cannot use your sketchbook folder as destination of downloaded cores.
I'd like to understand if this is a useful feature.
What do you think, @per1234 ?

sej7278, 2020-08-15:
hi @ubidefeo yes i've looked some more into this and understand now that tools (avr-gcc, avrdude etc.) are also downloaded into ~/.arduino15/packages, that i'm fine with.

but i'd still like a way to have the sketchbook/hardware folder used for the cores (perhaps without the numbering - surely that can be held in a file rather than as a subdirectory name?) to make it consistent with the "old" way of doing things and also the libraries which are downloaded to sketchbook/libraries/ using arduino-cli

i'm a fan of Makefiles (as a contributor to arduino-mk) so made this simple wrapper for arduino-cli, which i think is much easier than passing in all of the flags every time you run arduino-cli:

```make
FQBN = ATTinyCore:avr:attinyx5

all:
	arduino-cli compile --fqbn $(FQBN) $(notdir $(CURDIR)).ino

upload:
	arduino-cli upload -b $(FQBN) -P usbasp $(CURDIR)

clean:
	rm -rf $(CURDIR)/build
```

although as i recall with arduino-builder you could make a local json file to hold the config, can't seem to find the documentation for that now.

finally, does burn-bootloader set the fuses for cores that don't have bootloaders like attiny85? otherwise how do we set the fusebits for blank chips using this?

ubidefeo, 2020-08-15:
>but i'd still like a way to have the sketchbook/hardware folder used for the cores (perhaps without the numbering - surely that can be held in a file rather than as a subdirectory name?) to make it consistent with the "old" way of doing things
if you mean the core version that really seems to work for most cases. Not sure we'll ad extra configuration files just for this

>and also the libraries which are downloaded to sketchbook/libraries/ using arduino-cli
you may have a point here, but since also the Java IDE downloads cores and tools to this package folder we'll stick to that.

There are other `arduino-cli` users who implement it using makefiles, and I remember checking out _Arduino Makefile_ a while back. I like that approach :) 

>although as i recall with arduino-builder you could make a local json file to hold the config, can't seem to find the documentation for that now.
I'm not familiar with this configuration file, but we support a local `sketch.json` file that allows you to store board and port.
Try using the command `arduino-cli board attach -h` 

`burn-bootlaoder` operates based on how the core settings are configured. I believe if you create a board with burn bootloader directives that configure fuses but do not write a bootloader to the flash it should just work.
@per1234 knows way more about these things than I do :)

sej7278, 2020-08-15:
ok thanks, think i'll stick to git cloning cores to my hardware directory then, or maybe using arduino-cli and moving the files it downloads. wonder if symlinking would work - might mess up with those versioned directories though.

ubidefeo, 2020-08-15:
if it suits you :D 

the advantage of keeping it stock is that you'll be able to get updates via `arduino-cli` as well as retaining previous versions without having to use git :)

sej7278, 2020-08-15:
Yeah I get that, but the directory location and versioning breaks too many if my supporting scripts, I'll close this, thanks for the input.

closed 2020-08-15 by sej7278

per1234, 2020-08-15:
>you cannot use your sketchbook folder as destination of downloaded cores.

@ubidefeo what is the issue that prevents using the sketchbook folder as destination of downloaded cores? I just did a quick test with both `directories.user` and `directories.data` set to the same path and I didn't have any problems.

>does burn-bootloader set the fuses for cores that don't have bootloaders like attiny85? otherwise how do we set the fusebits for blank chips using this?

@sej7278 it does whatever the the author of the boards core configured it to do. Just as in the Arduino IDE, Arduino CLI doesn't impose any restriction in this matter.

For example, here's how the classic `attiny:avr` boards core is configured:
https://github.com/damellis/attiny/blob/6bba7d452af59d5190025bc870ec9e53d170e4d9/platform.txt
`arduino-cli burn-bootloader` executes two actions in sequence:
1. `erase`
1. `bootloader`

`erase` is traditionally used to do a chip erase (`-e`), put the lock fuse bits in the unlock state, and set the other fuse bits according to the board definition. You can see that the `attiny:avr` boards core doesn't touch the lock fuse, but does erase and set the other fuses.

`bootloader` is traditionally used to flash the bootloader and then put the lock fuse bits in the lock state. The `attiny:avr` boards core has no need to do any of that, so it simply runs an avrdude command that verifies the chip signature.

In the case of the more advanced `ATTinyCore:avr` boards core, some of the boards do use bootloaders, so it is configured to flash a bootloader file (but note they do it in the `erase` action in order to provide compatibility with certain programmers that don't like the two commands in quick succession):
https://github.com/SpenceKonde/ATTinyCore/blob/8faf2e3436114e51917536b56656581dbca94abb/avr/platform.txt#L127
A dummy bootloader file is used for the boards that don't use a bootloader:
https://github.com/SpenceKonde/ATTinyCore/blob/8faf2e3436114e51917536b56656581dbca94abb/avr/boards.txt#L309

So you can see the Arduino boards core system is flexible enough that the boards core authors are able to do amazing things!

ubidefeo, 2020-08-16:
@per1234 
>what is the issue that prevents using the sketchbook folder as destination of downloaded cores? I just did a quick test with both directories.user and directories.data set to the same path and I didn't have any problems.

I believe if you only use CLI you might have no issues, but if you also need to run the Java IDE and/or the Pro IDE then they'd source and store data in different places and there would be no consistency.
Am I getting it wrong?

matthijskooijman, 2020-08-16:
IIRC the Java IDE uses checks like "is this path inside the sketchbook" and "is this path inside my user directory" to retroactively decide where a core was loaded from. Not ideal, but works most of the time.

However, in the Java IDE, there is no way to change the user directory (it is always `~/arduino15` on Linux, unless you have a portable install), so I guess this is really a moot point here.

sej7278, 2020-08-16:
Yeah I can live with it, just been changing an awful lot of makefiles and scripts lol.

One thing I found odd was that the ide couldn't find the cores I'd installed until I added the additional URLs into it too, it didn't read the cli config yaml.