arduino/arduino-cli #899: Unable to include any other compile paths using --libraries

colerd24, 2020-08-15:
## Bug Report

### Current behavior

arduino-cli fails to include a path for additional source files using the --libraries argument

Trying to include test.cpp/.h in a subdirectory in the sketch_name folder:
```
/sketch_name

    /subdir

        test.cpp

        test.h

    sketch_name.ino
```

sketch_name.ino includes "test.h" and calls a function that is defined in test.cpp and has a prototype in test.h

Run the following command while in the sketch_name directory:
`arduino-cli compile --fqbn arduino:avr:pro --libraries /path/to/sketch_name/subdir sketch_name`

The result is:
`fatal error: test.h: No such file or directory`

I tried using quotes around the path, adding '/' after subdir in the path, and using absolute/relative paths. I also created a library.properties file in the subdir. In all cases the error is always the same

### Expected behavior

Sucessful compilation - test.h is found.

### Environment

- CLI version: 0.11.0-arch Commit: 0296f4df116385f868b67c5ffa7393936c3345c9
- OS and platform: Manjaro Linux kernel 5.4.52-1

### Additional context

N/A

Thanks for the assistance! 
[sketch_name.zip](https://github.com/arduino/arduino-cli/files/5079358/sketch_name.zip)

per1234, 2020-08-15:
Hi @colerd24. The path provided to the `--libraries` option must be to a folder that contains libraries, not to a library itself. This is similar to how the `libraries` subfolder of the user directory can not be a library itself, but instead must contain libraries each in their own folder, as defined by the[ library specification](https://arduino.github.io/arduino-cli/latest/library-specification/#layout-of-folders-and-files).

So if you change your sketch structure to this:
```
sketch_name
|_ sketch_name.ino
|_ subdir
    |_ test
        |_ test.cpp
        |_ test.h
```
You'll find that your command now works as expected.

colerd24, 2020-08-16:
@per1234 Thank you so much! I can't believe I got that mixed up, works great now.

closed 2020-08-16 by colerd24