arduino/arduino-cli #920: Version 0.12.0 does not accept json string "version": "2.0a4". Version 0.11.0 did accept it.

ArminJo, 2020-08-24:
## Bug Report

### Current behavior

https://github.com/ArminJo/Arduino-OpenWindowAlarm/runs/1023797774?check_suite_focus=true#step:4:42

arduino-cli core update-index --additional-urls http://drazzy.com/package_drazzy.com_index.json > /dev/null
Error updating index: invalid handle
arduino-cli core install ATTinyCore:avr --additional-urls http://drazzy.com/package_drazzy.com_index.json > /dev/null
Error installing: updating the core index: invalid package index in http://drazzy.com/package_drazzy.com_index.json: invalid patch version separator 'a'
The json contains
```
"tools": [
        {
          "name": "micronucleus",
          "version": "2.0a4",
          "systems": [
```

### Expected behavior

as in Version 0.11.0 https://github.com/ArminJo/Arduino-OpenWindowAlarm/runs/716439332?check_suite_focus=true#step:4:38

### Environment

- arduino-cli Version: 0.12.0 Commit: d68544e
- OS and platform: LINUX

ubidefeo, 2020-08-24:
this is an interesting one, 
thank you for reporting :)

ArminJo, 2020-08-24:
It is maybe not the string "version": "2.0a4", as I assumed, since the string worked for another [build ](https://github.com/ArminJo/Arduino-OpenWindowAlarm/runs/716439323?check_suite_focus=true#step:4:38) where it is also included in the [json](https://raw.githubusercontent.com/ArminJo/DigistumpArduino/master/package_digistump_index.json).

per1234, 2020-08-24:
The error is caused by the use of a different non-semver version. It's the platform version rather than the tool version from the snippet you posted:
https://github.com/SpenceKonde/ReleaseScripts/blob/354f8584e8523ed82f1766cd33be97e4ed8809cd/package_drazzy.com_index.json#L906
```json
          "name": "DxCore",
          "architecture": "megaavr",
          "version": "1.1.0a",
          "category": "Contributed",
```
In my tests, I didn't find the non-semver tool version caused any problems, though I certainly wouldn't recommend it.

---
>Version 0.11.0 did accept it.
The non-semver platform version is also not supported by Arduino CLI 0.11.0, so this is not a regression:
```
per@HAL MINGW64 /e/ProgramFiles/ArduinoIDE/arduino-cli-0.11.0
$ ./arduino-cli core update-index
Updating index: package_index.json downloaded/ 308.74 KiB   28.19% 00m01s
Updating index: package_drazzy.com_index.json downloaded7 KiB    0.00%
Error updating index: invalid package index in https://raw.githubusercontent.com/SpenceKonde/ReleaseScripts/354f8584e8523ed82f1766cd33be97e4ed8809cd/package_drazzy.com_index.json: invalid patch version separator 'a'
```

---
The error in the package index has already been fixed: https://github.com/SpenceKonde/ReleaseScripts/pull/34.

---
Perhaps we can consider making Arduino CLI resilient to this sort of thing. It's fine that non-semver versions are unsupported, but perhaps it would be better to make it just display a warning and otherwise ignore the definitions for the bad platform releases in the package index, rather than having everything come to a screeching halt if you have a package index with a bad version somewhere in it.

That appears to be the behavior of the Arduino IDE. When I tried this with the Arduino IDE, the only impact I could see from this issue is that DxCore 1.1.0a is not offered in the Boards Manager version menu. The other releases from the package index can still be used and it causes no problems for general operation.

---
Here's the package index as it was before the fix, to use for reproducing the issue:
https://raw.githubusercontent.com/SpenceKonde/ReleaseScripts/354f8584e8523ed82f1766cd33be97e4ed8809cd/package_drazzy.com_index.json

closed 2020-08-25 by ArminJo