arduino/arduino-cli #928: Clarification is required on the duplicate Library and LibraryRelease properties.

kittaakos, 2020-08-28:
The following properties are available on both the `LibraryRelease` and the `Library` objects. Is this something you’re planning to change later, or the gRPC API is like that by design. If it’s by design, why, and what does it solve?
 - `author`,
 - `maintainer`,
 - `sentence`,
 - `paragraph`,
 - `website`,
 - `category`,
 - `architectures`, and
 - `types`.

It does not cause any errors, no need to change it right now (if you want to change it at all), but as a gRPC API consumer, I want to understand the why.

Thank you!

----

https://github.com/arduino/arduino-cli/blob/3c16c05b0a240fc2b8b68dcebd6895be0935c44d/rpc/commands/lib.proto#L135-L155

https://github.com/arduino/arduino-cli/blob/3c16c05b0a240fc2b8b68dcebd6895be0935c44d/rpc/commands/lib.proto#L213-L232

cmaglie, 2020-09-01:
* `LibraryRelease` contains the metadata that comes from the `library_index.json`, so it maps the index 1:1 and it's basically the result of the `search` when you're looking for a not-installed library.
* `Library` contains the metadata that are extracted by scanning the installed libraries and looking into their `library.properties`. An installed library may not be present in the libraries index (because it may have been manually installed for example).

Some fields overlap (because they are both present in the index and in the library.properties) some other not (for example the URL to download the library is present only in `LibaryRelease`, or vice-versa for the `install_dir` is present only in the `Library`).

So:
`LibraryRelease` -> a library release in the library_index
`Library` -> an installed library in the user sketchbook

Now, for completeness let's see some operations on the libraries (I've omitted the parts that are not relevant):

```
message LibrarySearchResp {
    repeated SearchedLibrary libraries = 1;
}

message SearchedLibrary {
    // Library name.
    string name = 1;
    // The index data for the available versions of the library. The key of the
    // map is the library version.
    map<string, LibraryRelease> releases = 2;
    // The index data for the latest version of the library.
    LibraryRelease latest = 3;
}
```

The `search` operation returns all the release of a library as a `map<version, LibraryRelease>` because these metadata are obtained from the index (since the libraries are not installed yet). It also returns a `LibraryRelease` with the latest version available.

```
message LibraryListResp {
    repeated InstalledLibrary installed_library = 1;
}

message InstalledLibrary {
    // Information about the library.
    Library library = 1;
    // When the `updatable` field of the `LibraryList` request is set to `true`,
    // this will contain information on the latest version of the library in the
    // libraries index.
    LibraryRelease release = 2;
}
```

the `list` operation instead returns information about the installed libraries. So the structure `InstalledLibrary` contains a `Library` but... if an update of the library is available it also links a `LibraryRelease` (because the update is not installed but comes from the index!).

Hope this clarifies things a bit!

kittaakos, 2020-09-02:
Thank you! Unfortunately, I still cannot acquire the `provides_includes` from the available set of CLI commands.

When I run `lib search`, `provides_includes` is available:
```
% ./arduino-cli lib search "AllThingsTalk LoRaWAN SDK" --format json
{
  "libraries": [
    {
      "name": "AllThingsTalk LoRaWAN SDK",
      "releases": {
        "3.0.0": {
          "author": "AllThingsTalk",
          "version": "3.0.0",
          "maintainer": "Alain Cuypers \u003calain@allthingstalk.com\u003e",
          "sentence": "LoRaWAN SDK for AllThingsTalk Maker.",
          "paragraph": "Compatible with Microchip RN2483 and RN2903 devices.",
          "website": "https://github.com/allthingstalk/arduino-lorawan-sdk",
          "category": "Communication",
          "architectures": [
            "*"
          ],
          "types": [
            "Contributed"
          ],
          "resources": {
            "url": "http://downloads.arduino.cc/libraries/github.com/allthingstalk/AllThingsTalk_LoRaWAN_SDK-3.0.0.zip",
            "archivefilename": "AllThingsTalk_LoRaWAN_SDK-3.0.0.zip",
            "checksum": "SHA-256:b33d17c95e1b4c73cd39c7dab62ca97a8a31fbba37f36d95e4a03d7204aad7b7",
            "size": 52001,
            "cachepath": "libraries"
          },
          "provides_includes": [
            "ABPCredentials.h",
            "LoRaModem.h",
            "CborPayload.h"
          ]
        },
        "3.1.0": {
          "author": "AllThingsTalk",
          "version": "3.1.0",
          "maintainer": "Alain Cuypers \u003calain@allthingstalk.com\u003e",
          "sentence": "LoRaWAN SDK for AllThingsTalk Maker",
          "paragraph": "Compatible with Microchip RN2483 and RN2903 devices.",
          "website": "https://github.com/allthingstalk/arduino-lorawan-sdk",
          "category": "Communication",
          "architectures": [
            "*"
          ],
          "types": [
            "Contributed"
          ],
          "resources": {
            "url": "http://downloads.arduino.cc/libraries/github.com/allthingstalk/AllThingsTalk_LoRaWAN_SDK-3.1.0.zip",
            "archivefilename": "AllThingsTalk_LoRaWAN_SDK-3.1.0.zip",
            "checksum": "SHA-256:6c76df191f5189a8b80826bde0880dba97a6e3d27861cc8533ffce51f400ea48",
            "size": 484781,
            "cachepath": "libraries"
          },
          "provides_includes": [
            "AllThingsTalk_LoRaWAN.h"
          ]
        },
        "3.1.2": {
          "author": "AllThingsTalk",
          "version": "3.1.2",
          "maintainer": "Vanja \u003cvanja@allthingstalk.com\u003e",
          "sentence": "LoRaWAN SDK for AllThingsTalk Maker",
          "paragraph": "Compatible with Microchip RN2483 and RN2903 devices.",
          "website": "https://github.com/allthingstalk/arduino-lorawan-sdk",
          "category": "Communication",
[... 186 lines cut ...]
        "/Users/akos.kitta/Documents/Arduino/libraries/AllThingsTalk_LoRaWAN_SDK/examples/Rapid Development Kit/guard-your-stuff",
        "/Users/akos.kitta/Documents/Arduino/libraries/AllThingsTalk_LoRaWAN_SDK/examples/Rapid Development Kit/pushbutton",
        "/Users/akos.kitta/Documents/Arduino/libraries/AllThingsTalk_LoRaWAN_SDK/examples/SDK Examples/BinaryPayloads",
        "/Users/akos.kitta/Documents/Arduino/libraries/AllThingsTalk_LoRaWAN_SDK/examples/SDK Examples/GPSCBOR",
        "/Users/akos.kitta/Documents/Arduino/libraries/AllThingsTalk_LoRaWAN_SDK/examples/SDK Examples/HelloLora",
        "/Users/akos.kitta/Documents/Arduino/libraries/AllThingsTalk_LoRaWAN_SDK/examples/SDK Examples/IotDataPoint",
        "/Users/akos.kitta/Documents/Arduino/libraries/AllThingsTalk_LoRaWAN_SDK/examples/SDK Examples/LoraOptions",
        "/Users/akos.kitta/Documents/Arduino/libraries/AllThingsTalk_LoRaWAN_SDK/examples/SDK Examples/OTAARetry",
        "/Users/akos.kitta/Documents/Arduino/libraries/AllThingsTalk_LoRaWAN_SDK/examples/SDK Examples/instrumentation"
      ]
    },
    "release": {}
  }
```

Ouch, `release` is not there. But no worries, the [documentation](https://arduino.github.io/arduino-cli/latest/rpc/commands/#cc.arduino.cli.commands.InstalledLibrary) of the `InstalledLibrary` says:
> When the `updatable` field of the `LibraryList` request is set to `true`, this will contain information on the latest version of the library in the libraries index.

Let's force the `updatable` flag to access the `release` property:
```
% ./arduino-cli lib list AllThingsTalk_LoRaWAN_SDK --updatable --format json
[]%                                 
```

Ouch, the lib is already up to date, so I cannot get this info other than running `lib search` and `lib list` together and make some string matching based on the `name` of the library. Or is it the `real_name` I should use? (https://github.com/arduino/arduino-cli/issues/932)

kittaakos, 2020-09-02:
> running `lib search` and `lib list` together and make some string matching based on the `name` of the library

It works for user libraries, but it does not work for libs bundled with a core, for instance for `EEPROM`. `EEPROM` is not listed with `lib search`. Can you please recommend a workaround for this limitation, @cmaglie?

ubidefeo, 2020-09-02:
@kittaakos 
you have the `--all` flag to add libraries in cores to the list

kittaakos, 2020-09-02:
> @kittaakos
> you have the `--all` flag to add libraries in cores to the list

It's about `lib search`, @ubidefeo and not `lib list`.

`lib search` does not support `--all`:
```
% ./arduino-cli lib search EEPROM --all
Error: unknown flag: --all
```

cmaglie, 2020-09-02:
Do you need includes for the installed libraries?

kittaakos, 2020-09-02:
> Do you need includes for the installed libraries?

I am not sure if I understood it, but I want to clone this feature for the Pro IDE. I need the includes for user libs and libs packaged with cores.

[image]

kittaakos, 2020-09-04:
> Hope this clarifies things a bit!

Yes, thank you!

The rest was addressed as part of https://github.com/arduino/arduino-cli/pull/934.

closed 2020-09-04 by kittaakos