arduino/arduino-cli #929: Cannot upload on Arduino MEGA2560 R3

Hexer10, 2020-08-29:
## Bug Report

### Current behavior

This command: `arduino-cli upload arduino_.ino --port COM6 --fqbn "arduino:avr:mega:cpu=atmega2560"`
fails with the given error:
```
Error during Upload: incorrect FQBN: platform arduino:avr is not installed
```
### Expected behavior

Upload the code to Arduno MEGA2560

### Environment

- CLI version: 0.12.1 Commit: 48383da
- OS and platform: Windows 10

### Additional context

The FQBN should be right since the Arduino IDE uses that during the build (see -fqbn=arduino:avr:mega:cpu=atmega2560):
```
D:\Tools (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\Tools (x86)\Arduino\hardware -hardware C:\Users\Mattia\AppData\Local\Arduino15\packages -tools D:\Tools (x86)\Arduino\tools-builder -tools D:\Tools (x86)\Arduino\hardware\tools\avr -tools C:\Users\Mattia\AppData\Local\Arduino15\packages -built-in-libraries D:\Tools (x86)\Arduino\libraries -libraries D:\Users\Mattia\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -vid-pid=2341_0042 -ide-version=10813 -build-path C:\Users\Mattia\AppData\Local\Temp\arduino_build_70659 -warnings=none -build-cache C:\Users\Mattia\AppData\Local\Temp\arduino_cache_188058 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=D:\Tools (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=D:\Tools (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=D:\Tools (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=D:\Tools (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=D:\Tools (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=D:\Tools (x86)\Arduino\hardware\tools\avr -verbose D:\Users\Mattia\Documents\ClionProjects\arduino_\arduino_.ino

```

per1234, 2020-08-29:
Hi @Hexer10. Unlike the Arduino IDE, Arduino CLI doesn't come with any boards cores pre-installed. So, even with the Arduino AVR Boards core of your Mega that you are used to always having available with the Arduino IDE, you must install it first.

You can do that using the [`arduino-cli core install` command](https://arduino.github.io/arduino-cli/latest/commands/arduino-cli_core_install/):
```
arduino-cli core install arduino:avr
```

closed 2020-08-29 by per1234

Hexer10, 2020-08-29:
@per1234 Thanks for the quick reply! That solved my issue but still I cannot make it to upload my file, here is the commandline output:

```
D:\Users\Mattia\Documents\ClionProjects\arduino_test>arduino-cli compile arduino_test.ino --fqbn arduino:avr:mega:cpu=atmega2560
Sketch uses 16608 bytes (6%) of program storage space. Maximum is 253952 bytes.
Global variables use 688 bytes (8%) of dynamic memory, leaving 7504 bytes for local variables. Maximum is 8192 bytes.

D:\Users\Mattia\Documents\ClionProjects\arduino_test>arduino-cli upload arduino_test.ino --port COM6 --fqbn arduino:avr:mega:cpu=atmega2560 --input-dir build/arduino.avr.mega
avrdude: can't open input file build/arduino.avr.mega/..ino.hex: No such file or directory
avrdude: read from file 'build/arduino.avr.mega/..ino.hex' failed
Error during Upload: uploading error: uploading error: exit status 1
````

(I've tried with and without --input-dir)

per1234, 2020-08-30:
The problem is caused by running the command from inside the sketch folder and then also specifying the sketch argument to the command. When you run an `arduino-cli` command from inside the sketch folder, the sketch argument is optional. So your command can be simplified to:
```
arduino-cli upload --port COM6 --fqbn arduino:avr:mega:cpu=atmega2560 --input-dir build/arduino.avr.mega
```
which will work.

However, usage of your original command should be supported, so I would consider this is a bug. We have a previous report of this here: https://github.com/arduino/arduino-cli/issues/630#issuecomment-678694060, and a related report here: https://github.com/arduino/arduino-cli/issues/691, but we don't have a dedicated issue for this. Would you mind opening a dedicated issue for that bug?

---
FYI, it doesn't cause any problems, but the `--input-dir` option is only necessary when you want a custom input directory. In this case, you are using the build directory created by the standard `arduino-cli compile` command, so Arduino CLI automatically looks there and there is no need for you to add the `--input-dir` option.

ubidefeo, 2020-08-30:
@per1234 
I spoke to @cmaglie about the behaviour of `--input-dir` and I ended up making a mistake using it.
I believe its behaviour can be confusing, maybe a bit more documentation should be used to specify that when using it you are still required to either be in the Sketch folder or pass the Sketch as an input parameter.
I believe this also ties to `--input-file` for upload in some way

per1234, 2020-08-30:
>I believe its behaviour can be confusing, maybe a bit more documentation should be used to specify that when using it you are still required to either be in the Sketch folder or pass the Sketch as an input parameter.

I think that's a good idea. Although I think the option works just as it should, It's a bit unintuitive that the input file is determined by combining the sketch name argument with the path specified by the `--upload-dir` option. A little extra documentation in the command help would clear up any confusion someone might have.

However, just to avoid any confusion for the people reading this, I should note that the bug reported in https://github.com/arduino/arduino-cli/issues/929#issuecomment-683353999 is not related to the use of `--input-dir`. The same problem occurs even without that option in the command. It's purely about running an `arduino-cli upload` command from the sketch folder and using either the sketch filename or `.` for the sketch name argument. For example:
```
D:\Users\Mattia\Documents\ClionProjects\arduino_test>arduino-cli upload arduino_test.ino --port COM6 --fqbn arduino:avr:mega:cpu=atmega2560
```
or:
```
D:\Users\Mattia\Documents\ClionProjects\arduino_test>arduino-cli upload . --port COM6 --fqbn arduino:avr:mega:cpu=atmega2560
```

---
>I believe this also ties to --input-file for upload in some way

I just tried https://github.com/arduino/arduino-cli/pull/777 and it seems that you can specify any any filename in any path, as I'd expect:
```
per@HAL MINGW64 ~/go/src/github.com/arduino/arduino-cli ((1df9f2db...))
$ ./arduino-cli upload -p COM6 -b arduino:avr:mega --input-file /tmp/bar.hex
```

ubidefeo, 2020-09-22:
@per1234 
any chance we can close this one?
`--input-dir` has been refactored completely 2 weeks ago, any other reasons to keep this open?