arduino/arduino-cli #932: `name` vs `real_name` when executing a lib commands

kittaakos, 2020-09-02:
## Bug Report

 - The [`name`](https://github.com/arduino/arduino-cli/blob/ef57e49b6d6273653a051cfdc55bd9884f0d935a/rpc/commands/lib.proto#L220) of a library is the name of the folder in which the lib is contained: `{directories.user}/Arduino/${name}`
 - The [`real_name`](https://github.com/arduino/arduino-cli/blob/ef57e49b6d6273653a051cfdc55bd9884f0d935a/rpc/commands/lib.proto#L249) of a library is the human-readable name of the lib. It comes from the `library.properties`.

This might be the desired behavior but it's strange. When I do a `lib search` or `lib install`  the `real_name` is required, when I run a `lib list`,  the `name` is required.

### Current behavior

`lib search` works with `real_name`, not with `name`:
```
% ./arduino-cli lib search  AllThingsTalk_LoRaWAN_SDK               
No libraries matching your search.
Did you mean...
AllThingsTalk LoRaWAN SDK

% ./arduino-cli lib search "AllThingsTalk LoRaWAN SDK" 
Name: "AllThingsTalk LoRaWAN SDK"
  Author: AllThingsTalk
  Maintainer: Vanja <vanja@allthingstalk.com>
  Sentence: LoRaWAN SDK for AllThingsTalk Maker
  Paragraph: Compatible with Microchip RN2483 and RN2903 devices.
  Website: https://github.com/allthingstalk/arduino-lorawan-sdk
  Category: Communication
  Architecture: *
  Types: Contributed
  Versions: [3.0.0, 3.1.0, 3.1.2, 3.1.3, 3.1.4, 3.1.5, 3.1.6]
  Provides includes: AllThingsTalk_LoRaWAN.h

```

`lib install` works with `real_name` but does not work with `name`:
```
% ./arduino-cli lib install AllThingsTalk_LoRaWAN_SDK 
Error resolving dependencies for AllThingsTalk_LoRaWAN_SDK: looking for library: library AllThingsTalk_LoRaWAN_SDK not found
% ./arduino-cli lib install "AllThingsTalk LoRaWAN SDK"
AllThingsTalk LoRaWAN SDK depends on AllThingsTalk LoRaWAN SDK@3.1.6
Downloading AllThingsTalk LoRaWAN SDK@3.1.6...
AllThingsTalk LoRaWAN SDK@3.1.6 already downloaded
Installing AllThingsTalk LoRaWAN SDK@3.1.6...
Already installed AllThingsTalk LoRaWAN SDK@3.1.6
```

`lib list` works with `name`, but does not work with `real_name`:
```
% ./arduino-cli lib list AllThingsTalk_LoRaWAN_SDK
Name                      Installed     Available         Location Description                                                
AllThingsTalk_LoRaWAN_SDK 3.1.6         -                 user     LoRaWAN SDK for AllThingsTalk Maker                        

% ./arduino-cli lib list "AllThingsTalk LoRaWAN SDK"
No libraries installed.
```

I think this is a bug.

### Expected behavior

All lib commands work either with both `name` and `real_name` or all require either the `name` or `real_name`.

### Environment

- CLI version (output of `arduino-cli version`): `arduino-cli Version: 0.13.0-rc1 Commit: ef57e49`
- OS and platform: macOS Catalina (`10.15.6 (19G2021)`)

### Additional context

per1234, 2022-05-12:
I ended up taking a close look at this situation while reviewing https://github.com/arduino/arduino-cli/pull/1725. Here is a report on my findings:

## Recommendations

### Always use canonical library name

The library name is used as the sole unique identifier by the Arduino Library Manager system.

This name must be used whenever referring to the library in Arduino CLI output.

This name must be used whenever referring to the library in arguments to `arduino-cli` commands.

### Record the canonical name when loading libraries

Data on each installed library is gathered [during initialization](https://github.com/arduino/arduino-cli/blob/b6d36c6d3d515cfe445f81ede3eaf1e6e2c6e853/arduino/libraries/loader.go#L30).

Since there is some logic involved in determining the canonical name, it must be recorded at this time to avoid the need for code duplication by each consumer of the data:

- If the library has [a `library.properties` metadata file](https://arduino.github.io/arduino-cli/dev/library-specification/#libraryproperties-file-format) (i.e., ["1.5 format"](https://arduino.github.io/arduino-cli/dev/library-specification/#15-library-format-rev-22)), the library name is defined by the `name` field of `library.properties`.
- If the library does not have a `library.properties` metadata file (i.e., ["1.0" or "legacy" format](https://arduino.github.io/arduino-cli/dev/library-specification/#old-library-format-pre-15)), the library name is defined by the folder name.

### Redesign  [`github.com/arduino/arduino-cli/arduino/libraries.Library`](https://github.com/arduino/arduino-cli/blob/b6d36c6d3d515cfe445f81ede3eaf1e6e2c6e853/arduino/libraries/libraries.go#L53)

I believe the design of this struct is the source of some of the current problems, and will be the source of others in the future:

- `Name`: This is actually [the base library folder name](https://github.com/arduino/arduino-cli/blob/b6d36c6d3d515cfe445f81ede3eaf1e6e2c6e853/arduino/libraries/loader.go#L111), so only the canonical library name for outdated "1.0 format" libraries
- `RealName`: This is [the value of the `name` field of `library.properties`](https://github.com/arduino/arduino-cli/blob/b6d36c6d3d515cfe445f81ede3eaf1e6e2c6e853/arduino/libraries/loader.go#L112), so only the canonical library name for "1.5 format" libraries

So neither of these currently provides the canonical name in all cases.

The data they contain is useful in its own right. For example, they are both priority factors in [library dependency resolution](https://arduino.github.io/arduino-cli/dev/sketch-build-process/#dependency-resolution).

This means that an additional field must be added to store the canonical name. That solves the technical problem, but it will only make the human problem of the unintuitive struct design even worse. So the existing fields must be changed to have appropriate names. This will be a fairly significant change, so it must be done during the current development phase where breaking changes are already commonplace.

### Update the code for the new struct design

Examples:

- [`github.com/arduino/arduino-cli/arduino/libraries.Library.String`](https://github.com/arduino/arduino-cli/blob/b6d36c6d3d515cfe445f81ede3eaf1e6e2c6e853/arduino/libraries/libraries.go#L86)
- [High level canonicalization code in `github.com/arduino/arduino-cli/cli/lib/installedResult.String`](https://github.com/arduino/arduino-cli/blob/b6d36c6d3d515cfe445f81ede3eaf1e6e2c6e853/cli/lib/list.go#L132-L135)

## Examples of problems caused by the current system:

- https://github.com/arduino/arduino-cli/issues/1041
- https://github.com/arduino/arduino-cli/issues/1577
- https://github.com/arduino/arduino-cli/issues/1731

These issues also affect Arduino IDE and Arduino Web Editor.

## Demos

```text
$ arduino-cli version
arduino-cli.exe  Version: nightly-20220512 Commit: b6d36c6 Date: 2022-05-12T01:37:29Z
```

### `compile`

```text
$ SKETCH_PATH="/tmp/LibNameBug"

$ mkdir "$SKETCH_PATH"

$ echo "#include <ArduinoRobotMotorBoard.h>
void setup() {}
void loop() {}
" > "$SKETCH_PATH/LibNameBug.ino"

$ arduino-cli compile -b arduino:megaavr:nona4809 -v "$SKETCH_PATH"

[... 66 lines cut ...]

$ arduino-cli lib uninstall "Robot_Motor"
Uninstalling Robot_Motor@1.0.3...
```

🐛 `lib uninstall` command accepts incorrect library names, but equivalent command `lib install` does not.

### `lib upgrade`

```
$ arduino-cli lib install "Robot Motor"@1.0.2
Downloading Robot Motor@1.0.2...
Robot Motor@1.0.2 already downloaded
Installing Robot Motor@1.0.2...
Installed Robot Motor@1.0.2

$ arduino-cli lib upgrade
Downloading Robot Motor@1.0.3...
Robot Motor@1.0.3 already downloaded
Installing Robot Motor@1.0.3...
Replacing Robot_Motor@1.0.2 with Robot Motor@1.0.3...
Installed Robot Motor@1.0.3
```

🐛 Library is referred to incorrectly as "Robot_Motor", and inconsistently as well since it also referred to the library correctly as "Robot Motor" in the same sentence.

cmaglie, 2022-09-22:
Fixed by #1878

closed 2022-09-22 by cmaglie