arduino/arduino-cli #933: Arduino nano "ATmega328P (Old Bootloader)" Board Not Found

PhysWiza, 2020-09-02:
## Bug Report

### Current behavior

arduino-cli board list

No board FQBN or core is printed on terminal as if no board is connected. Although an arduino nano is plugged.

### Expected behavior

The board should be detected and FQBN and core information should be printed with the corrosponding port. 

### Environment

- CLI version : 0.10.0 
- OS and platform: Linux Ubuntu

### Additional context

When using the Arduino GUI, after selecting the arduino nano bord there is an option under tools-->Processor-->ATmega328P (Old Bootloader)

after selecting this option the GUI will succeed in uploading sketches. Otherwise I recive a can't open device error

per1234, 2020-09-02:
Hi @PhysWiza. We have provided an explanation about this situation in the FAQ:
https://arduino.github.io/arduino-cli/latest/FAQ/#arduino-unomegaduemilanove-is-not-detected
>## Arduino Uno/Mega/Duemilanove is not detected
>When you run arduino-cli board list, your board doesn't show up. Possible causes:
>
>...
>
>It mounts a USB2Serial converter like FT232 or CH340: these chips always report the same USB VID/PID to the operating system, so the only thing we know is that the board mounts that specific USB2Serial chip, but we don’t know which board that chip is on.

Although the Nano is not mentioned, the same applies to it as well. All Nanos use a general purpose USB to TTL serial adapter chip. The board detection system uses the VID/PID presented by the board, but these chips use the VID/PID provided by the manufacturer. So there is no way to know whether it's a Nano, another Arduino board, or even a non-Arduino device on the other side of that chip. So unfortunately your request is impossible to grant.

closed 2020-09-02 by per1234

PhysWiza, 2020-10-01:
Does this mean that I can't upload the sketch to my nano with Arduino-cli?
In the youtube tutorial he uses a nano. 

https://www.youtube.com/watch?v=J-qGn1eEidA&t=196s

per1234, 2020-10-01:
>Does this mean that I can't upload the sketch to my nano with Arduino-cli?

No. You can use your Nano with Arduino CLI, no problem. If you want help, just make a post on the Arduino Forum's dedicated board for Arduino CLI and I'll be happy to help you out:
https://forum.arduino.cc/index.php?board=145.0

>In the youtube tutorial he uses a nano.

He uses a Nano 33 IoT. That is a different situation from the classic Nano. The Nano 33 IoT has a unique VID/PID, so it is able to be identified by `arduino-cli board list`.

PhysWiza, 2020-10-03:
Yes that is the thread I created. Wow that reply was amazing. Thank you very much. So clear, concise and had everything I needed to know.