arduino/arduino-cli #950: Add option to flag libraries as system libraries with -isystem to disable warnings

cjoach, 2020-09-15:
I would like to disable all warnings in the libraries I'm using. Is it possible to add an option to flag libraries with `-isystem` instead of `-I`?

ubidefeo, 2020-09-16:
@cjoach 
can you be more specific?
This is not something I understand the need for.
In Arduino CLI you won't manually use `-I` for a path to include, that is in case a platform design implementation, and might require cherry-picking which ones you want to be included as such.
Not sure we have that level of granularity at this moment.

@per1234 @cmaglie this is more of your realm, what's your take?

matthijskooijman, 2020-09-16:
I'm assuming that when an include path is specified with `-isystem` rather than `-I`, the compiler will suppress warnings generated from files included through those paths.

Most things on the compiler commandline are decided by `platform.txt`, but I think the `-I` option used for library include paths is actually one of the few things that is hardcoded in (and thus decided by) arduino-cli. So I think we *could* implement this flag, but I'm not so sure we should.

I'd rather see *less* hardcoding in arduino-cli than more of it. Also, there might be [other effects](https://gcc.gnu.org/onlinedocs/cpp/System-Headers.html) of using `-isystem` rather than `-I` (such as changing the priority of the include) and at the very least it's conceptually incorrect: Your preferences about warnings do not suddenly change a library from or to a system library.

Also, implementing this would not actually prevent warnings in libraries entirely: When compiling .cpp files from the library itself, warnings will still be shown. If that library uses double-quote includes (e.g. `#include "foo.h"`) in its .cpp files, I suspect that the `-isystem` does not kick in, and warnings from the library's header files will also be displayed.

This request relates to #684 and https://github.com/arduino/Arduino/issues/4184 which are about enabling warnings by default, but also suggest a related change to enable/disable warnings for the sketch, core and libraries separately (which is essentially the same motivation driving this request, I think). Reading this request, I do realize that the previously suggested approach to this (e.g. compile library files with different warning options than sketch files) is actually still flawed: Warnings in library header files included by sketch files would still be shown. Using `-isystem` as suggested here could maybe fix this for libraries, but only on/off (no control about the warning level) and not for the core files, for which `-I` is encoded in `platform.txt` I believe.

As a local workaround, @cjoach could try the `#pragma GCC diagnostic` to disable warnings before including library headers, or `#pramga GCC system_header` inside a library file to mark it as a system header. See http://dbp-consulting.com/tutorials/SuppressingGCCWarnings.html for some examples.

ubidefeo, 2020-09-16:
Thank you @matthijskooijman for the exhaustive and informative answer :)

cjoach, 2020-09-28:
I had problems with multiple pragma GCC diagnostic. I was not able to disable flags for the cpp files as @matthijskooijman mentioned. I have control only on the .h files. This indeed makes my problem more complicated.

My request boils down to having a way to activate warnings for MY code only since I have essentially no control over others libraries and especially not the core libs. Simply not adding the warning flags to any file that is not part of the sketch and passing any include outside of the sketch as -isystem would resolve this problem (for me at least).

I understand that this issue needs to be adressed properly for all cases but I think having a warning_flags for sketch, libs and core would be a sensible choice and has the potential to not change the default behavior. And I could disable warnings in .h files using the GCC diagnostic pragma until something better is implemented.