arduino/arduino-cli #951: custom "Arduino15" path

harryberlin, 2020-09-16:
## Feature Request

### Current behavior

the Arduino15 path is placed in "%USERPATH%\AppData\Local\" and is used by CLI and IDE

### Expected behavior

add option portable folder to place the Arduino15 files there, like IDE portable function
or add option to place Arduino15 in the same path, where arduino-cli.exe exist

### Environment

- CLI version (output of `arduino-cli version`): 0.10.0 Commit: ec5c3ed
- OS and platform: Windows 10 64bit

### Additional context

so it's possilbe to use the CLI standalone

matthijskooijman, 2020-09-16:
I believe that arduino-cli indeed usally loads its config file from the Arduino15 path, but inside that file you can specify a different path for storing e.g. the downloaded cores. Did you see https://arduino.github.io/arduino-cli/latest/configuration/ ? Maybe that would solve your need?

Furthermore, you can even change the location of the configfile using the `--config-file` option (might not be terribly convenient).

Lastly, `arduino-cli` will try to read a config file from the current directory, or its parents, which might also be helpful (though that feature is somewhat controversial and might be removed or modified in the future, see #758).

harryberlin, 2020-09-16:
Do you have an command line example to use the custom data path for compile?

in my case: `arduino-cli.exe compile -b arduino:avr:nano -o Release\AVR_IBus_NANO C:\Release`

matthijskooijman, 2020-09-16:
You would have to specify a custom data path in the `arduino-cli.yaml` file, e.g. something like:

```yaml
directories:
  data: /home/user/arduino-cli-data
  downloads: /home/user/arduino-cli-data/staging
  user: /home/user/arduino-cli-sketchbook
```

The put this file in your `Arduino15` directory, or elsewhere and pass its filename to `--config-file`.

Does that help?

harryberlin, 2020-09-16:
i have a complete copy of the Arduino15 path, where the arduino-cli.exe exist.

yaml file ist this and placed in arduino-cli.exe:
`
directories:
  data: /Arduino15
`

commandline ist this:
`
arduino-cli.exe --config-file arduino-cli.yaml compile -b arduino:avr:nano -o Release\AVR_IBus_NANO C:\Sketchpath
`

but it's not working.

matthijskooijman, 2020-09-16:
I'm not sure how this works exactly on Windows, but on Linux starting a path with a `/` makes it absolute:

```
data: /Arduino15
```

Maybe that should contain a full path to the Arduino15 directory, or maybe just remove the leading `/` would be sufficient to make it relative to the current directory (which could still be fragile, unless arduino-cli would interpret it relative to the config file itself, not sure if it does).

harryberlin, 2020-09-16:
same result without /
that's why, the best solution would be a "portable" option

per1234, 2021-02-03:
Closing because Arduino CLI already provides the capability to specify a custom `directories.data` as Matthijs explained.

closed 2021-02-03 by per1234