arduino/arduino-cli #955: Please review the exit code of the `lib list -b` command

kittaakos, 2020-09-17:
## Bug Report

This is a minor thing, maybe not even an issue but the expected behavior. When I run `lib list -b` command without any libraries **and** any cores installed, I always get a `0` exit code. However, when I have at least one user library installed, but still no cores, I always get a non-zero, `1` exit code. This feels inconsistent.

### Current behavior

```
% rm -rf ~/Library/Arduino15 && rm -rf ~/Documents/Arduino/libraries
% ./arduino-cli version
arduino-cli Version: 0.13.0 Commit: 693a045
% ./arduino-cli core update-index
Downloading missing tool builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 downloaded                                                                                                                          
Installing builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 installed
Downloading missing tool builtin:serial-discovery@1.0.0...
builtin:serial-discovery@1.0.0 downloaded                                                                                                                       
Installing builtin:serial-discovery@1.0.0...
builtin:serial-discovery@1.0.0 installed
Updating index: library_index.json downloaded                                                                                                                   
Updating index: package_index.json downloaded                                                                                                                   
Updating index: package_index.json.sig downloaded                                                                                                               
Updating index: package_index.json downloaded                                                                                                                   
Updating index: package_index.json.sig downloaded                                                                                                               
% ./arduino-cli core list        
% ./arduino-cli lib list -b arduino:samd:mkr1000                    
No libraries installed.
% echo $?                                                           
0
% ./arduino-cli lib install "AllThingsTalk LoRaWAN SDK"             
AllThingsTalk LoRaWAN SDK depends on AllThingsTalk LoRaWAN SDK@3.1.6
Downloading AllThingsTalk LoRaWAN SDK@3.1.6...
AllThingsTalk LoRaWAN SDK@3.1.6 downloaded                                                                                                                      
Installing AllThingsTalk LoRaWAN SDK@3.1.6...
Installed AllThingsTalk LoRaWAN SDK@3.1.6
% ./arduino-cli lib list -b arduino:samd:mkr1000       
Error listing Libraries: loading board data: platform arduino:samd is not installed
% echo $?                                              
1
% 
```

### Expected behavior

I would expect to have `1` exit code with a similar error message (`platform x:y is not installed`) when no user libraries **and** no cores are installed.

### Environment

- CLI version (output of `arduino-cli version`): `arduino-cli Version: 0.13.0 Commit: 693a045`
- OS and platform: macOS Catalina (`10.15.6 (19G2021)`)

### Additional context

If this is the expected behavior, please close the issue with a remark. Thank you!

ubidefeo, 2020-09-17:
I believe the issue lies in the sequence by which errors are handled.
An empty list of libraries returns before entering the filter phase.
Maybe in this case we should first address the filter and handle its absence in case.

ubidefeo, 2020-09-20:
found one more bug on the `lib list -b FQBN`:
libraries which are not compatible with the core are listed.
to replicate:
- install Arduboy
- `arduino-cli lib list -b arduino:samd:mkrwifi1010`

the library `arduiboyTones` is only compatible with AVR, but is still listed

PR #978 (2020-09-29): Libraries compatibility is now shown when calling lib list with fqbn
**Please check if the PR fulfills these requirements**

- [x] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-cli/pulls)
      before creating one)
- [x] The PR follows
      [our contributing guidelines](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/#pull-requests)
- [x] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

* **What kind of change does this PR introduce?**
Changes the b [... cut]
closed 2020-09-29 by silvanocerza