arduino/arduino-cli #965: IDE-Friendly Compile Error Paths from Local Header Files

ecorm, 2020-09-22:
## Bug Report

When `arduino-cli compile` prints compile errors from local header files, it displays the path from the temporary `/tmp/arduino-sketch-FOO` directory of the sketch being compiled. By "local header files", I mean header files that live alongside the `foo.ino` sketch file, as well as those under the `src` subdirectory.

My IDE (QtCreator) parses compiler errors and displays them in a user-friendly way. When I click on these parsed error messages, it conveniently takes me to the file and line that produced the error.

Unfortunately, arduino-cli defeats the convenience of these parsed error messages when they originate from local header files. When I click on them, it takes me to the header file that's been copied to the `/tmp/arduino-sketch-FOO` directory instead of the original header file directory.

### Current behavior

```
arduino-ide compile
...
In file included from /home/me/myproject/myproject.ino:10:0:
/tmp/arduino-sketch-5AA431FD41D5B5DF00FF21C5123C1125/sketch/header.h:100:5: error: 'printeff' was not declared in this scope
```

### Expected behavior

I would expect `arduino-ide compile` to substitute `/tmp/arduino-sketch-5AA431FD41D5B5DF00FF21C5123C1125/sketch/header.h` in the error message with the original path of the header file: `/home/me/myproject/header.h`

### Environment

arduino-cli Version: 0.13.0 Commit: 693a045
Linux Mint 20 Cinnamon

matthijskooijman, 2020-09-24:
Note that the Java IDE already has an output post-processing step that does pretty much this, though I couldn't find the code quickly just now (I think it's in a bit of an obscure place in the code, or maybe not even in the Java code but in Arduino-builder?).

cmaglie, 2021-03-17:
Fixed by #1224

closed 2021-03-17 by cmaglie