arduino/arduino-cli #967: upload does not treat -P list as an instruction to list programmers

PeterWone, 2020-09-23:
## Bug Report

### Current behavior

```
arduino-cli upload --fqbn arduino:avr:nano --programmer list
```

Fails to treat `list` as a special symbol.

### Expected behavior

The documentation says it will list available programmers.

### Environment

- CLI version (output of `arduino-cli version`): 0.13.0
- OS and platform: Windows 10

### Additional context

`arduino-cli burn-bootloader --fqbn arduino:avr:nano --programmer list` works as advertised. 

The capability to get a list of supported programmers is available from burn-bootloader, perhaps update the documentation to suggest that.

ubidefeo, 2020-09-23:
hi @PeterWone 
because this syntax was pretty cumbersome and programmers are part of platform definition and related to a specific board, we introduced a new flag for `boards detail` to just show programmers

`arduino-cli board details -b arduino:avr:uno --list-programmers`

the fact that `burn-bootloader` still has that is a leftover that we'll have to cleanup :) (@silvanocerza )

PeterWone, 2020-09-24:
In that case this issue is actually about the docs being behind the reality. Thanks for sharing the reality!

ubidefeo, 2020-09-24:
hey @PeterWone 
you are right, we need to update the documentation.
Than you for pointing it out, some things end up slipping through the cracks :)

per1234, 2020-10-22:
The documentation has now been corrected by https://github.com/arduino/arduino-cli/pull/982.

Thanks PeterWone!

closed 2020-10-22 by per1234