arduino/arduino-cli #973: More than one custom library get stuck arduino-cli in endless loop + SIGSEGV

xplsek01, 2020-09-23:
## Bug Report

### Current behavior

Cause infinite loop:
arduino-cli compile --libraries libraries/Lib1,libraries/Lib2 -b esp8266:esp8266:nodemcu:CpuFrequency=80,VTable=heap,FlashSize=4M1M,LwIPVariant=v2mss1460,FlashErase=none,UploadSpeed=115200 --verbose

The same for:
arduino-cli compile --libraries libraries/Lib1 --libraries libraries/Lib2 -b esp8266:esp8266:nodemcu:CpuFrequency=80,VTable=heap,FlashSize=4M1M,LwIPVariant=v2mss1460,FlashErase=none,UploadSpeed=115200 --verbose

SIGSEGV crash:
arduino-cli compile --libraries libraries/Lib1, libraries/Lib2 -b esp8266:esp8266:nodemcu:CpuFrequency=80,VTable=heap,FlashSize=4M1M,LwIPVariant=v2mss1460,FlashErase=none,UploadSpeed=115200 --verbose

### Expected behavior
Compilation successful. No crash.

### Environment

- CLI version:  0.13.0
- Linux Ubutnu (Gitlab-CI)

ubidefeo, 2020-09-23:
@xplsek01 
This is an interesting bug.
Funny enough if you only have one library it compiles successfully

Mind that in order to reproduce I've tried to change the target board with an Arduino Uno.
Also your parameters don't seem to be recognised by the latest ESP8266 platform, I had to rename most of them
`arduino-cli compile --libraries=libraries/Lib1 -b esp8266:esp8266:nodemcu:xtal=80,vt=heap,eesz=4M1M,wipe=none,baud=115200 --verbose`

this will of course complain for the absence of `lib2.h` which will be resolved by removing the `#include` directive

I also had another issue with your test sketch because you omit `void` before `setup()` and `loop()` but once fixed it compiles correctly when only including the first library and removing a reference to `Lib2` into the command line.

Thank you, we'll investigage

xplsek01, 2020-10-26:
@ubidefeo Can I ask you when do you plan to make a fix? Sorry, that I urge you but my other activities depends on it. If I should start to think about long-term workaround, or to wait. Thank you very much.

ubidefeo, 2020-10-28:
@xplsek01 
sorry for the late reply, we are having an incredibly busy week at Arduino.
@silvanocerza has found the source of the bug, and we have it planned for a fix in the next sprint beginning next week.
I guess it will be in the nightly in about a week or two, I'll make a note to notify you when a build is ready so you can test your specific case 👍

PR #1066 (2020-11-17): Fix compile command stuck in a loop when including multiple custom libraries
**Please check if the PR fulfills these requirements**

- [x] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-cli/pulls)
      before creating one)
- [x] The PR follows
      [our contributing guidelines](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/#pull-requests)
- [x] Tests for the changes have been added (for bug fixes / features)
- [x] Docs have been added / updated (for bug fixes / features)

* **What kind of change does this PR introduce?**

Fixe [... cut]
closed 2020-11-17 by silvanocerza