arduino/arduino-cli #975: Installing STM32duino gives Runtime: out of memory on Raspberry Pi 3 model B

ekawahyu, 2020-09-24:
Arduino-cli 0.13.0 on Raspberry Pi 3 model B

JSON file:
https://github.com/SnapBloks/BoardManagerFiles/blob/feature/tools-for-raspberrypi-host/STM32/package_stm_index.json

JSON is still for testing on STM32Tools v1.4.0 (no ARM support yet) + gcc-arm-none-eabi v9.3.1-1.2 (with ARM support)

I am expecting to see that the installation to go smooth, and then I can start building the proper STM32Tools with ARM support.

Here is what I got when trying to install STM32:stm32, it basically says Out of Memory when installing (or unpacking) gcc:

```
pi@raspberrypi:~ $ ./bin/arduino-cli core install STM32:stm32
Downloading packages...
STM32:xpack-arm-none-eabi-gcc@9.3.1-1.2 already downloaded
STM32:STM32Tools@1.4.0 already downloaded
STM32:CMSIS@5.5.1 already downloaded
STM32:stm32@2.0.0-dev already downloaded
Installing STM32:xpack-arm-none-eabi-gcc@9.3.1-1.2...
fatal error: runtime: out of memory

runtime stack:
runtime.throw(0x954e45, 0x16)
        /usr/local/go/src/runtime/panic.go:774 +0x5c
runtime.sysMap(0x40800000, 0x2000000, 0x106d938)
        /usr/local/go/src/runtime/mem_linux.go:169 +0xa8
runtime.(*mheap).sysAlloc(0x10604d0, 0x2000000, 0x1, 0x4813c)
        /usr/local/go/src/runtime/malloc.go:701 +0x180
runtime.(*mheap).grow(0x10604d0, 0x1000, 0xffffffff)
        /usr/local/go/src/runtime/mheap.go:1252 +0x2c
runtime.(*mheap).allocSpanLocked(0x10604d0, 0x1000, 0x106d948, 0x35128)
        /usr/local/go/src/runtime/mheap.go:1163 +0x290
runtime.(*mheap).alloc_m(0x10604d0, 0x1000, 0x64140101, 0x5ad3f8c0)
        /usr/local/go/src/runtime/mheap.go:1015 +0xd0
runtime.(*mheap).alloc.func1()
        /usr/local/go/src/runtime/mheap.go:1086 +0x3c
runtime.(*mheap).alloc(0x10604d0, 0x1000, 0x10101, 0x283a000)
        /usr/local/go/src/runtime/mheap.go:1085 +0x60
runtime.largeAlloc(0x1fffe00, 0x2830101, 0x6392e3e0)
        /usr/local/go/src/runtime/malloc.go:1138 +0x74
runtime.mallocgc.func1()
        /usr/local/go/src/runtime/malloc.go:1033 +0x38
runtime.systemstack(0x0)
        /usr/local/go/src/runtime/asm_arm.s:354 +0x84
runtime.mstart()
        /usr/local/go/src/runtime/proc.go:1146

goroutine 1 [running]:
runtime.systemstack_switch()
        /usr/local/go/src/runtime/asm_arm.s:298 +0x4 fp=0x46114d8 sp=0x46114d4 pc=0x6aaac
runtime.mallocgc(0x1fffe00, 0x853e48, 0x1, 0x0)
        /usr/local/go/src/runtime/malloc.go:1032 +0x8a0 fp=0x461153c sp=0x46114d8 pc=0x1cc84
runtime.makeslice(0x853e48, 0x1fffe00, 0x1fffe00, 0x3c86af0)
        /usr/local/go/src/runtime/slice.go:49 +0x6c fp=0x4611550 sp=0x461153c pc=0x56910
bytes.makeSlice(0x1fffe00, 0x0, 0x0, 0x0)
        /usr/local/go/src/bytes/buffer.go:229 +0x60 fp=0x4611584 sp=0x4611550 pc=0xf0b4c
bytes.(*Buffer).grow(0x4d4c520, 0x200, 0x1600)
        /usr/local/go/src/bytes/buffer.go:142 +0x138 fp=0x46115a8 sp=0x4611584 pc=0xf04e4
bytes.(*Buffer).ReadFrom(0x4d4c520, 0xa9dc48, 0x3c86b00, 0x63922888, 0x4d4c520, 0x1, 0x29)
        /usr/local/go/src/bytes/buffer.go:202 +0x48 fp=0x46115e4 sp=0x46115a8 pc=0xf0948
io.copyBuffer(0xa9ca60, 0x4d4c520, 0xa9dc48, 0x3c86b00, 0x0, 0x0, 0x0, 0x20, 0x8b1ff8, 0x1, ...)
        /usr/local/go/src/io/io.go:388 +0x2fc fp=0x4611624 sp=0x46115e4 pc=0x7a230
io.Copy(...)
        /usr/local/go/src/io/io.go:364
github.com/codeclysm/extract/v3.copyCancel(0xaa7d00, 0x4d12860, 0xa9ca60, 0x4d4c520, 0xa9c988, 0x28c4240, 0x0, 0x0, 0x0, 0x0)
        /github/home/go/pkg/mod/github.com/codeclysm/extract/v3@v3.0.1/cancelable_reader.go:10 +0x94 fp=0x4611654 sp=0x4611624 pc=0x37cbd4
github.com/codeclysm/extract/v3.(*Extractor).Tar(0x4611900, 0xaa7d00, 0x4d12860, 0xa9e740, 0x4ce9320, 0x4970e10, 0x29, 0x0, 0x94f562, 0x11)
        /github/home/go/pkg/mod/github.com/codeclysm/extract/v3@v3.0.1/extractor.go:148 +0x2d0 fp=0x4611764 sp=0x4611654 pc=0x37d884
github.com/codeclysm/extract/v3.(*Extractor).Gz(0x4611900, 0xaa7d00, 0x4d12860, 0xa9e740, 0x4ce9300, 0x4970e10, 0x29, 0x0, 0x94e549, 0x10)
        /github/home/go/pkg/mod/github.com/codeclysm/extract/v3@v3.0.1/extractor.go:91 +0x15c fp=0x46117c0 sp=0x4611764 pc=0x37d538
github.com/codeclysm/extract/v3.(*Extractor).Archive(0x4611900, 0xaa7d00, 0x4d12860, 0xa9e740, 0x4ce9300, 0x4970e10, 0x29, 0x0, 0x1cf10, 0x4c11a18)
        /github/home/go/pkg/mod/github.com/codeclysm/extract/v3@v3.0.1/extractor.go:46 +0x1a8 fp=0x461182c sp=0x46117c0 pc=0x37d00c
github.com/codeclysm/extract/v3.Archive(...)
        /github/home/go/pkg/mod/github.com/codeclysm/extract/v3@v3.0.1/extract.go:50
github.com/arduino/arduino-cli/arduino/resources.(*DownloadResource).Install(0x2a56f60, 0x2a47488, 0x2a47698, 0x4c11a18, 0x0, 0x0)
        /__w/arduino-cli/arduino-cli/arduino/resources/install.go:59 +0x3b0 fp=0x4611910 sp=0x461182c pc=0x381ccc
github.com/arduino/arduino-cli/arduino/cores/packagemanager.(*PackageManager).InstallTool(0x2a80520, 0x2a80740, 0xb, 0x4970db0)
        /__w/arduino-cli/arduino-cli/arduino/cores/packagemanager/install_uninstall.go:118 +0xf4 fp=0x4611960 sp=0x4611910 pc=0x4c84c8
github.com/arduino/arduino-cli/commands.InstallToolRelease(0x2a80520, 0x2a80740, 0x4c11660, 0x0, 0x0)
        /__w/arduino-cli/arduino-cli/commands/bundled_tools.go:51 +0x258 fp=0x4611a80 sp=0x4611960 pc=0x6281dc
github.com/arduino/arduino-cli/commands/core.installPlatform(0x2a80520, 0x2bdac80, 0x4ce9160, 0x3, 0x4, 0x4ce9140, 0x4c11660, 0x0, 0x8, 0x87a1b8)
        /__w/arduino-cli/arduino-cli/commands/core/install.go:99 +0x408 fp=0x4611e04 sp=0x4611a80 pc=0x75dec4
github.com/arduino/arduino-cli/commands/core.PlatformInstall(0xaa7d20, 0x2896010, 0x2916080, 0x4ce9140, 0x4c11660, 0x1, 0x2, 0x0)
        /__w/arduino-cli/arduino-cli/commands/core/install.go:52 +0x210 fp=0x4611e54 sp=0x4611e04 pc=0x75d940
github.com/arduino/arduino-cli/cli/core.runInstallCommand(0x2a73080, 0x2a46fc8, 0x1, 0x1)
        /__w/arduino-cli/arduino-cli/cli/core/install.go:105 +0x294 fp=0x4611ec0 sp=0x4611e54 pc=0x761f5c
github.com/spf13/cobra.(*Command).execute(0x2a73080, 0x2a46fb8, 0x1, 0x1, 0x2a73080, 0x2a46fb8)
        /github/home/go/pkg/mod/github.com/spf13/cobra@v1.0.1-0.20200710201246-675ae5f5a98c/command.go:846 +0x1f4 fp=0x4611f28 sp=0x4611ec0 pc=0x6dd07c
github.com/spf13/cobra.(*Command).ExecuteC(0x28c3b80, 0x0, 0x0, 0x28000e0)
        /github/home/go/pkg/mod/github.com/spf13/cobra@v1.0.1-0.20200710201246-675ae5f5a98c/command.go:950 +0x26c fp=0x4611f90 sp=0x4611f28 pc=0x6dda1c
github.com/spf13/cobra.(*Command).Execute(...)
        /github/home/go/pkg/mod/github.com/spf13/cobra@v1.0.1-0.20200710201246-675ae5f5a98c/command.go:887
main.main()
        /__w/arduino-cli/arduino-cli/main.go:31 +0x24 fp=0x4611fa4 sp=0x4611f90 pc=0x7b41f4
runtime.main()
        /usr/local/go/src/runtime/proc.go:203 +0x208 fp=0x4611fe4 sp=0x4611fa4 pc=0x41e5c
runtime.goexit()
        /usr/local/go/src/runtime/asm_arm.s:868 +0x4 fp=0x4611fe4 sp=0x4611fe4 pc=0x6c8f8

goroutine 19 [syscall]:
os/signal.signal_recv(0x0)
        /usr/local/go/src/runtime/sigqueue.go:147 +0x130
os/signal.loop()
        /usr/local/go/src/os/signal/signal_unix.go:23 +0x14
created by os/signal.init.0
        /usr/local/go/src/os/signal/signal_unix.go:29 +0x30

goroutine 5 [select]:
go.bug.st/cleanup.InterruptableContext.func1(0xaa7d00, 0x4d12860, 0x4c11a40)
        /github/home/go/pkg/mod/go.bug.st/cleanup@v1.0.0/context.go:21 +0xe0
created by go.bug.st/cleanup.InterruptableContext
        /github/home/go/pkg/mod/go.bug.st/cleanup@v1.0.0/context.go:18 +0x50
```

@ubidefeo I know that you made it working on Pi Zero W, but I don't seem to be able to install STM32duino with Arduino-cli v0.10.0 to v0.13.0. So this seems to me as really run out of memory? How much memory does it need to install 3rd party hardware? Thank you.

ubidefeo, 2020-09-24:
@ekawahyu 
I have to say that after that time I haven't been able to install some cores on the Zero, this out of memory message has been haunting me and I gave up, but never had it on the 3 or 4.
I'll have to test again. In the meantime, can you add a `-v` flag and give me a more verbose description of the failure point?

ekawahyu, 2020-09-24:
Verbose output:

```
pi@raspberrypi:~ $ ./bin/arduino-cli -v core install STM32:stm32
INFO[0000] Using config file: /home/pi/.arduino15/arduino-cli.yaml 
INFO[0000] arduino-cli version 0.13.0                   
INFO[0000] Checking signature                            index=/home/pi/.arduino15/package_index.json signatureFile=/home/pi/.arduino15/package_index.json.sig trusted=true
INFO[0000] Checking signature                            error="opening signature file: open /home/pi/.arduino15/package_stm_index.json.sig: no such file or directory" index=/home/pi/.arduino15/package_stm_index.json signatureFile=/home/pi/.arduino15/package_stm_index.json.sig
INFO[0000] Checking if CLI is Bundled into the IDE      
INFO[0000] Loading hardware from: /home/pi/.arduino15/packages 
INFO[0000] Loading package builtin from: /home/pi/.arduino15/packages/builtin 
INFO[0000] Checking existence of 'tools' path: /home/pi/.arduino15/packages/builtin/tools 
INFO[0000] Loading tools from dir: /home/pi/.arduino15/packages/builtin/tools 
INFO[0000] Loaded tool                                   tool="builtin:ctags@5.8-arduino11"
INFO[0000] Loaded tool                                   tool="builtin:serial-discovery@1.0.0"
INFO[0000] Adding libraries dir                          dir=/home/pi/Arduino/libraries location=user
INFO[0003] Executing `arduino core install`             
INFO[0003] Running from console, will run post-install by default 
Downloading packages...
STM32:xpack-arm-none-eabi-gcc@9.3.1-1.2 already downloaded
STM32:STM32Tools@1.4.0 already downloaded
STM32:CMSIS@5.5.1 already downloaded
STM32:stm32@2.0.0-dev already downloaded
INFO[0016] Installing tool                               Tool="STM32:xpack-arm-none-eabi-gcc@9.3.1-1.2"
Installing STM32:xpack-arm-none-eabi-gcc@9.3.1-1.2...
fatal error: runtime: out of memory

runtime stack:
runtime.throw(0x954e45, 0x16)
        /usr/local/go/src/runtime/panic.go:774 +0x5c
runtime.sysMap(0x3fc00000, 0x2000000, 0x106d938)
        /usr/local/go/src/runtime/mem_linux.go:169 +0xa8
runtime.(*mheap).sysAlloc(0x10604d0, 0x2000000, 0x1, 0x4813c)
        /usr/local/go/src/runtime/malloc.go:701 +0x180
runtime.(*mheap).grow(0x10604d0, 0x1000, 0xffffffff)
        /usr/local/go/src/runtime/mheap.go:1252 +0x2c
runtime.(*mheap).allocSpanLocked(0x10604d0, 0x1000, 0x106d948, 0x35128)
        /usr/local/go/src/runtime/mheap.go:1163 +0x290
runtime.(*mheap).alloc_m(0x10604d0, 0x1000, 0x62ef0101, 0x94a44e92)
        /usr/local/go/src/runtime/mheap.go:1015 +0xd0
runtime.(*mheap).alloc.func1()
        /usr/local/go/src/runtime/mheap.go:1086 +0x3c
runtime.(*mheap).alloc(0x10604d0, 0x1000, 0x3010101, 0x638e5778)
        /usr/local/go/src/runtime/mheap.go:1085 +0x60
runtime.largeAlloc(0x1fffe00, 0x60101, 0x639ceb20)
        /usr/local/go/src/runtime/malloc.go:1138 +0x74
runtime.mallocgc.func1()
        /usr/local/go/src/runtime/malloc.go:1033 +0x38
runtime.systemstack(0x0)
        /usr/local/go/src/runtime/asm_arm.s:354 +0x84
runtime.mstart()
        /usr/local/go/src/runtime/proc.go:1146

goroutine 1 [running]:
runtime.systemstack_switch()
        /usr/local/go/src/runtime/asm_arm.s:298 +0x4 fp=0x5df14d8 sp=0x5df14d4 pc=0x6aaac
runtime.mallocgc(0x1fffe00, 0x853e48, 0x1, 0x0)
        /usr/local/go/src/runtime/malloc.go:1032 +0x8a0 fp=0x5df153c sp=0x5df14d8 pc=0x1cc84
runtime.makeslice(0x853e48, 0x1fffe00, 0x1fffe00, 0x4be02e0)
        /usr/local/go/src/runtime/slice.go:49 +0x6c fp=0x5df1550 sp=0x5df153c pc=0x56910
bytes.makeSlice(0x1fffe00, 0x0, 0x0, 0x0)
        /usr/local/go/src/bytes/buffer.go:229 +0x60 fp=0x5df1584 sp=0x5df1550 pc=0xf0b4c
bytes.(*Buffer).grow(0x55560a0, 0x200, 0x1600)
        /usr/local/go/src/bytes/buffer.go:142 +0x138 fp=0x5df15a8 sp=0x5df1584 pc=0xf04e4
bytes.(*Buffer).ReadFrom(0x55560a0, 0xa9dc48, 0x4be02f0, 0x638e1eb8, 0x55560a0, 0x1, 0x5)
        /usr/local/go/src/bytes/buffer.go:202 +0x48 fp=0x5df15e4 sp=0x5df15a8 pc=0xf0948
io.copyBuffer(0xa9ca60, 0x55560a0, 0xa9dc48, 0x4be02f0, 0x0, 0x0, 0x0, 0x20, 0x8b1ff8, 0x1, ...)
        /usr/local/go/src/io/io.go:388 +0x2fc fp=0x5df1624 sp=0x5df15e4 pc=0x7a230
io.Copy(...)
        /usr/local/go/src/io/io.go:364
[... 24 lines cut ...]
github.com/spf13/cobra.(*Command).ExecuteC(0x30cde40, 0x0, 0x0, 0x30000e0)
        /github/home/go/pkg/mod/github.com/spf13/cobra@v1.0.1-0.20200710201246-675ae5f5a98c/command.go:950 +0x26c fp=0x5df1f90 sp=0x5df1f28 pc=0x6dda1c
github.com/spf13/cobra.(*Command).Execute(...)
        /github/home/go/pkg/mod/github.com/spf13/cobra@v1.0.1-0.20200710201246-675ae5f5a98c/command.go:887
main.main()
        /__w/arduino-cli/arduino-cli/main.go:31 +0x24 fp=0x5df1fa4 sp=0x5df1f90 pc=0x7b41f4
runtime.main()
        /usr/local/go/src/runtime/proc.go:203 +0x208 fp=0x5df1fe4 sp=0x5df1fa4 pc=0x41e5c
runtime.goexit()
        /usr/local/go/src/runtime/asm_arm.s:868 +0x4 fp=0x5df1fe4 sp=0x5df1fe4 pc=0x6c8f8

goroutine 19 [syscall]:
os/signal.signal_recv(0x0)
        /usr/local/go/src/runtime/sigqueue.go:147 +0x130
os/signal.loop()
        /usr/local/go/src/os/signal/signal_unix.go:23 +0x14
created by os/signal.init.0
        /usr/local/go/src/os/signal/signal_unix.go:29 +0x30

goroutine 7 [select]:
go.bug.st/cleanup.InterruptableContext.func1(0xaa7d00, 0x5512840, 0x5421960)
        /github/home/go/pkg/mod/go.bug.st/cleanup@v1.0.0/context.go:21 +0xe0
created by go.bug.st/cleanup.InterruptableContext
        /github/home/go/pkg/mod/go.bug.st/cleanup@v1.0.0/context.go:18 +0x50
```

ubidefeo, 2020-09-24:
@ekawahyu are you running Raspian or Raspberry Pi OS 64 bit?
care to drop the output of `lsb_release -a`? (install `lsb_release` if you don't have it but you might)

Your Pi should have 1GB of RAM, which should be enough to decompress and install the XPack release of `arm-none-eabi-gcc`, but because I've seen the exact error on my Pi Zero W I'd like to see this looked into.
@cmaglie @silvanocerza can you investigate? ⬆️

ekawahyu, 2020-09-24:
```
pi@raspberrypi:~/pi-wake-on-rtc $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~/pi-wake-on-rtc $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.13 (stretch)
Release:        9.13
Codename:       stretch
pi@raspberrypi:~/pi-wake-on-rtc $ uname -a
Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
```

ubidefeo, 2020-09-25:
ok, so it's not the 64 bit version.
we'll investigate this one and get back to you in a few days 
thanks again for the detailed report 👍

cmaglie, 2020-09-25:
I've pushed a patch for the high memory consumption https://github.com/arduino/arduino-cli/pull/981
Would like to try this builds:

Linux ARM64: https://github.com/arduino/arduino-cli/suites/1244362101/artifacts/18868999
Linux ARMv6: https://github.com/arduino/arduino-cli/suites/1244362101/artifacts/18869000
Linux ARMv7: https://github.com/arduino/arduino-cli/suites/1244362101/artifacts/18869001

and report if they works now?

ubidefeo, 2020-09-25:
@ekawahyu download the release @cmaglie posted above.
You'll need the ARMv7 one

ekawahyu, 2020-09-25:
Confirmed that there is no more out of memory issue. Installation is successful for ARMv7 by running it stand alone as is.

Question, can I overwrite the existing arduino-cli in the bin directory and test the rest of functionalities with the sketch?

ubidefeo, 2020-09-25:
@ekawahyu 
do that but remember that it's not been merged into main so it's just a patch release we made for you to test.
Beginning next week we'll merge it and it will be in the Nightly Build the following day.
Check the changelog to see if it gets updated.

enjoy the weekend :)

ekawahyu, 2020-09-25:
Alright, sounds good! Thank you for the quick turnaround @ubidefeo @cmaglie I am closing this issue for now. Have a great weekend to both of you too!

PR #981 (2020-09-28): Fix high memory usage during archive unpack
Upstream fix:
https://github.com/codeclysm/extract/pull/10
https://github.com/codeclysm/extract/commit/54043e81a1eca308b94f4d242776baabb9c7f7ce

Fix #975
commit 439c4735 (2020-09-28): Fix high memory usage during archive unpack (#981)

Related to:
https://github.com/codeclysm/extract/pull/10
https://github.com/codeclysm/extract/commit/54043e81a1eca308b94f4d242776baabb9c7f7ce

Fix #975
closed 2020-09-25 by ekawahyu