arduino/arduino-cli #976: Is buildCtx.OtherLibrariesDirs even used?

Alfrederson, 2020-09-24:
## Bug Report

### Current behavior

cmd.exe /r "C:\Users\pc\Documents\tool\arduino-cli.exe" compile "C:\Users\pc\Documents\tool\sketch\sketch.ino" -b arduino:avr:diecimila --libraries "C:\Users\pc\Documents\tool\libs" --output-dir "C:\Users\pc\Documents\tool\sketch\out" --verbose

It says it can't find a header file inside \tool\libs directory (with #include "header.h") .

### Expected behavior

It should find the file like it does when the \libs\ folder is inside the same directory as the sketch (#include "libs/header.h")

### Environment

- CLI version (output of `arduino-cli version`):
- OS and platform:

### Additional context

Looking at the source code it seems like the values passed to --libraries are added to a list but aren't referenced ever again. Maybe this is the cause of the issue.

ubidefeo, 2020-09-24:
hi @Alfrederson 

it seems that the `--libraries` flag is causing some issues, we had another report #973 with a different behaviour caused by the same flag and I'm having this investigated.

Thank you for reporting

per1234, 2021-02-03:
Hi @Alfrederson. If I understand what you wrote correctly, the problem is that you are misusing the `--libraries` flag. From the command reference:
https://arduino.github.io/arduino-cli/latest/commands/arduino-cli_compile/#options
```
--libraries strings            List of custom libraries paths separated by commas. Or can be used multiple times for multiple libraries paths.
```
The path specified via the `--libraries` flag is to a libraries folder. Like the `libraries` subfolder of `directories.user` (AKA "the sketchbook folder"), this is a folder containing libraries, like this:
```
C:\Users\pc\Documents\tool
|_ libs
    |_ BarLib
    |    |_ header.h
    |_ FooLib
        |_ library.properties
        |_ src
            |_ header.h
```

It sounds like your `C:\Users\pc\Documents\tool\libs` is actuall a library folder:
```
C:\Users\pc\Documents\tool
|_ libs
    |_ header.h
```
which is not supported.

closed 2021-02-03 by per1234