arduino/arduino-cli #983: compile flag -o removed, how to do this in current version?

harryberlin, 2020-09-25:
## Bug Report

### Current behavior

compile -b arduino:avr:nano -o C:\temp\MySketch\Release\MYNANO C:\Users\MySketch

### Expected behavior

how to use the -o flag with current version?

### Environment

- CLI version (output of `arduino-cli version`): 0.13.0 32bit
- OS and platform: windows 10

### Additional context

ubidefeo, 2020-09-26:
hi @harryberlin 
I think we lost the short flag format for that at some point, but not sure why (@cmaglie / @silvanocerza ?)
you can use `--output-dir` which has recently gotten some love and works very well for what you need to do :)

harryberlin, 2020-09-27:
sorry, it's not the same. now, i get a subfolder.
is there a solution to only get the .hex file without bootloader and other without .eep + .elf?
had a workaroudn to compile with -o flag and delete the other not needed files.

ubidefeo, 2020-09-28:
@harryberlin I've assigned this task
we'll get back to you soon

ubidefeo, 2020-09-28:
@harryberlin 
We discussed this one and essentially the old `-o` was causing issues because not every 3rd party platform developer follows specs and we had complaints because things just broke.
a few releases ago we moved to a system that outputs all the required files into a folder (`build` or one of your choice using the `--output-dir` flag).
We won't reintroduce that flag, so I invite you to either use `--output-dir` and have a cleanup script to be called after your CLI operations or to embrace this behaviour.
Hope it's not too much of an inconvenience, but at least this doesn't break when a 3rd party developer does things their way without reading documentation  😬

closed 2020-09-30 by per1234