arduino/arduino-cli #985: Document / specify core and variant requirements and workings

matthijskooijman, 2020-09-27:
We need some more exhaustive documentation about the "interface" that a core or variant should expose. There is [this wiki page](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification), which documents a good deal of the structure and has examples, but is not exhaustive or explicit. In particular:
- All of the required keys in platform.txt, boards.txt and programmers.txt, and what they are used for, should be listed. These keys form the "external interface" of the core, which can be used by the IDE or arduino-builder. A core can also contain arbitrary other keys, which are used for variable substitutions internally, but cannot be relied on to exist is all cores. Currently, this distinction is blurry and ad-hoc.
- All of the macros, functions and variables that a variant of the Arduino core should expose should be documented. Some of these are mean to be used by the sketch directly (and should actually be documented as part of the normal Arduino reference), but some of them are only used internally by the Arduino core as part of another API. Currenly, no definitive list of these exists, making variants (both third-party and Arduino ones) inconsistent or incomplete.

Both of these pieces of documentation should be updated when the requirements change in future IDE or core versions, with the changes mentioned explicitly and with the IDE or core version that introduced them.

Related issues are arduino/Arduino#3371 and arduino/Arduino#3681.

matthijskooijman, 2020-11-27:
I just noticed that in the area of versioning, the SAMD core already applies some versioning to its variants by letting the variants define a constant to indicate what version of the core-variant interface they implement (I think). I dont' think there's any specific documentation about what this version number implies exactly, though.

See: https://github.com/arduino/ArduinoCore-samd/blob/56e13aa5ebb21f8aca8805e2bc0835a1f1fea372/cores/arduino/Arduino.h#L52

ianfixes, 2021-01-11:
Subscribing for interest, via https://github.com/Arduino-CI/arduino_ci/issues/251

I am trying to put boundaries around what would be considered the proper set of compilation mocks for Arduino, and the lack of official documentation on stuff [like these](https://github.com/arduino/ArduinoCore-avr/blob/366b1b8/cores/arduino/Arduino.h#L182-L184):

```c++
#define portOutputRegister(P) ( (volatile uint8_t *)( pgm_read_word( port_to_output_PGM + (P))) )
#define portInputRegister(P) ( (volatile uint8_t *)( pgm_read_word( port_to_input_PGM + (P))) )
#define portModeRegister(P) ( (volatile uint8_t *)( pgm_read_word( port_to_mode_PGM + (P))) )
```

I can't find any official-looking documentation to say whether these are meant to exist for the long term and/or what their function should be (so that I can create proper unit tests for my mockups of them).