arduino/arduino-cli #986: Unkown board name

dontfindavailablename, 2020-09-28:
## Bug Report
Hello everyone!

I encounter a problem with arduino-cli: the board name is unknown.

I newly installed arduino-cli on ubuntu.
I updated the cores indexes, installed arduino:avr and samd cores, installed different versions (1.6.23 and 1.8.3).

Still when I list the connected boards, the board name FQBN core remains Unknown.

This is the case for all my boards ([arduino nano](https://robotdyn.com/nano-v3-atmega-328-usb-ttl-ch340g-micro-usb.html), mega2560).
This issue doesn't happen with the arduino IDE.

What can I do to fixe this problem?

Thank you for your help!

ubidefeo, 2020-09-28:
hi @dontfindavailablename 

If the board doesn't have specific, fixed `VID/PID` values on the USB device it cannot be recognised automatically.
There are some cores that allow a wide range of VID/PID to be resolved, but that is not necessarily true.

are your boards manufactured by Arduino or are they from another manufacturer?

dontfindavailablename, 2020-09-28:
Thanks for your post.

My board was manufactured by RobotDyn.

I don't understand what is VID/PID.

What should I do?

Thanks a lot!

dontfindavailablename, 2020-09-28:
I've been looking for VIP/PID on internet.

When I run lsusb, I get:
Bus 001 Device 031: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter

ubidefeo, 2020-09-28:
hi @dontfindavailablename 
this manufacturer uses generic usb devices, so it cannot be detected as any particular board

the auto-discovery is based on specific numbers being mapped in the platform files.
Our original boards are recognised because we pay for our own VID and PID numbers.

Hope that answers your question, but there is nothing we can do about this, so I'll close the issue 

regards,
ubi

closed 2020-09-28 by ubidefeo