arduino/arduino-cli #993: Recommend a license for third-party libraries in documentation

Coding-Badly, 2020-09-30:
Please unify the libraries under a single license.  Currently, at least six different licenses are used.

Please also recommend a license for third-party libraries.

per1234, 2021-02-03:
Considering that the Arduino software doesn't currently have any sort of library license detection/enforcement/etc. system, I don't think the specification would be an appropriate place for this. It would make more sense to put it in another document with a different focus, perhaps here: https://www.arduino.cc/en/Hacking/LibraryTutorial

There is now an official tool for encouraging best practices in addition to enforcing specification compliance: [Arduino Lint](https://github.com/arduino/arduino-lint). Arduino Lint does have a rule that checks for the presence of one of the recognized license files [GitHub's automated license detection feature](https://docs.github.com/en/free-pro-team@latest/github/creating-cloning-and-archiving-repositories/licensing-a-repository) uses. This is only a warning when Arduino Lint is in the "specification" or "permissive" compliance modes, but it's upgraded to an error when used in "strict" compliance mode. It doesn't currently parse the contents of the file to determine the license type as GitHub's system does with the[ licensee Ruby gem](https://github.com/licensee/licensee). It would be interesting to try to add that capability to Arduino Lint eventually in a rule that fails if the license type was not recognized.

per1234, 2021-06-07:
I mentioned Arduino Lint's license file check above. [The new Library Manager system](https://blog.arduino.cc/2021/06/07/new-workflow-for-arduino-library-submissions/) takes another step in encouraging licence documentation in Arduino libraries by running Arduino Lint on all libraries submitted to the Arduino Library Manager index, and on every release of the libraries in the index.

At the [compliance setting](https://arduino.github.io/arduino-lint/latest/#compliance-setting) used by the Library Manager system, lack of a licence file only results in a warning, but this warning is displayed during the submission process and also in [the indexer logs](https://github.com/arduino/library-registry/blob/main/FAQ.md#can-i-check-on-library-releases-being-added-to-library-manager). I believe this approach will result in improvements to the overall level of license documentation in the Arduino ecosystem without requiring a huge effort or causing a lot of disruption.