arduino/arduino-cli #995: Feature request: Field incompatible-with in library.properties

bengtmartensson, 2020-09-30:
Some libraries are incompatible with others. Fact of life. I propose a field `incompatible-with` in library.properties, analogues to `depends` and with the same semantic (several entries divided by commas), but with the opposite semantic: The library manager should refuse to install a library if any of the libraries in `incompatible-with` is present. Ideally there should be a granularity down to version too, but let's leave that out for the moment (there is no versions in `depends` either).

Only in this way it is possible to cleanly publish a completely compatible replacement for an existing library.

ubidefeo, 2020-09-30:
@bengtmartensson 
I like the concept, but refusing to install a library because incompatible with other ones already present would not be ideal.
It is down to the sketch level that you want such incompatibility to alert the user

matthijskooijman, 2020-09-30:
Also, often such "incompatible" libraries can still be made compatible by e.g. including an "ImprovedFoo.h" or "OriginalFoo.h" header file that is unique to the improved and/or original version, so a sketch can select one or the other.