arduino/arduino-cli #996: Missing FQBNs for non AVR boards

PeterWone, 2020-09-30:
## Bug Report

### Current behavior
Boards listed for the arduino:avr core specify FQBNs. 
```
  {
    "ID": "arduino:avr",
    "Latest": "1.8.3",
    "Name": "Arduino AVR Boards",
    "Maintainer": "Arduino",
    "Website": "http://www.arduino.cc/",
    "Email": "packages@arduino.cc",
    "Boards": [
      {
        "name": "LilyPad Arduino USB",
        "fqbn": "arduino:avr:LilyPadUSB"
      },
      {
        "name": "Arduino Industrial 101",
        "fqbn": "arduino:avr:chiwawa"
      },
```

Boards listed for other cores do not provide FQBN.

```
  {
    "ID": "arduino:sam",
    "Latest": "1.6.12",
    "Name": "Arduino SAM Boards (32-bits ARM Cortex-M3)",
    "Maintainer": "Arduino",
    "Website": "http://www.arduino.cc/",
    "Email": "packages@arduino.cc",
    "Boards": [
      {
        "name": "Arduino Due"
      }
    ]
  },
  {
    "ID": "arduino:samd",
    "Latest": "1.8.8",
    "Name": "Arduino SAMD Boards (32-bits ARM Cortex-M0+)",
    "Maintainer": "Arduino",
    "Website": "http://www.arduino.cc/",
    "Email": "packages@arduino.cc",
    "Boards": [
      {
        "name": "Arduino MKR WiFi 1010"
      },
```
The FQBN is _required_ by `arduino-cli compile ` and other commands.

```arduino-cli core search --format json ```

### Expected behavior
All boards for which an FQBN is defined should specify FQBN in the response to `core search`

### Environment

- CLI version (output of `arduino-cli version`): arduino-cli Version: 0.13.0 Commit: 693a045
- OS and platform: Windows 10 build 2004

### Additional context

67 boards are reported for various cores. 30 are from arduino:avr and only these report FQBN.

ubidefeo, 2020-09-30:
@PeterWone 
that looks odd
I have tried the same and it works
```
{
    "ID": "arduino:sam",
    "Latest": "1.6.12",
    "Name": "Arduino SAM Boards (32-bits ARM Cortex-M3)",
    "Maintainer": "Arduino",
    "Website": "http://www.arduino.cc/",
    "Email": "packages@arduino.cc",
    "Boards": [
      {
        "name": "Arduino Due (Programming Port)",
        "fqbn": "arduino:sam:arduino_due_x_dbg"
      },
      {
        "name": "Arduino Due (Native USB Port)",
        "fqbn": "arduino:sam:arduino_due_x"
      }
    ]
  }
```

and 
```
{
        "name": "Arduino MKR WiFi 1010",
        "fqbn": "arduino:samd:mkrwifi1010"
      }
```

PeterWone, 2020-09-30:
I have an hypothesis that I am currently testing. This one may also turn out to be a documentation thing.

PeterWone, 2020-09-30:
I can now reproduce consistent behaviour.

A board's FQBN is available _only when the core for that board is installed_.

ubidefeo, 2020-09-30:
@PeterWone the timing is right
I just found out another bug with the `core search` in that it only searches within the platform's name.
Thank you for filing #997 
I'll close this one and move the discussion there

closed 2020-09-30 by ubidefeo