arduino/arduino-cli #997: Board names are inconsistent between core installed and core not installed

PeterWone, 2020-09-30:
## Bug Report

### Current behavior
Without the arduino:avr core installed, `arduino-cli core search arduino:avr --format json` yields the following. In particular, note the presents of a board with the name "Arduino ATMegaNG"
```
[
  {
    "ID": "arduino:avr",
    "Latest": "1.8.3",
    "Name": "Arduino AVR Boards",
    "Maintainer": "Arduino",
    "Website": "http://www.arduino.cc/",
    "Email": "packages@arduino.cc",
    "Boards": [
      {
        "name": "Arduino Yún"
      },
      {
        "name": "Arduino Uno"
      },
      {
        "name": "Arduino Uno WiFi"
      },
      {
        "name": "Arduino Diecimila"
      },
      {
        "name": "Arduino Nano"
      },
      {
        "name": "Arduino Mega"
      },
      {
        "name": "Arduino MegaADK"
      },
      {
        "name": "Arduino Leonardo"
      },
      {
        "name": "Arduino Leonardo Ethernet"
      },
      {
        "name": "Arduino Micro"
      },
      {
        "name": "Arduino Esplora"
      },
      {
        "name": "Arduino Mini"
      },
      {
        "name": "Arduino Ethernet"
      },
      {
        "name": "Arduino Fio"
      },
      {
        "name": "Arduino BT"
      },
      {
        "name": "Arduino LilyPadUSB"
      },
      {
        "name": "Arduino Lilypad"
      },
      {
        "name": "Arduino Pro"
      },
      {
        "name": "Arduino ATMegaNG"
      },
      {
        "name": "Arduino Robot Control"
      },
      {
        "name": "Arduino Robot Motor"
      },
      {
        "name": "Arduino Gemma"
      },
      {
        "name": "Adafruit Circuit Playground"
      },
      {
        "name": "Arduino Yún Mini"
      },
      {
        "name": "Arduino Industrial 101"
      },
      {
        "name": "Linino One"
      }
    ]
  }
]
```

After installing the arduino:avr core the same command produces this.

```
[
  {
    "ID": "arduino:avr",
    "Latest": "1.8.3",
    "Name": "Arduino AVR Boards",
    "Maintainer": "Arduino",
    "Website": "http://www.arduino.cc/",
    "Email": "packages@arduino.cc",
    "Boards": [
      {
        "name": "Arduino Gemma",
        "fqbn": "arduino:avr:gemma"
      },
      {
        "name": "Adafruit Circuit Playground",
        "fqbn": "arduino:avr:circuitplay32u4cat"
      },
      {
        "name": "Arduino Pro or Pro Mini",
        "fqbn": "arduino:avr:pro"
      },
      {
        "name": "Arduino Yún Mini",
        "fqbn": "arduino:avr:yunmini"
      },
      {
        "name": "Arduino Robot Motor",
        "fqbn": "arduino:avr:robotMotor"
      },
      {
        "name": "Arduino Robot Control",
        "fqbn": "arduino:avr:robotControl"
      },
      {
        "name": "Arduino Ethernet",
        "fqbn": "arduino:avr:ethernet"
      },
      {
        "name": "Arduino Leonardo",
        "fqbn": "arduino:avr:leonardo"
      },
      {
        "name": "Arduino Mini",
        "fqbn": "arduino:avr:mini"
      },
      {
        "name": "Arduino Fio",
        "fqbn": "arduino:avr:fio"
      },
      {
        "name": "Arduino Esplora",
        "fqbn": "arduino:avr:esplora"
      },
      {
        "name": "Arduino Industrial 101",
        "fqbn": "arduino:avr:chiwawa"
      },
      {
        "name": "Arduino Yún",
        "fqbn": "arduino:avr:yun"
      },
      {
        "name": "Linino One",
        "fqbn": "arduino:avr:one"
      },
      {
        "name": "Arduino Micro",
        "fqbn": "arduino:avr:micro"
      },
      {
        "name": "Arduino Nano",
        "fqbn": "arduino:avr:nano"
      },
      {
        "name": "Arduino BT",
        "fqbn": "arduino:avr:bt"
      },
      {
        "name": "Arduino Duemilanove or Diecimila",
        "fqbn": "arduino:avr:diecimila"
      },
      {
        "name": "Arduino Uno",
        "fqbn": "arduino:avr:uno"
      },
      {
        "name": "Arduino Uno WiFi",
        "fqbn": "arduino:avr:unowifi"
      },
      {
        "name": "Arduino Mega or Mega 2560",
        "fqbn": "arduino:avr:mega"
      },
      {
        "name": "Arduino Leonardo ETH",
        "fqbn": "arduino:avr:leonardoeth"
      },
      {
        "name": "Arduino NG or older",
        "fqbn": "arduino:avr:atmegang"
      },
      {
        "name": "LilyPad Arduino USB",
        "fqbn": "arduino:avr:LilyPadUSB"
      },
      {
        "name": "Arduino Mega ADK",
        "fqbn": "arduino:avr:megaADK"
      },
      {
        "name": "LilyPad Arduino",
        "fqbn": "arduino:avr:lilypad"
      }
    ]
  }
]
```
 "Arduino ATMegaNG" has vanished, replaced by "Arduino NG or older".

### Expected behavior
The names should be consistent before and after core installation.

### Environment

- CLI version (output of `arduino-cli version`): arduino-cli Version: 0.13.0 Commit: 693a045
- OS and platform: Windows 10 build 2004

### Additional context

Board only has two properties. FQBN isn't available before the relevant core is installed. I install the core when the user selects a board that lacks an FQBN and then do a fixup of my board objects. Except the fixup barfs because the names don't all match before and after.

ubidefeo, 2020-09-30:
@silvanocerza whilst we're looking into the behaviour of `core search` let's keep this in mind

PeterWone, 2020-09-30:
You know, I was just looking through your repo and found this
```
          "boards": [
            {"name": "Arduino Yún"},
            {"name": "Arduino Uno"},
            {"name": "Arduino Diecimila"},
            {"name": "Arduino Nano"},
            {"name": "Arduino Mega"},
            {"name": "Arduino MegaADK"},
            {"name": "Arduino Leonardo"},
            {"name": "Arduino Micro"},
            {"name": "Arduino Esplora"},
            {"name": "Arduino Mini"},
            {"name": "Arduino Ethernet"},
            {"name": "Arduino Fio"},
            {"name": "Arduino BT"},
            {"name": "Arduino LilyPadUSB"},
            {"name": "Arduino Lilypad"},
            {"name": "Arduino Pro"},
            {"name": "Arduino ATMegaNG"},
            {"name": "Arduino Robot Control"},
            {"name": "Arduino Robot Motor"}
          ],
```

It repeats for each platform version but I strongly suspect that if the FQBNs were added to this JSON they would be delivered.

ubidefeo, 2020-09-30:
@PeterWone 
I'm not sure it's because of that, but I'll prompt for an examination.
Thank you for helping us troubleshoot 👍

per1234, 2020-09-30:
Closing as duplicate of https://github.com/arduino/arduino-cli/issues/620

To make things clear, the source of the boards list when the boards platform is not installed comes from the `packages[].platforms[].boards` field of its [package index](https://arduino.github.io/arduino-cli/latest/package_index_json-specification/) (in this case https://downloads.arduino.cc/packages/package_index.json):
```json
        {
          "name": "Arduino AVR Boards",
          "architecture": "avr",
          "version": "1.8.3",
          "category": "Arduino",
          "help": {
            "online": "http://www.arduino.cc/en/Reference/HomePage"
          },
          "url": "http://downloads.arduino.cc/cores/avr-1.8.3.tar.bz2",
          "archiveFileName": "avr-1.8.3.tar.bz2",
          "checksum": "SHA-256:de8a9b982477762d3d3e52fc2b682cdd8ff194dc3f1d46f4debdea6a01b33c14",
          "size": "4941548",
          "boards": [
            {"name": "Arduino Yún"},
            {"name": "Arduino Uno"},
            {"name": "Arduino Uno WiFi"},
            {"name": "Arduino Diecimila"},
            {"name": "Arduino Nano"},
            {"name": "Arduino Mega"},
            {"name": "Arduino MegaADK"},
            {"name": "Arduino Leonardo"},
            {"name": "Arduino Leonardo Ethernet"},
            {"name": "Arduino Micro"},
            {"name": "Arduino Esplora"},
            {"name": "Arduino Mini"},
            {"name": "Arduino Ethernet"},
            {"name": "Arduino Fio"},
            {"name": "Arduino BT"},
            {"name": "Arduino LilyPadUSB"},
            {"name": "Arduino Lilypad"},
            {"name": "Arduino Pro"},
            {"name": "Arduino ATMegaNG"},
            {"name": "Arduino Robot Control"},
            {"name": "Arduino Robot Motor"},
            {"name": "Arduino Gemma"},
            {"name": "Adafruit Circuit Playground"},
            {"name": "Arduino Yún Mini"},
            {"name": "Arduino Industrial 101"},
            {"name": "Linino One"}
          ],
          "toolsDependencies": [
            {
              "packager": "arduino",
              "name": "avr-gcc",
              "version": "7.3.0-atmel3.6.1-arduino7"
            },
            {
              "packager": "arduino",
              "name": "avrdude",
              "version": "6.3.0-arduino17"
            },
            {
              "packager": "arduino",
              "name": "arduinoOTA",
              "version": "1.3.0"
            }
          ]
        },
```
After the boards platform is installed, the boards list comes from the `BOARD_ID.name` properties in [boards.txt](https://arduino.github.io/arduino-cli/latest/platform-specification/#boardstxt) file:
https://github.com/arduino/ArduinoCore-avr/blob/1.8.3/boards.txt

The first is only intended to provide a summary of the boards provided by the platform in the Boards Manager installation interface. The latter defines the actual boards provided by the platform. So you can't expect the two to be identical.

closed 2020-09-30 by per1234