arduino/arduino-cli #999: Global system platform.txt

NicoHood, 2020-10-01:
It would be nice to have a platform.txt that cannot be overwritten by the avr-core.
For example I need to change `compiler.path={runtime.tools.avr-gcc.path}/bin/` to `compiler.path=/usr/bin/`. But I also need to add some extra cpp flags that are specific for the avr-gcc 6.x.

I am still not sure whats the best way to solve this, but if we can make the distribution avr-gcc packages more accessible its simpler to package arduino for linux and also especially for archlinux never avr-gcc can already be tested for bugs.

per1234, 2020-10-01:
If I understand the request correctly, we now have this feature, but only for Arduino IDE:
https://arduino.github.io/arduino-cli/latest/platform-specification/#global-platformtxt

ubidefeo, 2020-10-01:
@per1234 
you beat me to the punch as usual :D

fabianoriccardi, 2022-09-17:
I agree with the previous posts, I would be great to have this feature back or, at least, another system to run pre-/ post- scripts before/after a build.

Is still planned?

closed 2025-07-14 by alessio-perugini

per1234, 2025-07-14:
Thanks for your request @NicoHood. This was actually resolved long ago by https://github.com/arduino/arduino-cli/pull/472. However, that was not known due to a misimpression that the feature was specific to the IDE-bundled packages folder unique to the Arduino IDE 1.x application. It turns out that global `platform.txt` files are (and always were) recognized in any of the packages folders; not the Arduino IDE 1.x bundled packages folder exclusively. Thus, we now understand that this issue is resolved.

The misinformation in the Arduino Platform Specification regarding the scope of the global `platform.txt` file has been corrected: https://github.com/arduino/arduino-cli/pull/2952. In addition, the locations of all the packages folders have been documented:

https://arduino.github.io/arduino-cli/dev/platform-specification/#global-platformtxt