arduino/arduino-cli #1002: Can't compile for esp32 when setting config directories outside default Arduino15 on windows

lazaroofarrill, 2020-10-02:
## Bug Report

### Current behavior
I'm trying to make a portable setup of arduino-cli with the platforms bundled next to the app but i get this error when trying to compile for esp32 platforms.

this is what i get when loading the cli with a custom data folder next to the executable

`
PS E:\Software\Development\ARDUINO\arduino-cli\arduino-cli_0.13.0_Windows_64bit> arduino-cli.exe compile -b esp32:esp32:esp32doit-devkit-v1 .\sketcher\
In file included from e:\software\development\arduino\arduino-cli\arduino-cli_0.13.0_windows_64bit\arduinodata\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\algorithm:60:0,
                 from E:\Software\Development\ARDUINO\arduino-cli\arduino-cli_0.13.0_Windows_64bit\ArduinoData\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Arduino.h:142,
                 from C:\Users\LAZARO~1\AppData\Local\Temp\arduino-sketch-4E21FC4C5D191BC180FDD3CAC2586C3F\sketch\sketcher.ino.cpp:1:
e:\software\development\arduino\arduino-cli\arduino-cli_0.13.0_windows_64bit\arduinodata\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\utility:68:28: fatal error: bits/c++config.h: No such file or directory
compilation terminated.
Error during build: exit status 1`

### Expected behavior
This is what i get when compiling using default directories
`PS E:\> arduino-cli.exe compile -b esp32:esp32:esp32doit-devkit-v1 .\sketcher\
Sketch uses 206423 bytes (15%) of program storage space. Maximum is 1310720 bytes.
Global variables use 14556 bytes (4%) of dynamic memory, leaving 313124 bytes for local variables. Maximum is 327680 bytes.
PS E:\>`

### Environment

- CLI version (arduino-cli.exe Version: 0.13.0 Commit: 693a045`):
- Windows 10

So far this problem is exclusive to esp32 boards, arduino avr boards compile with no problem in here.
Can anyone help me

matthijskooijman, 2020-10-03:
Some additional info that could be useful:
 - The ESP32 core version you have installed
 - Running arduino-cli with -v, to show the exact compiler commands it ran.

It seems like there is something wrong with the builtin include paths from the compiler, which is weird. Could you check:
 - That inside your "portable" data dir, this file exists: `packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits\c++config.h`
 - Show the output of this command: `e:\software\development\arduino\arduino-cli\arduino-cli_0.13.0_windows_64bit\arduinodata\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\bin\xtensa-esp32-elf-g++ -E -v -x c++ empty.file.cpp` (you'll have to create an empty file in the current directory called `empty.file.cpp` first).

lazaroofarrill, 2020-10-03:
Yes, I confirm the file existed. I even tried copying all the files in bits to a bits folder higher in the file hierarchy and the compilation advanced beyond that point only to stop in a similar error, i repeated the procedure over and over again until i had duplicated files everywhere, but new errors kept showing.

I can't reproduce the exact same error at the time because i found this workoround.

**I managed to compile for the platform deleting the cores installed by the board manager and cloning the esp32 arduino github repository and installing it in my custom sketchfolder.**

now i'm getting the same error when compiling for stm32 platform, but i could'nt solve it the same way because even the development version of the github repo depends on a previous installation through the board manager of the tools. Like with the esp32, installing to the default directory in appdata\Arduino15 compiles correctly

command:
`.\arduino-cli.exe compile -b stm32duino:STM32F1:genericSTM32F103C6 .\sketcher\ -v`

error message:
`In file included from E:\Software\Development\ARDUINO\arduino-cli\arduino-cli_0.13.0_Windows_64bit\ArduinoData\packages\stm32duino\hardware\STM32F1\2020.8.9\cores\maple\cxxabi-compat.cpp:1:0:
e:\software\development\arduino\arduino-cli\arduino-cli_0.13.0_windows_64bit\arduinodata\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\c++\4.8.3\cstdlib:41:28: fatal error: bits/c++config.h: No such file or directory
 #include <bits/c++config.h>
                            ^
compilation terminated.
Error during build: exit status 1`

full log:
[log.txt](https://github.com/arduino/arduino-cli/files/5321858/log.txt)

> Show the output of this command: e:\software\development\arduino\arduino-cli\arduino-cli_0.13.0_windows_64bit\arduinodata\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\bin\xtensa-esp32-elf-g++ -E -v -x c++ empty.file.cpp

I can't reproduce that same command so im using this that i'm gessing is more or less the same:
```
PS E:\Software\Development\ARDUINO\arduino-cli\arduino-cli_0.13.0_Windows_64bit\ArduinoData\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\bin> .\arm-none-eabi-g++.exe -E -v -x c++ empty.file.cpp
Using built-in specs.
COLLECT_GCC=E:\Software\Development\ARDUINO\arduino-cli\arduino-cli_0.13.0_Windows_64bit\ArduinoData\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\bin\arm-none-eabi-g++.exe
Target: arm-none-eabi
Configured with: /home/build/work/GCC-4-8-build/src/gcc/configure --build=i686-linux-gnu --host=i686-w64-mingw32 --target=arm-none-eabi --prefix=/home/build/work/GCC-4-8-build/install-mingw --libexecdir=/home/build/work/GCC-4-8-build/install-mingw/lib --infodir=/home/build/work/GCC-4-8-build/install-mingw/share/doc/gcc-arm-none-eabi/info --mandir=/home/build/work/GCC-4-8-build/install-mingw/share/doc/gcc-arm-none-eabi/man --htmldir=/home/build/work/GCC-4-8-build/install-mingw/share/doc/gcc-arm-none-eabi/html --pdfdir=/home/build/work/GCC-4-8-build/install-mingw/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-headers=yes --with-newlib --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/home/build/work/GCC-4-8-build/install-mingw/arm-none-eabi --with-libiconv-prefix=/home/build/work/GCC-4-8-build/build-mingw/host-libs/usr --with-gmp=/home/build/work/GCC-4-8-build/build-mingw/host-libs/usr --with-mpfr=/home/build/work/GCC-4-8-build/build-mingw/host-libs/usr --with-mpc=/home/build/work/GCC-4-8-build/build-mingw/host-libs/usr --with-isl=/home/build/work/GCC-4-8-build/build-mingw/host-libs/usr --with-cloog=/home/build/work/GCC-4-8-build/build-mingw/host-libs/usr --with-libelf=/home/build/work/GCC-4-8-build/build-mingw/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='GNU Tools for ARM Embedded Processors' --with-multilib-list=armv6-m,armv7-m,armv7e-m,armv7-r
Thread model: single
gcc version 4.8.3 20140228 (release) [ARM/embedded-4_8-branch revision 208322] (GNU Tools for ARM Embedded Processors)
COLLECT_GCC_OPTIONS='-E' '-v'
 e:/software/development/arduino/arduino-cli/arduino-cli_0.13.0_windows_64bit/arduinodata/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/cc1plus.exe -E -quiet -v -iprefix e:\software\development\arduino\arduino-cli\arduino-cli_0.13.0_windows_64bit\arduinodata\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\bin\../lib/gcc/arm-none-eabi/4.8.3/ -isysroot e:\software\development\arduino\arduino-cli\arduino-cli_0.13.0_windows_64bit\arduinodata\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\bin\../arm-none-eabi -D__USES_INITFINI__ empty.file.cpp
ignoring duplicate directory "e:/software/development/arduino/arduino-cli/arduino-cli_0.13.0_windows_64bit/arduinodata/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/lib/gcc/../../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/include/c++/4.8.3"
ignoring duplicate directory "e:/software/development/arduino/arduino-cli/arduino-cli_0.13.0_windows_64bit/arduinodata/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/lib/gcc/../../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/include/c++/4.8.3/arm-none-eabi"
[... cut ...] _windows_64bit/arduinodata/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/;e:/software/development/arduino/arduino-cli/arduino-cli_0.13.0_windows_64bit/arduinodata/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../arm-none-eabi/lib/
COLLECT_GCC_OPTIONS='-E' '-v'
PS E:\Software\Development\ARDUINO\arduino-cli\arduino-cli_0.13.0_Windows_64bit\ArduinoData\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\bin>
```

matthijskooijman, 2020-10-08:
I'm totally unsure what causes this problems. The include paths look good, in particular this one should contain `bits/c++config.h` AFAIU:

     e:\software\development\arduino\arduino-cli\arduino-cli_0.13.0_windows_64bit\arduinodata\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\bin\../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/include/c++/4.8.3/arm-none-eabi

So I don't know why gcc doesn't find it. Maybe there is some virus scanner or other protection active that somehow triggers on this file but not others?

jdressu, 2021-03-31:
> The include pa

Hey @matthijskooijman , i was having a very similar problem yesterday. But i think my console output was a bit more "vague" (for a lack of a better work). Was cracking my head around the problem because i didn't see anything/anyone referencing my problem.

I'm not sure if this is the detail you need, but i solved my problem by moving everything (cli + sketch) to the operation system's main drive.

I say that because i noticed that @lazaroofarrill 's console outputs show '2 different'/'multiple' drives 'C:' and 'E:' (like in my case 'C:' and 'U:').
And after i moved everything onto a single drive the cli seemed to get the paths correctly and compiled without any problems.

lazaroofarrill, 2021-03-31:
Forgot about this thread. The problem is with windows max path length. The package was too deep in my file system and the path got too long, that is why it only fails on windows.

closed 2021-03-31 by lazaroofarrill