arduino/arduino-cli #1004: can't install arduino-cli on an aarch64 machine: official arm64 binary is probably compiled for arm32

nestukh, 2020-10-03:
## Bug Report
_Bug originally mentined in https://github.com/arduino/arduino-cli/issues/292#issuecomment-699613053_

A Vero 4K+ (by OSMC) is a Kodi box which has Debian with aarch64 (arm64) architecture. It's very versatile but it hasn't Xorg installed. It has its own debian repo `apt.osmc.tv` and uses the official `http://ftp.debian.org/debian stretch main contrib non-free` as well. Yes the stable version is still on Debian Stretch but it's very stable.
It can be accessed on the command line via ssh over Wi-Fi. The default user is named `osmc`, who has sudo privileges as well.

### Current behavior
installing with the command
`curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh`
I obtain
```
Installing in /home/osmc/bin
ARCH=ARM64
OS=Linux
Using curl as download tool
TAG=0.13.0
CLI_DIST=arduino-cli_0.13.0_Linux_ARM64.tar.gz
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_0.13.0_Linux_ARM64.tar.gz
main: line 204: /home/osmc/bin/arduino-cli: No such file or directory
Failed to install arduino-cli
```
But the binary `/home/osmc/bin/arduino-cli` exists, ergo it's made for the wrong arch, like it was happening in [issue 307](https://github.com/arduino/arduino-cli/pull/307). I manually installed the `arduino-cli` binary [from arduino-cli-152-PR307-linuxarm64.tar.bz2 found in a comment there](https://github.com/arduino/arduino-cli/pull/307#issuecomment-514613081) and it works: the output of `arduino-cli version` with the 152-PR307 version is
`arduino-cli Version: 152-PR307 Commit: 77d5613`
and can run other subrutines as well.

I probably could install the 32 bit armhf architeture like suggested online in some places (like [here](https://askubuntu.com/a/1090440)), but it don't like to mess up things and hope for the best. Space is also limited.
It there is a clean solution that you know of, the `avr-g++` binary from the main arduino installer has the same problem on this machine: running a normal command like `arduino-cli compile --fqbn arduino:avr:uno Sketch01`, I'm getting a classic error:
```
Error: build failed: fork/exec /home/osmc/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++: no such file or directory
Error during build
```
`avr-g++` exists but it's for the wrong arch for sure. But this problem cannot be solved removing the `.arduino15` folder like suggested in many places.

matthijskooijman, 2020-10-09:
> I probably could install the 32 bit armhf architeture like suggested online in some places (like here)

As we figured out in https://github.com/arduino/toolchain-avr/issues/73, the problem is actually reversed: You have a 32-bit userspace already, so should be using a 32-bit arduino-cli rather than a 64-bit arduino-cli.

I can't explain why the older PR307 version you linked, which is also 64-bit, *does* work, but I don't find it particularly interesting either, since I don't think the 64-bit version can be expected to work on your system.

So, the bug I see here is that the arduino-cli `install.sh` script misdetects your system as a 64-bit system, presumably because it is running a 64-bit kernel, even though the userspace is 32-bit. This *is* a bug, since the installer script should detect and install the right version, even on a more complicated system like this, I think. I'm not sure how easy this is to fix, though.

umbynos, 2022-11-24:
we use `uname` to detect the architecture [in the install script](https://github.com/arduino/arduino-cli/blob/d86bc138c186db94cc76a488b02682eb8a50d739/install.sh#L37). Could you please run `uname -a` and tell us what is reporting?
To me it seems a strange setup to support

nestukh, 2022-11-24:
> we use `uname` to detect the architecture [in the install script](https://github.com/arduino/arduino-cli/blob/d86bc138c186db94cc76a488b02682eb8a50d739/install.sh#L37). Could you please run `uname -a` and tell us what is reporting? To me it seems a strange setup to support

On the Vero 4K+, with the latest updates, `uname -a` returns
`Linux hostname 4.9.269-16-osmc #1 SMP PREEMPT Tue Nov 1 15:44:25 UTC 2022 aarch64 GNU/Linux`

cmaglie, 2022-11-25:
From the `uname` we get the `aarch64`, but apparently, this is wrong. Do you know how to reliably detect the user-space architecture?

nestukh, 2022-11-26:
both `dpkg -l | grep " apt " | awk '{ print $4 }'` and `dpkg --print-architecture` return `armhf`

`arch` returns `aarch64`

`lscpu | grep -i Architecture` returns 
```
Architecture:        aarch64
Byte Order:          Little Endian
CPU(s):              4
On-line CPU(s) list: 0-3
Thread(s) per core:  1
Core(s) per socket:  4
Socket(s):           1
Vendor ID:           ARM
Model:               4
Model name:          Cortex-A53
Stepping:            r0p4
CPU max MHz:         1512.0000
CPU min MHz:         100.0000
BogoMIPS:            48.00
Flags:               fp asimd evtstrm aes pmull sha1 sha2 crc32
```

`file -L /sbin/init` returns 
```
/sbin/init: ELF 32-bit LSB pie executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, BuildID[sha1]=a255bbe9c1a8e632c6190fbe1d3e7009c8456210, for GNU/Linux 3.2.0, stripped
```