arduino/arduino-cli #1005: does arduino-cli require admin right on Windows OS ?

VercorsSciences, 2020-10-04:
Hello,
I'm wondering if the arduino-cli requires admin right on windows OS ? For now, I've tried on some Linux and Windows with admin rights profile and it works properly.

Once I try to use cli with user right only, it fails with an invalid handle error (seems to come from req.GetInstance().GetId()  )

```
C:\Users\xxx\Downloads>arduino-cli core update-index --additional-urls https://downloads.arduino.cc/packages/package_staging_index.json
Downloading missing tool builtin:ctags@5.8-arduino11
Error updating index: invalid handle
```

- CLI version 0.13.0
- OS : Windows 10 x64

Thanks

ubidefeo, 2020-10-05:
@VercorsSciences 
I think the user running the command only needs rights on its own folder, since tools and platforms are saved to the AppData folder in the user's home.
I have just tried that and it works on my Windows 10 x64 install, I only have one user and it's the admin, though

I have tried removing the Arduino15 folder from AppData and tried again, worked

can you run the command passing `-v` as a flag and post the output?
thanks

VercorsSciences, 2020-10-06:
Hi @ubidefeo ,
Thanks for your feedback.
Here is the log :
```
C:\Users\xxx\Downloads>arduino-cli.exe core update-index -v
INFO[0000] Config file not found, using default values
INFO[0000] arduino-cli.exe version 0.13.0
INFO[0000] Checking if CLI is Bundled into the IDE
INFO[0000] Loading hardware from: C:\Users\xxxx\AppData\Local\Arduino15\packages
INFO[0000] Adding libraries dir                          dir="C:\\Users\\xxxx\\Documents\\Arduino\\libraries" location=user
Downloading missing tool builtin:ctags@5.8-arduino11...
INFO[0000] Executing `arduino core update-index`
Error updating index: invalid handle
ERRO[0000] Error updating index: invalid handle
```

I understand that it's not a bug related to arduino-cli but I would like to find a way to investigate in.
What does `invalid handle` stand for ? Is it related to its own PID (instance) number ?

ubidefeo, 2020-10-06:
@VercorsSciences 
this is something that I don't believe we've seen reported yet

@silvanocerza @cmaglie what do you think is happening here?

cmaglie, 2020-10-06:
It may be a firewall/antivirus?

VercorsSciences, 2020-10-06:
Interesting, and yes it could be... McCafee is running on and it could be disabled (as explained, I don't have admin rights)
Could you more elaborate your thoughs and how would a firewall/antivirus affect the app and generate an invalid handle ?
Does this issue come from an unreachable repo or from an internal error ?

Groupe-DEMS, 2022-11-25:
Buongiorno,
Thanks for your reminder. As time has passed, I'm not sure to set up the same environment... so many things have changed.
It's an old issue. Maybe it's time to close this one. I will open a new one if I can repoduce it.

VercorsSciences, 2022-11-25:
Buongiorno,
Thanks for your reminder. As time has passed, I'm not sure to set up the same environment... so many things have changed.
It's an old issue. Maybe it's time to close this one. I will open a new one if I can repoduce it.

closed 2022-11-25 by VercorsSciences